letscontrolit/ESPEasy #4230: ESP_Easy_mega_20220809_neopixel_ESP8266_4M1M onboard led

fly74, 2022-08-29:
@tonhuisman Hi ton, why the onboard led is always on in ESP_Easy_mega_20220809_neopixel_ESP8266_4M1M with Output - NeoPixel (BusFX) ?

tonhuisman, 2022-08-29:
That depends on the GPIO used for connecting the NeoPixels, most likely you use GPIO-2, and that's also where the Led is connected to. One way to 'silence' it is by 'uninstalling' it from the board, or by using another GPIO.

fly74, 2022-08-29:
The problem is gpio 2 can not be changed and is default in the plugin. 🤨

tonhuisman, 2022-08-29:
Unfortunately, that is a limitation of the library used, that can only work with GPIO-2, on ESP8266.
You could use an ESP32 board, like [this one](https://aliexpress.com/item/32858054775.html) (hope that Ali-link still works) that is nearly as small as a D1 Mini, as for ESP32 the library allows pin selection.

NB: By 'uninstalling' I intended to say to cut or de-solder the led from the board 😮

fly74, 2022-08-29:
It looks like it I will cut it off.

closed 2022-08-29 by fly74

ghtester, 2022-08-29:
You can cover the LED with non-transparent adhesive tape instead of removing it.  ;-)