letscontrolit/ESPEasy #4234: New CSS style creates ugly sub-tables

uwekaditz, 2022-08-30:
With the last two commits ( #4228 and [Build CDN] Serve ESPEasy auto css by default) the style of sub-tables looks ugly.
The auto-size functionality of columns does not work any more.

Here are some examples:
`P016_IR:`
**Old CSS** (before the above mentioned changes):
[image]
**New CSS**
[image]
`P036_FrameOLED:`
**Old CSS (using xwide)** (before the above mentioned changes):
[image]
**Old CSS (without xwide)** (before the above mentioned changes):
[image]
**New CSS**
[image]

TD-er, 2022-08-30:
@chromoxdor 
Can you have a look here?

chromoxdor, 2022-08-30:
working on it..

chromoxdor, 2022-08-30:
So the css fix is easy but the table still breaks the hole site if viewed on small displays:

This is how the table is looks now:
[image]
[image]

And this is how it should look (with a class name i can address the table and it doesn’t mess up the rest of the css)
[image]
[image]

So P016 has to be edited ... i would really appreciate it if somebody could do this otherwise i have to get myself familiar with the code...

chromoxdor, 2022-08-30:
ok fixed it in P016 by making the table a "normal" one...
@uwekaditz  can you check if it works for you?

TD-er, 2022-08-31:
Just keep in mind, we have control of both the HTML and the CSS.
So if you can't fix it in CSS, then adding a class to some HTML elements isn't that much of a hassle.
Which table should have which class added?

chromoxdor, 2022-08-31:
Here the fix: #4238 
@uwekaditz can you test it? And where do i find the subtable in P036_FrameOLED? It is not in mega right? I would like to test this too

TD-er, 2022-08-31:
@chromoxdor See: https://github.com/letscontrolit/ESPEasy/pull/3997

chromoxdor, 2022-08-31:
and i guess i have to add "xwide" to the css eventually ...

uwekaditz, 2022-08-31:
@chromoxdor Looks much better
P036:
[image]
or P016:
[image]
In P016 you forgot one code line:
```
addHtml(F("<TR><TD colspan=\"2\">"));
addRowLabel(F("Code"));   // <- this code was forgotten
html_table(F("sub"));
```
The old "xwide" is just setting the width to 100%
```
input.xwide,
select.xwide {
	max-width: 500px;
	width: 100%
}
```
In order to have a defined gap between the columns the following code has to be used, too:
`html_TD(F("padding-right: 8px")); // text box is (100% - 8 pixel) on right side wide`

It would be great if this can be directly done by
calling `addTextBox()` with the "xwide" class name.

uwekaditz, 2022-08-31:
@chromoxdor By the way is there a css setting to adjust the Up/Down buttons better to the dark mode:
Dark mode:
[image]
Bright mode:
[image]

chromoxdor, 2022-08-31:
> @chromoxdor By the way is there a css setting to adjust the Up/Down buttons better to the dark mode:

There is always a way 😁
But...... the css and the webui as well are full of compromises to accomondate to the filesize. 
So we could make a perfect webui with perfect css but it will always come at a cost.

chromoxdor, 2022-08-31:
> In P016 you forgot one code line:

actually i removed it because i thought it is redundant to the headline.
But for consistency with at least P036 i can add it back...

TD-er, 2022-08-31:
> But...... the css and the webui as well are full of compromises to accomondate to the filesize.
> So we could make a perfect webui with perfect css but it will always come at a cost.

At least the costs of me frowning upon those and asking to defend the cost of those extra bytes.
So better be prepared then!

chromoxdor, 2022-08-31:
> The old "xwide" is just setting the width to 100%
> 
> ```
> input.xwide,
> select.xwide {
> 	max-width: 500px;
> 	width: 100%
> }
> ```

with setting this in css you don´t need this anymore:
[image]

tonhuisman, 2022-08-31:
> for consistency with at least P036 i can add it back...

I only remove that label if I have a lot of columns in the table, like P104, and also now P123 (still in PR, and probably will have the label), but for UI consistency it might be better to keep the label when it fits.

uwekaditz, 2022-08-31:
I know, but for using a text box (`addTextBox()`) following css style is valid:
```
input.wide,select.wide {
	margin-bottom:5px;
	max-width:350px;
	width:95% 
} 
```
In my old css style for xwide the width was set to 100% and to generate the same gap to the next right column the parent TD was created with `html_TD(F("padding-right: 8px")`. The padding-left was already set somewhere.

chromoxdor, 2022-08-31:
i made a new PR (the old one was broken) #4239 

I also realized, that in P036 the buttons at the bottom are in the wron column:
[image]

TD-er, 2022-09-01:
@uwekaditz Is it fixed with the proposed PR https://github.com/letscontrolit/ESPEasy/pull/4239  ?

uwekaditz, 2022-09-01:
Yes, it looks nice again with the proposed PR #4239 
[image]

TD-er, 2022-09-01:
Great!
Now it has been merged :)

closed 2022-09-01 by TD-er

chromoxdor, 2022-09-03:
> I also realized, that in P036 the buttons at the bottom are in the wrong column:

by the way.. this is nonsense since it is the right place for buttons! i was comparing it only quickly with the rules page und got confused since there is no second column so the buttons are at a different place. 🙄