letscontrolit/ESPEasy #4248: Plugin P007 doesn't work since release mega-20220328

jc-59, 2022-09-06:
All values Voltage, Power, Current, Pulses = 0

The plugin P007 is working will with release mega-20211224

TD-er, 2022-09-06:
You mention since 20220328.
That was the last build this plugin was still working?
Possible related commit: https://github.com/letscontrolit/ESPEasy/commit/92ec7a8752ee5f03f4144a93a8cda52321f10166#diff-a68c1771e9e67694d8f64a81c385403315ae80b953367cfbe8c06f83244bd45f

ghtester, 2022-09-06:
I can't confirm the issue. On one of my ESP test nodes, running the Custom firmware build dated 20220808 this plugin (PCF8591) still looks working fine (at least with 0x48 i2c address, quad output values, 4 single ended inputs).

jc-59, 2022-09-06:
Sorry I mad a mistake it's P077 plugin

TD-er, 2022-09-06:
Which exact build are you using?
What is the bin filename?

tonhuisman, 2022-09-06:
Hmm, might be this part of the code: https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P077_CSE7766.ino#L155-L159
The call to `CseReceive(event)` is only reached if the `BUILD_NO_DEBUG` flag is _not_ active, but I expect that to be active in the POW builds, at least it's active in the 1M builds.

TD-er, 2022-09-06:
Yep, that's very plausible.
It is for sure a bug

TD-er, 2022-09-06:
Made a quick patch based on @tonhuisman suggestions
If it is ready, can you test this GH Actions build? https://github.com/letscontrolit/ESPEasy/actions/runs/3002806279

tonhuisman, 2022-09-06:
Before fix:
[image]
No measurement.

After fix:
[image]
Yay, voltage and pulses (no load connected) 👍 

(Background alternating color changed from `even` to `odd` recently)

PR #4249 (2022-09-07): [CSE7766] Fix wrong debug check wrapper P077 (#4248)
Fixes: #4248
commit 141e8081 (2022-09-07): [CSE7766] Fix wrong debug check wrapper P077 (#4248)

Fixes: #4248
closed 2022-09-07 by TD-er