letscontrolit/ESPEasy #4255: Dark web color schema

fly74, 2022-09-08:
[image]

The partition table ESP32 in dark colors are nor readable.

Action Build: ESP_Easy_mega_20220907_neopixel_ESP32_4M316k

TD-er, 2022-09-08:
SVG can also use CSS, so we could replace some colors by the CSS class thingies.

tonhuisman, 2022-09-08:
I'll include this fix in the open PR that improves the CSS:
Light mode:
[image]

Dark/Auto mode:
[image]

@fly74 Don't close this issue manually please, I'll connect it to the PR so it will be closed automatically when merged.

PR #4243 (2022-09-12): [UI] Improvements for Devices page and css
Features:
- Show configured Controller name on Device configuration page and re-arrange the IDX value, that's optionally shown
- Updated `espeasy_default.css` to enable easy switching between Light and Dark mode, and by default follow the OS setting if this code is included (excluded from minimal builds)
- Removed other, unneeded, .css files from build.
- Add `tworow` and `even` attribute classes for `multirow` class for alternating background colors in tables per 2 or 1 row.

Partially based on suggestions by `chromoxdor` and `TD-er`

TODO:
- [ ] Tes [... cut]
closed 2022-09-12 by TD-er