letscontrolit/ESPEasy #4261: ESP_Easy_mega_20220910_normal_ESP8266_4M1M Action build crashes

fly74, 2022-09-11:
When ESPEasy P2P Networking is enabled and a sensor is tried to enable send to controller it crashes in ESP_Easy_mega_20220910_normal_ESP8266_4M1M. Rollback to actual release ist work as espected.

Maybe it is helpfull.

6334 : Info   : NTP  : NTP replied: delay 20 mSec Accuracy increased by 0.569 seconds
6337 : Info   : Time set to 1662884322.569 Time adjusted by 2635.86 msec. Wander: 0.000 msec/second Source: NTP
6340 : Info   : Local time: 2022-09-11 10:18:42
26693 : Info   :  Webserver 2 Arguments: 0: 'index' length: 1 1: 'page' length: 1
26723 : Info   : BMx280: SHOW_ERRORSTATE_OPT
28537 : Info   :  Webserver 25 Arguments (First 20): 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 2 3: 'TDN' length: 3 4: 'TDE' length: 2 5: 'i2c_addr' length: 3 6: 'p028_elev' length: 2 7: 'p028_tempoffset' length: 3 8: 'p028_err' length: 1 9: 'TDSD1' length: 2 10: 'TDID1' length: 1 11: 'TDSD2' length: 2 12: 'TDT' length: 3 13: 'TD
                                                                                                                   --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x40294584 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff76cb depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff2cf0 end: 3fff3310 offset: 0190
3fff2e80:  3fff2ed0 00000089 00000001 4022ea11
3fff2e90:  00000000 00000003 3fff2f5c 402952e8
3fff2ea0:  00000000 00000000 3fff2f5c 40286079
3fff2eb0:  0000001a 00000002 00000003 3fff2f5c
3fff2ec0:  3fff0c3c 3fff081c 3fff76b4 4022eb94
3fff2ed0:  010203ff 421c0000 00004b4c 00000000
3fff2ee0:  00000000 00000000 00000000 00000000
3fff2ef0:  6d655400 61726570 65727574 00000000
3fff2f00:  00000000 00000000 48000000 64696d75
3fff2f10:  00797469 00000000 00000000 00000000
3fff2f20:  00000000 65725000 72757373 00000065
3fff2f30:  00000000 00000000 00000000 00000000
3fff2f40:  00000000 00000000 00000000 00000000
3fff2f50:  00000000 00000000 3fff2f00 73657200
3fff2f60:  65727573 80000000 3fff2fa8 80286079
3fff2f70:  0000001c 3fff2f3f 80ff2fec 00000000
3fff2f80:  00000000 00000000 00000000 00000000
3fff2f90:  3fff3020 00000087 0000000a 4022ed59
3fff2fa0:  3fff3020 00000087 00000007 402477ac
3fff2fb0:  44898000 80000000 3fff2fec 40246bea
3fff2fc0:  3fff2fec 00000000 43960000 40286098
3fff2fd0:  00003300 3fff3080 01b46d58 00000000
3fff2fe0:  0000000d 3fff19e0 3fff3080 40286079
3fff2ff0:  00006e00 3fff0c3c 00000000 000003ec
3fff3000:  00000480 3fff0c3c 00000001 4027a48c
3fff3010:  ffffffff 3fff3090 00000000 40286098
3fff3020:  3fff3000 3fff3090 80000000 4024d000
3fff3030:  3fff3080 80000007 3ffe8300 7fffffff
3fff3040:  80ff3081 402dcc00 00006f7a 80000005
3fff3050:  00000000 3fff3080 80ff3080 00000000
3fff3060:  00000000 00000000 00000000 00000000
3fff3070:  00000000 00000000 03010000 4000ff00
3fff3080:  33534400 000e0000 80ff30b0 40286079
3fff3090:  00000001 3fff30d0 00000001 40286098
3fff30a0:  402e1023 3fff30d0 00000000 4027942c
3fff30b0:  00000000 402e1023 80ff30f0 ffffffff
3fff30c0:  ffffffff ffffffff 3fff30f0 40286079
3fff30d0:  3fff0a59 0000001a 3fff0c3c 0000000a
3fff30e0:  00000003 00000003 00000000 40279476
3fff30f0:  65646e00 00650078 80000000 3fff0c3c
3fff3100:  0000001c 00000001 00000000 40281f4f
3fff3110:  006c6500 00000855 80000855 40101383
3fff3120:  00006f48 ffffffff 3fff8504 00000001
3fff3130:  000000da 3fff121c 00000020 4010163b
3fff3140:  000000da 3fff31c8 402438d8 00000001
3fff3150:  000000da 3fff121c 3fff31f8 40286079
3fff3160:  3fff8ecc 3fff3180 3fff8c00 40286098
3fff3170:  00000008 3fff121c 3fff8c00 00000008
3fff3180:  00000001 3fff31d0 00000000 40286c1c
3fff3190:  3fff31f0 3fff31f0 3fff31d0 402790fc
3fff31a0:  00000003 00000001 3fff5e2c 402934fa
3fff31b0:  3fff31f0 3fff1238 3fff5e2c 401000d9
3fff31c0:  3fff5e2c 3fff1238 3fff5e2c 40279134
3fff31d0:  76656400 73656369 80006c00 80000000
3fff31e0:  3fff5e2c 3fff1238 3fff11f8 40244ee2
3fff31f0:  7665642f 73656369 88ff7e00 0000012f
3fff3200:  80006976 4bc6a7f0 40100c36 076c8b43
3fff3210:  00000000 00000001 3fff12a8 000000da
3fff3220:  00000003 0000000f 3fff06ec 4024273c
3fff3230:  00000000 3fff0700 3fff21dc 3fff335c
3fff3240:  00000001 3fff121c 3fff11f8 40245037
3fff3250:  3ffe8568 00000000 40100c36 3fff335c
3fff3260:  00006f33 3fff0700 3fff0704 3fff335c
3fff3270:  00006f33 3fff0700 3fff0704 40245135
3fff3280:  01b272ac 00000000 3fff1a88 40243200
3fff3290:  01b272db 00000000 402952e8 3fff335c
3fff32a0:  3fffdad0 00000000 3fff0c3c 3fff335c
3fff32b0:  3ffe8568 00000000 3fff0c3c 40245510
3fff32c0:  00000000 feefeffe feefeffe feefeffe
3fff32d0:  00000000 00000000 00000001 3fff335c
3fff32e0:  3fffdad0 00000000 3fff3318 40211088
3fff32f0:  feefeffe feefeffe feefeffe 40287acf
3fff3300:  feefeffe feefeffe 3ffe86d0 40100541
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
U206 : Info   : WIFI : Set WiFi to OFF
324 : Info   :

INIT : Booting version: ESP_Easy_mega_20220910_normal_ESP8266_4M1M, (GitHub Actions) HEAD_1006dbb (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
325 : Info   : INIT : Free RAM:28144
326 : Info   : INIT : Exception #2 Last Action before Reboot: Background Task Last systime: 1662884344 - Restart Reason: Exception
327 : Info   : FS   : Mounting...
351 : Info   : FS   : Mount successful, used 73041 bytes of 957314
397 : Info   : CRC  : SecuritySettings CRC   ...OK
398 : Error  : FS   : Error while reading/writing security.dat in 26832
400 : Info   : INIT : I2C
402 : Info   : INIT : SPI not enabled
413 : Info   : Time set to 1662884344.000
416 : Info   : Current Time Zone:  DST time start: 2022-03-27 02:00:00 offset: 120 min STD time start: 2022-10-30 03:00:00 offset: 60 min
419 : Info   : Local time: 2022-09-11 10:19:04
431 : Info   : WiFi : Start network scan all channels
443 : Info   : INIT : Free RAM:25656

TD-er, 2022-09-11:
It was an exception, so it is possible some memory is allocated which failed.
Do you have any idea about the amount of free memory when saving the settings?
How many p2p nodes do you have running in your network?
Which plugin did you try to save using the p2p?

fly74, 2022-09-11:
For the memory I have no idea. It is a network with 6 units. And the only plugin is the BME280 sensor.

TD-er, 2022-09-11:
Can you disable the p2p plugin for that task and add a sysinfo task to get some idea on the memory usage of the node.

I recently did merge some stuff related to p2p nodes from the ESPEasy-NOW mesh PR to make that PR actually smaller (something like a pre-merge)
So it is for sure possible that my pre-merge was incomplete.
But to get some idea of where to look, I need to know whether it is something like a missing check for a nullpointer, divide by zero, etc.

fly74, 2022-09-11:
With the disabled controller the sensor send to can be changed without problems. With enabled controller it crashed reproduceable. The http controller hasn't this issue.

Sysinfo:

load:9.74
freeheap:15864.00
web:0.17
freestack:3264.00

TD-er, 2022-09-11:
OK, wil look into it later this evening.

TD-er, 2022-09-13:
I am looking into this right now.
What I did experience is that the last line of the sysvars could somehow cause a crash:
```
Unit to IP: %c_u2ip%(%unit%, 2)
```
I will make sure that will no longer crash.

Still looking into the other crashes you mentioned.

fly74, 2022-09-13:
With a disabled controller you can activate send to in the plugin and after that you can activate the controller as a workaround.

TD-er, 2022-09-20:
It should be fixed by this PR: https://github.com/letscontrolit/ESPEasy/pull/4272
@fly74 @chromoxdor can you test this please?

fly74, 2022-09-20:
@TD-er I would, but the build is still running :)

TD-er, 2022-09-20:
which one do you need?

chromoxdor, 2022-09-20:
looks good so far 🙂

fly74, 2022-09-20:
> which one do you need?

8266 4M1M normal

chromoxdor, 2022-09-20:
[ESP_Easy_mega_20220920_normal_ESP8266_4M1M.bin.gz](https://github.com/letscontrolit/ESPEasy/files/9611142/ESP_Easy_mega_20220920_normal_ESP8266_4M1M.bin.gz)

fly74, 2022-09-20:
Build was not ok??

[build (esp8266, normal_ESP8266_4M1M)](https://github.com/letscontrolit/ESPEasy/actions/runs/3093162691/jobs/5005299181)
The job was canceled because "esp8266_energy_ESP8266_4M1M" failed.

fly74, 2022-09-20:
> [ESP_Easy_mega_20220920_normal_ESP8266_4M1M.bin.gz](https://github.com/letscontrolit/ESPEasy/files/9611142/ESP_Easy_mega_20220920_normal_ESP8266_4M1M.bin.gz)

works for me too!

TD-er, 2022-09-20:
> The job was canceled because "esp8266_energy_ESP8266_4M1M" failed.

Probably due to bin size.... as always :(

closed 2022-09-20 by TD-er

chromoxdor, 2022-09-20:
> Probably due to bin size

140byte off... that was at least very close 🙂

TD-er, 2022-09-20:
```
Flash: [==========]  100.0% (used 1044496 bytes from 1044464 bytes)
```

chromoxdor, 2022-09-20:
you can do it!!!
Go @TD-er  go.....🥳

TD-er, 2022-09-20:
> you can do it!!! Go @TD-er go.....🥳

I still need to fix the other bug I introduced while reducing build size.
The issue you reported...

PR #4272 (2022-09-20): [ESPEasy p2p] Fix crashes (#4261)
Fixes: #4261
commit 77e05143 (2022-09-20): [ESPEasy p2p] Fix crashes (#4261)

Fixes: #4261