letscontrolit/ESPEasy #4264: update pubsubclient lib

uzi18, 2022-09-13:
@TD-er @tonhuisman  When You look at https://github.com/knolleary/pubsubclient/pulls there are tons of stuff already implemented ... into this lib.

TD-er, 2022-09-13:
Any suggestion on what's being fixed what we need?
The problem is, I did change quite a few things, so updating isn't that easy.

TD-er, 2022-09-13:
This seems to be the first (useful) merged one which isn't merged into ESPEasy: https://github.com/knolleary/pubsubclient/pull/571

TD-er, 2022-09-13:
And this one is also useful: https://github.com/knolleary/pubsubclient/pull/568
The rest is either already implemented (due to long waiting PRs I cherry picked) or not really something we're missing.

uzi18, 2022-09-13:
I think it should be converted i to module as external fork of original lib. So rest of the world could also make use of it.

TD-er, 2022-09-13:
For quite some time, Tasmota did use the literal code of the ESPEasy version of the pubsubclient code.