letscontrolit/ESPEasy #4265: I2C Error

fly74, 2022-09-17:
I'm facing an I²C problem. I have 2 devices at the bus. A BME280 and a BH1750. The BH1750 alone is detected and works. With the BME280 (and he alone too) I get in I²C Scan and no device is detected:

<html>
<body>

SDA low at address 0x01 Reset bus attempted
--
SDA low at address 0x02 Reset bus attempted
SDA low at address 0x03 Reset bus attempted
SDA low at address 0x04 Reset bus attempted
SDA low at address 0x05 Reset bus attempted
SDA low at address 0x06 Reset bus attempted
SDA low at address 0x07 Reset bus attempted
SDA low at address 0x08 Reset bus attempted

</body>
</html>

Version is ESP_Easy_mega_20220809_normal_ESP8266_4M1M and I tried 2 different BME280.
Has anybody an idea?

tonhuisman, 2022-09-17:
Are you sure the BME280 is connected correctly? As it is not recognized by the I2C Scanner, and looking at the log you show, either there is a short-circuit between SDA and GND (soldering checked?) or the BME280 board is broken/kaput.
These BME boards usually come with 'different' quality from a lot of Chinese suppliers, so it is entirely possible you received a set of defective boards/chips. If you have a stand-alone I2C device validator, you could try that to see if the board is recognized there.
You may need to use the warranty procedure to have them replaced.

fly74, 2022-09-17:
I tried 2 of the delivery and it looks like a failure for all. It is soldered correctly. :(

TD-er, 2022-09-17:
Is it possible you have bought the 5V versions of the sensor?
N.B. SDA/SCL are probably pulled up to 5V then, so you should not combine those with 3V3 I2C devices.

fly74, 2022-09-17:
I sent it back. It was the default config D1 SCL and D2 SCA. The light sensor works perfectly with the configuration. Only the BMEs were causing the problem. It was a pack of cheap ones. 3 for 11€. Normally 1 costs 16 €. So it looks like the quality was low.

closed 2022-09-17 by fly74