letscontrolit/ESPEasy #4270: constant reboots with latest mega build

chromoxdor, 2022-09-20:
Just build myself a ESP_Easy_mega_20220920_normal_ESP8266_4M1M.bin and updated an esp.
now i get constant reboots caused by a task with a bme280...
[image]

Did recently something in the plugin or i2c related was changed?

Revert to an older build (only tried july) or deactivating the task solves the issue

chromoxdor, 2022-09-20:
Hmmm... disabling the p2p controller seems to solve this issue as well... therefore my next best guess is, that this is an p2p related problem.

fly74, 2022-09-20:
@TD-er looks like in https://github.com/letscontrolit/ESPEasy/issues/4261

chromoxdor, 2022-09-20:
Yes, it seems to be the same issue...

TD-er, 2022-09-20:
Yep that's not yet fixed...
So this is a duplicate of #4261 and thus I will close this one.

closed 2022-09-20 by TD-er