letscontrolit/ESPEasy #4276: mega-20220809 : src/Helpers/Hardware.h: No such file or directory

PandCall, 2022-09-21:
Hello,
I wanted to test the source version to compile to change 2 or 3 things.

I started from scratch
VM Windows 10 21H2 completely updated
Arduino IDE 1.8.19
ESP12E ( NodeMCU V1.0 board)
Download Release mega-20220809 src
https://github.com/letscontrolit/ESPEasy/archive/refs/tags/mega-20220809.zip
Decompression ZIP 
Copy of libraries and completely updated
Change scr name to ESPEasy
open ESPEasy.ino
And compile test to see if the base version is ok.
And I have this message.

In file included from C:\Users\User\Downloads\ESPEasy-mega-20220809\ESPEasy\_P002_ADC.ino:6:
C:\Users\User\Downloads\ESPEasy-mega-20220809\ESPEasy\src\PluginStructs\P002_data_struct.h:8:11: fatal error: src/Helpers/Hardware.h: No such file or directory
    8 | # include "src/Helpers/Hardware.h"
      |           ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Erreur de compilation pour la carte NodeMCU 1.0 (ESP-12E Module)

Is there something I missed in the use?
I even tested with the sources that are in 
https://github.com/letscontrolit/ESPEasy/releases/download/mega-20220809/ESPEasy_ESP82xx_mega-20220809.zip

If I flash the BIN ESP_Easy_mega_20220809_normal_ESP8266_4M1M.bin
everything works fine, so we can rule out a hardware problem.

Thanks for the help.
Regards

tonhuisman, 2022-09-21:
This is the MSDT (Manufacturer Suggested Development Tooling) for ESPEasy: [PlatformIO & Starter guide for development on ESPEasy](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html)

PandCall, 2022-09-21:
To make it simple, the recommended development platform is PlatformIO and not Arduino IDE?
And so no explanation to this error ? 
Regards

tonhuisman, 2022-09-21:
... and why I steered toward the Developer guide. 😃  Trying to do development with Arduino IDE is IMHO making it harder for yourself than needed.

PandCall, 2022-09-21:
Yes, I thought I had seen that this problem had been fixed, but apparently it's happening again.
In the meantime, I'm doing some tests with PlatformIO,
I'll come back here to give you an update :)
But I will look at all your leads.
Best regards

TD-er, 2022-09-21:
It has been fixed, but there has not yet been a new official build since it was fixed.
So if you download the source code from the last build, you still get the old source code.
But if you download the ZIP as I showed in my screenshot, you have the latest code.

TD-er, 2022-09-21:
Please download the ZIP from where I showed you and not using the zip from the last build.

For example the "boards" folder is not included in that ZIP (reported by another user) and it should be there when you download the ZIP from where I showed you in the screenshot.

TD-er, 2022-09-21:
You can also install the (Chrome) plugin "GitPod" and then you even can build it all in your browser without ever downloading and installing anything :)

PandCall, 2022-09-21:
Thanks for the information, I would test it out of curiosity.
But no problem for me, the Arduino IDE and PlatformIO environments, I use them for other products and environments.
But before installing on my Dev machine, I install in a clean VM to see the behaviors.
Thanks for the product, for the fast support and the good advices.
Regards

closed 2022-09-21 by TD-er

uzi18, 2022-09-21:
@PandCall gitpod is nice and fast idea if you don't want to mess with your computer, just try it. 
But remember they limited work time to 50hours per month in free plan.