letscontrolit/ESPEasy #4277: Sensor information and vlan

anno006, 2022-09-22:
situation: 

esp on vlan 50 for example 
And domoticz on vlan 1 
When trying to send sensor information, for example DHT22 or BME280 there where no errors in both the logs untill firmware 20220809.

But the sensor in domoticz will not update. So the information is lost. On the onther hand alle http commands send from domoticz will all work. 

Firmware 20220809 gives the next error: Error : HTTP : C001 x.x.1.x get... failed http code: 401 (it will repeat 11 times with every sensor update)

TD-er, 2022-09-22:
Can you try one of the latest builds, as I have been fixing some HTTP related issues since the last build.

For example this one (the last completed build on the Actions tab): https://github.com/letscontrolit/ESPEasy/actions/runs/3108390225

anno006, 2022-09-23:
I did manage to find the build and install it. After some time waiting for connect/disconnect actions (a lot, other build also, other problem) the esp was running and, nice dark color, a lot less white. but it didn't fix this problem. Still C001 erros and no update to the sensor.

TD-er, 2022-09-23:
OK, will also test here using Domoticz via HTTP.

In the mean time, can you give slightly more info on the VLAN/routing config in your network?
Are Domoticz and the ESPs in the same VLAN/subnet?
If not, how do you route between the networks? L3 switch? Router?
What is routed and what isn't?

Static IP or dynamic IP?

anno006, 2022-09-23:
I will send you a mail about what you ask.

TD-er, 2022-09-23:
Via emails, it appears to be a routing issue between the VLAN/subnets

closed 2022-09-23 by TD-er

anno006, 2022-09-23:
For now i did get 2 extra esp on to the other subnet. But 2 others i did run into a lot of errors, other errors this time. Auth expired for example. But it is possible a esp will get working out of the blue. It is not a routing issue between the vlan/subnets.