letscontrolit/ESPEasy #4279: Webinterface looks minimal

fly74, 2022-09-27:
What could be the reason for look like this?:

[image]

tonhuisman, 2022-09-27:
The browser you are using doesn't seem to have access to the internet, to download the css files from the CDN. (Should only be needed 'once', as it has a very long time to live in the browser cache).
Alternatively you can upload these files to the filesystem of your ESP, instructions are here in [Read The Docs](https://espeasy.readthedocs.io/en/latest/Reference/ExternalHostedStaticFiles.html)

fly74, 2022-09-27:
[image]

Maybe from an action build releated issue. Some internet ressources are loaded ?!?

fly74, 2022-09-27:
Only the ESP8266 Version has the look - ESP_Easy_mega_20220910_neopixel_ESP8266_4M1M, ESP_Easy_mega_20220910_neopixel_ESP32_4M316k on ESP32 looks fine. The devices are not restricted for internet access.

fly74, 2022-09-27:
works with ESP_Easy_mega_20220921_neopixel_ESP8266_4M1M

closed 2022-09-27 by fly74

tonhuisman, 2025-08-24:
Hmm, that's rather silly as many apps rely on their js stuff provided via that CDN 🤔