letscontrolit/ESPEasy #4287: Feature Request. I2C rotary encoder support.

Budman1758, 2022-10-05:
I stumbled into these at Adafruit the other day and methinks it would be pretty awesome if these could have support. Would make for much simpler setups. Also ability to support many of them over only 2 GPIO.

https://www.adafruit.com/product/4991

tonhuisman, 2022-10-08:
I'm planning to add support for this type of encoders, but I've yet been unable to source that Adafruit device locally. I'll keep searching. (trying to avoid high transport cost from US to EU, and haven't yet found a clone at Aliexpress)

Did find, and bought, these alternatives:
- [M5Stack Encoder](https://docs.m5stack.com/en/unit/encoder)
- [DFRobot Rotary encoder module](https://wiki.dfrobot.com/SKU_SEN0502_Rotary_Encoder_Module_I2C)

I intend to not use their specific libraries, but add direct support for these devices, to avoid the overhead/code duplication they seem to introduce by the I2C support library, like DFRobot has. I'll re-use the code from the available libraries, of course 😉.

Budman1758, 2022-10-09:
I would be quite willing to send you one of mine. Due to the fact that I can't program worth a shit it would be cheap for me. :)
What part of Europe do you live in? Send me an email with your address and it will be on its way.

TD-er, 2022-10-09:
Both Ton and I live in the Netherlands, but I guess Ton is lately way more productive regarding coding Plugins.

Budman1758, 2022-10-10:
Ordered 2 of those encoders and they are shipping to Tons place direct. Hopefully you should see them within a couple weeks at most.

Budman1758, 2022-11-19:
Left a comment in the wrong place.  lol

https://github.com/letscontrolit/ESPEasy/pull/4356#issuecomment-1320765119

PR #4356 (2023-01-01): [P143] Add plugin I2C Rotary encoders
Resolves: #4287 

Add plugin `Switch input - I2C Rotary encoders`
Features:
- Support for [Adafruit I2C Rotary encoder](https://www.adafruit.com/product/4991) (up to 8).
- Support for [M5Stack I2C Rotary encoder](https://docs.m5stack.com/en/unit/encoder) (max. 1).
- Support for [DFRobot I2C Rotary encoder](https://wiki.dfrobot.com/SKU_SEN0502_Rotary_Encoder_Module_I2C) (up to 4).
- Support for button (all above encoders have a pushbutton included).
- Support for button-longpress, resulting in state=10/11, similar to Switch plugin. (Not supported by DFRobot [... cut]
closed 2023-01-01 by TD-er