letscontrolit/ESPEasy #4292: Feature Request.Add DS28EA00 chip chain feature support

3dalex07, 2022-10-10:
It is posible to add chain feature support for the Maxim DS28EA00 temperature sensor
maybe that [library](https://github.com/mortonkopf/OneWire_DS28EA00)  may help to do that 
Thanks

TD-er, 2022-10-10:
I have not yet seen this sensor myself.
Have to look into it to see if it can be used with other 1-wire sensors too.
For future reference, [sensor info](https://www.maximintegrated.com/en/products/sensors/DS28EA00.html/product-details/tabs-3)

3dalex07, 2022-10-10:
I have chain of that sensors 20 pck. It work with standart plugin P004  but sensors adresses reads randomly.

TD-er, 2022-10-10:
Do you have the pull-up resistor present?

3dalex07, 2022-10-10:
yes 2,2koM

TD-er, 2022-10-10:
Wow, those are rather expensive.
Mouser asks 12 euro per piece and only starting at 10+ it drops to 7 euro.

3dalex07, 2022-10-11:
more interesting price https://www.digikey.lv/en/products/detail/analog-devices-inc-maxim-integrated/DS28EA00U-T-R/1769510

3dalex07, 2022-10-11:
I may connect my sensor chain to esp and give you access to it

TD-er, 2022-10-11:
> I may connect my sensor chain to esp and give you access to it

I had to use my logic analyzer extensively to make the current Dallas code work.
So I don't think it will be very useful unless I have some of my own.

Anyway, I first have to check the datasheet to see if there might be some very obvious issue (e.g. more bits required to scan for an address or something similar)

3dalex07, 2022-10-19:
> I have chain of that sensors 20 pck. It work with standart plugin P004 but sensors adresses reads randomly.

test it again / only adresses reads / temperature for all sensor 0 pullup 2.2k\

test only one connected sensor, adress reads but temperature show 0

3dalex07, 2022-10-19:
src/src/Helpers/Dallas1WireHelper.cpp 
add line 56 
`    case 0x42: return F("DS28EA00"); `
change line 392 to
` if ((ROM[0] == 0x28) || (ROM[0] == 0x3b) || (ROM[0] == 0x22) || (ROM[0] == 0x42)) // DS18B20 or DS1825 or DS1822 or DS28EA00`
now it reading temperature

TD-er, 2022-10-19:
Ah great!
So that's a very simple fix :)

TD-er, 2022-10-20:
Created a PR with essentially your suggested code changes in it: https://github.com/letscontrolit/ESPEasy/pull/4310
You could try the [test build](https://github.com/letscontrolit/ESPEasy/actions/runs/3288375780) when ready, but I guess it can be merged as soon as that build has been completed without errors.

PR #4310 (2022-10-23): [Dallas] Add DS28EA00 chip (#4292)
For future reference, see the [ID list](https://github.com/owfs/owfs-doc/wiki/1Wire-Device-List) 

Thanks @3dalex07 for debugging and handing the copy/paste fix.

Fixes: #4292
commit 46d848b5 (2022-10-23): [Dallas] Add DS28EA00 chip (#4292)

For future reference, see the [ID list](https://github.com/owfs/owfs-doc/wiki/1Wire-Device-List)
Fixes: #4292

Thanks @3dalex07 for debugging and handing the copy/paste fix.
closed 2022-10-23 by TD-er