letscontrolit/ESPEasy #4295: OLED Framed timeout not working correctly

Budman1758, 2022-10-13:
On some of the latest builds from the github "actions" area the OLED Framed plugin timeout setting seems broken. Does not seem to matter what the timeout setting is set to the display goes off at somewhere between 30 and 60 seconds. Have not put a stopwatch on it but it seems a bit random as to exactly how long. Have tried 300, 600, 900 and other seconds settings and it seems to make no difference. However, if set to zero for no timeout that seems to work fine.

This is true on 8266 and ESP32 builds. Have not tested on the released binaries. Will do so and see.

tonhuisman, 2022-10-14:
The `airvalve` and `power` tasks _are_ enabled, right? As disabled tasks will result in empty results.

Budman1758, 2022-10-15:
> The `airvalve` and `power` tasks _are_ enabled, right? As disabled tasks will result in empty results.

Indeed they are enabled.

tonhuisman, 2022-10-15:
Which plugins are used for the `airvalve` and `power` tasks?

tonhuisman, 2022-10-15:
Fixed the formatting issue, 1 down, 1 to tackle 😃

tonhuisman, 2022-10-15:
Can you try a with build from PR #3997 as that might be merged in the near future (development is completed), and my ESP8266 test-unit running that PR build is reliably turning both connected displays off after the set interval (2 minutes and 3 minutes, just for 'fun' 😉).

Budman1758, 2022-10-15:
> Which plugins are used for the `airvalve` and `power` tasks?

[image]

Budman1758, 2022-10-15:
> Can you try a with build from PR #3997 as that might be merged in the near future (development is completed),

Using builds from this [link](https://github.com/letscontrolit/ESPEasy/actions/runs/3252784397). Think thats the one you mean. Have display timeout set to 600 seconds and its going off inside of 2 minutes. Transformations not working either.....

tonhuisman, 2022-10-15:
> Have display timeout set to 600 seconds and its going off inside of 2 minutes

Hmm, that's surprising. I'll test with 600 seconds here

> Transformations not working either.....

That fix _has_ been merged into mega, but _not yet_ in that PR branch. For testing the formatting fix you can pick the most recent merge build, [here](https://github.com/letscontrolit/ESPEasy/actions/runs/3255460785)

tonhuisman, 2022-10-15:
Just had a 💡 and looked at the variable that holds the time-out, but that is an `uint8_t` meaning the largest value it can hold is 255... That's a clear explanation why it won't get at 600 seconds 🤣 
Will ask for a fix in that PR, so it can be included when that is merged.

Budman1758, 2022-10-15:
> Just had a 💡 and looked at the variable that holds the time-out, but that is an `uint8_t` meaning the largest value it can hold is 255... That's a clear explanation why it won't get at 600 seconds 🤣 

Well, guess that makes sense. 🤣 

Installed a bin from that link at it appears that transformations are functional now. 

BTW. I REALLY like the dark mode.  WAAAAAAAAYYYYYY better!!!  That needs to go in every build.

TD-er, 2022-10-15:
Maybe also have a look at the rules editor window ;)

Budman1758, 2022-10-15:
> Maybe also have a look at the rules editor window ;)

I did!!  Love it!!

TD-er, 2022-10-15:
> > Maybe also have a look at the rules editor window ;)
> 
> I did!! Love it!!

You have to thank @chromoxdor for it :)

Budman1758, 2022-10-16:
@chromoxdor   Great job on the Rules editor. Pretty awesome stuff!!!

Budman1758, 2022-10-16:
> Just had a 💡 and looked at the variable that holds the time-out, but that is an uint8_t meaning the largest value it can hold is 255... That's a clear explanation why it won't get at 600 seconds 🤣

BTW. I set the timeout on my unit at 255 and the display stays on for 4 min 15 secs.

TD-er, 2022-10-27:
Fixed by https://github.com/letscontrolit/ESPEasy/pull/3997

closed 2022-10-27 by TD-er