letscontrolit/ESPEasy #4303: [CSS] textarea is way too large

TD-er, 2022-10-16:
@chromoxdor 

The textarea fields are now set to `60vh`
This may be fine for the logviewer, but not for all other fields.
For example the command output field on the Tools page is now way too large.

It seems to come from this section in the `espeasy_default.css`:

```css
.logviewer,
textarea {
	font-family: 'Lucida Console', Monaco, monospace;
	height: 60vh;
	width: -webkit-fill-available;
	width: -moz-available
}
```

When this height field is left out, the text area boxes look OK in most cases.

The `addTextArea` function does add a `rows` and `cols` attribute.
Maybe we can compute the desired height based on these?

chromoxdor, 2022-10-16:
Can you send a screenshot... i am not sure what you mean.

TD-er, 2022-10-16:
These are for the (upcoming) MQTT TLS controller:
[image]

N.B. the Certficate info field was some experiment of mine to see if setting attributes would help here.

TD-er, 2022-10-16:
Output after a command on the tools page:

[image]

chromoxdor, 2022-10-16:
Ahh ok .... i´ll look into it

TD-er, 2022-10-16:
N.B. the logviewer is fine with this height setting.

chromoxdor, 2022-10-16:
We could assign the height only to the rules and logviewer window and the rest would than be "fit-content"

chromoxdor, 2022-10-16:
> When this height field is left out, the text area boxes look OK in most cases.

would it look better with "fit-content"?

chromoxdor, 2022-10-16:
this should fix everything...
`height: fit-content;
  max-height: 60vh;`
i can make a commit tomorrow

TD-er, 2022-10-16:
For the log viewer this looks very weird.

tonhuisman, 2022-10-16:
```css
.logviewer,
textarea {
	font-family: 'Lucida Console', Monaco, monospace;
	height: fit-content;
        max-height: 60vh;
	width: -webkit-fill-available;
	width: -moz-available
}
.logviewer {
	height: 60vh !important;
}
```

AFAICS, this should fix it. (untested)

Grovkillen, 2022-10-16:
You could look into my CSS framework... I'm going to get back into this
rabbit hole now that I'm settled in my new life.

Here is my GUI project:
https://github.com/letscontrolit/GUIEasy/tree/master/src

The framework for the graphics is here:
https://github.com/letscontrolit/GUIEasy/blob/master/src/gui_easy.css

chromoxdor, 2022-10-16:
> For the log viewer this looks very weird.

ohh... sorry
this should work fine with everything:

```css
textarea {
	font-family: 'Lucida Console', Monaco, monospace;
	height: fit-content;
        max-height: 60vh;  
	width: -webkit-fill-available; 
	width: -moz-available
}

.logviewer {
	height: 60vh
}
```

chromoxdor, 2022-10-17:
I made a PR here: https://github.com/letscontrolit/ESPEasy/pull/4304

and some explanation i put in the commit comment which is not placed very obvious so i copy it in here too:

> textarea height is given by "cols" in the tag and it ignores things like "fit-content" so i removed it and this also makes "max-height" unnecessary....
Auto resize the textarea to fit content is possible but adds more extra code:
Example: https://codepen.io/phpcodertech/pen/oNXVzWW

TD-er, 2022-10-17:
I think the option with the span is the better one as it does also resize on long lines.
[image]

chromoxdor, 2022-10-17:
But the question is, do we need auto resize for anything? 
I don’t see a real use for it right now....

TD-er, 2022-10-17:
See the example for the certificates on the MQTT TLS I'm working on.
I do compute the nr. of lines and then set the nr of rows accordingly in the HTML
But if the window is too small, it may still add a scroll slider.
Not sure if that's really a problem though.

chromoxdor, 2022-10-17:
> See the example for the certificates on the MQTT TLS I'm working on. I do compute the nr. of lines and then set the nr of rows accordingly in the HTML But if the window is too small, it may still add a scroll slider. Not sure if that's really a problem though.

I actually wouldn’t add it because either way.. you have to scroll down to see more and without auto resize it doesn’t move the rest of the page down

TD-er, 2022-10-17:
OK, you're the expert here... I don't speak CSS/JS, I only copy/paste it from Stackoverflow :)

chromoxdor, 2022-10-17:
I am very honored that you call me an expert, i wish i could agree... 😁

TD-er, 2022-10-17:
In the land of the blind, one-eye is king

chromoxdor, 2022-10-19:
Can this be closed?

TD-er, 2022-10-19:
Yep

closed 2022-10-19 by TD-er