letscontrolit/ESPEasy #4311: [GitHub Actions] Deprecating save-state and set-output commands

TD-er, 2022-10-20:
GitHub Actions now shows this deprecation warning:

> The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

@mcspr Have you already looked into this?

Since I am unfamiliar with the original syntax, I wonder whether this would be the correct change:

Build.yml:
Current:
```
echo "::set-output name=matrix::[${JSONI}]"
```

Change into:
```
echo "matrix=[${JSONI}]" >> $GITHUB_OUTPUT
```

mcspr, 2022-10-20:
ref. 'official implementation', which is also a source of the warning when some 3rd-party workflow has pinned an older version
https://github.com/actions/toolkit/blob/b36e70495fbee083eb20f600eafa9091d832577d/packages/core/src/core.ts#L193
https://github.com/actions/toolkit/blob/b36e70495fbee083eb20f600eafa9091d832577d/packages/core/src/file-command.ts#L11

Python could just open that file directly - `open(os.environ["GITHUB_OUTPUT"], "wa")` file or fallback to print
(i.e. github runner is simply reading our stdout, it does not execute any shell commands but instead uses a special command syntax)

Blogpost also has weird wording that seem to suggest we should use OS envrion instead of output, URL goes to the very first entry that uses `GITHUB_ENVIRONMENT`; we need to do some scrolling to find the `GITHUB_OUTPUT` entry
https://docs.github.com/en/actions/using-workflows/workflow-commands-for-github-actions#setting-an-output-parameter

TD-er, 2022-10-20:
Hmm, to be honest this change feels more of a "step back" regarding security and safety checks.
It seems like there used to be some kind of parser module which did expect only texts with a rather specific format.
And now they just let scripts set environment variables and run bash like scripts.

I will have another look at this tomorrow.

Thanks anyway for the links.

closed 2023-01-13 by TD-er