letscontrolit/ESPEasy #4312: ESP32-S2 parameters not correctly utilized, wrong number of GPIO

Jollerprutt, 2022-10-22:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

Yes I self compile.

### Summarize of the problem
Number of GPIO pins for ESP-S2 wrongly evaluated during compilation.
Not sure what component name I should give, but I guess this belongs to some kind of core.

### Expected behavior
in [/src/src/CustomBuild/ESPEasyLimits.h](https://github.com/letscontrolit/ESPEasy/blob/317f69496b096fbefee8e2c34e46374b321fe61f/src/src/CustomBuild/ESPEasyLimits.h#L41) the MAX_GPIO is expected to be evaluated to 46 for ESP32-S2.
https://github.com/letscontrolit/ESPEasy/blob/317f69496b096fbefee8e2c34e46374b321fe61f/src/src/CustomBuild/ESPEasyLimits.h#L41

### Actual behavior
MAX_GPIO is evaluated as 39.

### Steps to reproduce
1. Set a S2 environment for compilation, I used "collection_A_ESP32s2_4M316k"
2. Compile and upload
3. Check highest GPIO number webui

To verify the problem is with this code I tried commenting out the evaluation and manually defining MAX_GPIO as 46.
This works and GPIO 40 and higher, show up in the web ui. The pins are also usable and I used 40 & 41 for the QT PYs i2c successfully.

### System configuration
Build:⋄
20221022  - Mega32-s2
System Libraries:⋄
ESP32 SDK v4.4.3-149-g10ab11e815
Git Build:⋄
pygit2_not_installed
Plugin Count:⋄
80 [Normal][Collection_A ESP32]
Build Time:⋄
Oct 22 2022 18:07:47
Binary Filename:⋄
ESP_Easy_mega_20221022_collection_A_ESP32s2_4M316k
Build Platform:⋄
Windows-10-10.0.19042-SP0
Git HEAD:⋄
pygit2_not_installed

Hardware:
Run on [Adafruit QT PY ESP32-S2](https://learn.adafruit.com/adafruit-qt-py-esp32-s2/pinouts)

ESP Easy version: 
ESP Easy mega branch clean pull today (22 October 2022)
Self compiled.

TD-er, 2022-10-22:
Hmm that's very strange as nobody has complained about it and the related files seem to be changed about 10 months ago.

I have to admit, I have not tested the ESP32-S2 boards for a while since I moved back in Feb/March, as I seem to have misplaced my ESP32-S2 boards since.

Anyway, I think there should be some extra define (or define check).
In `boards/esp32s2.json` there is this line for `extra_flags`, which could be extended with something like this:
`  "extra_flags": "-DBOARD_HAS_PSRAM -DESP32_4M -DESP32S2 -DCONFIG_IDF_TARGET_ESP32S2=1",`

Or we should add some extra checks in the `define_plugin_checks.h` checking for `#ifdef ESP32S2`

But since this `CONFIG_IDF_TARGET_ESP32S2` is also checked often in the arduino-esp32 SDK code, I think it is best to define it in the board JSON file.

Jollerprutt, 2022-10-22:
Tried editing the json as suggested, but it still evaluates to 39 GPIOs.
Also tried a manual define of CONFIG_IDF_TARGET_ESP32S2 in include/ESPEasy_config.h, but it did not affect anything.
I'm thinking maybe CONFIG_IDF_TARGET_ESP32 is set somewhere and therefore CONFIG_IDF_TARGET_ESP32S2is never considered.

TD-er, 2022-10-22:
@tonhuisman had it already working (reproducible) , but not 100% sure what else he did.
Just make sure to perform a clean build first, so there is nothing left from a previous build.

Jollerprutt, 2022-10-22:
e4e883f8f58d361286450b844b180cbc84b1bf0f solves it :)
Tested on QT PY ESP32-S2.

PR #4309 (2022-10-23): [Various] Check Arduino OTA setting, increase max. plugin count
Features/improvements:
- Initialize Arduino OTA only when enabled in Advanced settings page
- Increase max. number of plugins to allow all in MAX builds (140)
- Logging improvements
- Enable I2C Device scan for MAX builds
- Updated I2C scanner with some missing devices
- [P039] Allow GPIO-0 (D3 on NodeMCU/Wemos boards) to be used for CS (Bugfix)
- [Boards] Additional IDF flag for ESP32s2 and, currently unsupported ESP32c3, boards to get correct hardware configuration
- [SPI] 3rd SPI GPIO pin should be output only (and not Generic  [... cut]
closed 2022-10-23 by TD-er