letscontrolit/ESPEasy #4314: [P014] shows wrong temperature in recent builds

chromoxdor, 2022-10-24:
As described here: https://www.letscontrolit.com/forum/viewtopic.php?t=9312
the HTU21D shows the wrong temperature. In fact it only shows 128,86.
Going back to the latest release (20220809)solves this issue.

TD-er, 2022-10-24:
@fmuntean can you have a look at this? You were the last to extend this plugin.

chromoxdor, 2022-10-24:
My test setup already had a newer build. I hooked up an htu21d that I had laying around just to see if I can replicate the thread starter issue. Unfortunately, I won't be at home this week, so I can't test further.
Of course, I always have a few espeasy devices nearby, but no sensor. 🙂

fmuntean, 2022-10-24:
The PCONFIG has changed in between the versions. Thus my recommendation to remove then add the plugin back.
The only other thing I could think of is the heater on that device somehow starts as I guess the 128.86 is in Celsius.

Unfortunately I do not have the HTU21D device to test with myself but the fact that moving back to the old firmware (most likely not deleting/adding the plugin in between) works tells me that might be the PCONFIG issue here.

chromoxdor, 2022-10-24:
> The PCONFIG has changed in between the versions. Thus my recommendation to remove then add the plugin back.

Like i said.. i started with a fresh recent build to test this issue. So it had nothing to do with updating in my case..
I´ll invite the tread-starter to this conversation. Maybe he can try your recommendations.....

fmuntean, 2022-10-24:
As I do not have the sensor myself I will need some logs please.
looking for any logs starting with "SI70..."

fmuntean, 2022-10-28:
pull request started: https://github.com/letscontrolit/ESPEasy/pull/4321
We will use the chip_id=50 to detect the HTU21D which does not have all the commands as SI70xx

TD-er, 2022-10-28:
When the GitHub Actions build is ready, could someone try it?
https://github.com/letscontrolit/ESPEasy/actions/runs/3345553088

chromoxdor, 2022-10-28:
I will… but not before monday

chemmex, 2022-10-28:
To be honest, no luck with ESP8266

`136769: SI70xx : Not available at address: 0`

each 10 seconds in the log.

The HTU21 is detected by I2C scan

[image]

fmuntean, 2022-10-28:
It says address zero.
Have you configured the plugging ?

chemmex, 2022-10-28:
Just re-init the plugin: 
```
632587: EVENT: TaskInit#HTU=1,0
632598: SI7013 : sn=5fe22-32114854
632598: P014: chip_id=50
632659: static_file: favicon.ico
632875: P014: Temperature: 23.61 Humidity: 45.50
632879: EVENT: HTU#All=23.61
```

chemmex, 2022-10-28:
But, as you may see, only temperature in an All event, and also only temperature on Devices tab

fmuntean, 2022-10-28:
Looks that it works now.
What is the issue you still face?

chemmex, 2022-10-28:
upd: I got the readings only once upon re-init. Cannot get them with TaskRun or by schedule

fmuntean, 2022-10-28:
can you provide the screenshot for the plugin settings ? (when you click the "edit" button)

chemmex, 2022-10-28:
Aha, new for me, this task set Single output by default, so it was only Temperature, indeed. Changing to Dual or Temp/Hum corrects the whole thing

I only left a bit confused by this plenty of choices not quite relevant to this particular sensor
[image]

fmuntean, 2022-10-28:
This sensor is piggy backing on the real SI70xx sensors that have more options.

chemmex, 2022-10-28:
I'm not sure how for long I didn't configure this sensor manually. I have a preset config.dat from older versions which I just uploaded to the devices

chemmex, 2022-10-28:
Hmm, seems that at this point there is no more compatibility with older configs. It now defaults to single output despite that I had two in my config file based on 09.2022 custom build

chemmex, 2022-10-28:
And last but not least, when I upload the most recent config, it doesn't init the sensor immediately. I had to reboot the ESP or submit the task once more. This doesn't happen with BME280

fmuntean, 2022-10-29:
> And last but not least, when I upload the most recent config, it doesn't init the sensor immediately. I had to reboot the ESP or submit the task once more. This doesn't happen with BME280

Please explain this in more details. Not sure I understand the scenario and not sure if related to the plugin itself.

chemmex, 2022-10-30:
Just checked that once more. My expectation was that if I provision the clean unit by uploading a config.dat, the sensors must activate immediately without a reboot. Now I see it's not the case with most I2C sensors I have regardless of firmware version. So please disregard my last message

TD-er, 2022-10-30:
Nope, when uploading a config.dat file you _must_ reboot or else the settings file may become corrupt as parts are still active in memory and parts aren't.
So there will be inconsistencies if you continue making changes (and saving them) after uploading the settings file.

chemmex, 2022-10-30:
I probably was misguided by the fact that task values remain in RAM or RTC so they may reappear when I delete the task and add it at the same place (or upload the new config file)

TD-er, 2022-10-30:
The `config.dat` file consists of several sections.
- SettingsStruct
- ExtraTaskSettings per task (e.g. task names, task value names, etc.)
- Custom Task settings (some tasks need extra settings, which are very specific to that task. For example the Framed OLED task.

etc.
Of this, only the SettingsStruct is kept in memory and this one contains very basic task settings which do apply to (almost) all tasks.
Everything used in the source code starting with `PCONFIG` is taken from this SettingsStruct.

These will all be mixed up if you edit settings partly still in memory and partly existing in the newly uploaded file.

The task values will be kept in RTC memory, which may survive a reboot and those will be restored at reboot.
However I have to admit, I don't know what may happen when booting a different settings file.
Maybe not a bad idea to clear RTC when a different settings file is loaded...

chemmex, 2022-10-30:
Clearing RTC makes sense as I often saw the situation when after loading config file old values are not zeroed

chromoxdor, 2022-10-31:
As promised i tested it today and it looks good so far...

TD-er, 2022-10-31:
> As promised i tested it today and it looks good so far...

Which sensor(s) did you test?

chromoxdor, 2022-10-31:
htu21d

PR #4321 (2022-11-02): [P014 si70xx] fixing support for htu21D
Fixes: #4314
closed 2022-11-02 by TD-er