letscontrolit/ESPEasy #4315: senseair s8 on esp32

anno006, 2022-10-24:
- can't give the task a name
- can't use the formula.
- after giving it a name and or formula and save the page will be white for some time and than i get error, see attachment.
[image]

tonhuisman, 2022-10-24:
> Build: [ESP_Easy_mega_20220809_normal_ESP32_4M316k Aug 9 2022]
> 
>     * can't give the task a name
> 
>     * can't use the formula.
> 
>     * after giving it a name and or formula and save the page will be white for some time and than i get error, see attachment.
> ...
> the problem is not presend in the Build: [ESP_Easy_mega_20221010_normal_ESP32_4M316k Oct 10 2022] (did install it on a esp32 i have laying around, so there is no s8 sensor connected to it, but i can give the task a name and formula

That is a bug that's been fixed after the 20220809 release. You can use the 20221010 build without problems, and has the benefit of also having many more issues fixed.
There is expected to be a new official release soon.

TD-er, 2022-10-24:
One note of advice...
I would not advice to use Serial0 for a sensor.
The pins are probably fine, but rather set it to "Serial1" or "Serial2", whichever is available.

Serial0 is used by the log and handling inputs via the serial port. So this will interfere with the sensor and no idea what the SenseAir sensor will do when it receives all kinds of data it doesn't understand (like logs)

anno006, 2022-10-24:
> One note of advice... I would not advice to use Serial0 for a sensor. The pins are probably fine, but rather set it to "Serial1" or "Serial2", whichever is available.
>
> Serial0 is used by the log and [...]

Done, not a problem.

tonhuisman, 2023-09-09:
This seems to be solved, so can be closed.

closed 2023-09-09 by TD-er