letscontrolit/ESPEasy #4324: No event from Generic - MQTT Import device

dkotte, 2022-10-31:
### Summarize of the problem/feature request

The following rule works until i upgrade today. May be i skip some releases ...
on MQTT#NoSleep do
  if [MQTT#NoSleep] > 0 
     ....
  endif
endon

data for "MQTT#NoSleep" came from Generic - MQTT Import

Expected behavior, feature or bug? 
THX, Dirk

### Expected behavior

the code within rule should be executed

### Actual behavior

the code within rule is not executed

### System configuration

Hardware: Memos D1 mini

ESP Easy version:   mega-20220809_57b0c0c

### Rules or log data

```
on MQTT#NoSleep do
  if [MQTT#NoSleep] > 0 
     ....
  endif
endon
```

tonhuisman, 2022-10-31:
Can you show your Device configuration? (screenshot)

TD-er, 2022-10-31:
I think the task value may already have been cleared.
But why don't you use the `%eventvalue1%` here as the value you need is already included in the event itself and that value is for sure reflecting the correct state at the moment the event was generated.

```
on MQTT#NoSleep do
  if %eventvalue1% > 0 
     ....
  endif
endon
```

TD-er, 2022-10-31:
By the way, can you also test one of the latest GitHub Actions test builds, as we have fixed quite a few bugs in the past 2 months.
For example this build: https://github.com/letscontrolit/ESPEasy/actions/runs/3352900207

dkotte, 2022-10-31:
Seems git replace some text within code-fields ... i never seen or wrote " %eventvalue1%" before ...
I paste the following:
on MQTT#NoSleep do
  if [MQTT#NoSleep] > 0 
     ...
  endif
endon

I can use the "  if [MQTT#NoSleep] > 0        ...      endif  "   within other "on ..." parts of rules (an on that runs every minute for example) 
So i have the value ... but the "on ..." part is not triggert.

I'll try the latest GitHub test build tomorrow. THX!!

Devide-Config:
[image]

tonhuisman, 2022-10-31:
Most likely, enabling the `Generate events for accepted topics` checkbox will fix the issue.
Might probably have been better to auto-enable that setting for existing installations when I added a lot of features to that plugin (though not available in this build for .bin-size reasons) 🤔

TD-er, 2022-10-31:
> . i never seen or wrote " %eventvalue1%" before ...

See the [documentation](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html?highlight=%25eventvalue#event-value-eventvalue) on the use of eventvalues.
It's not a new feature, so you could have used it with (much) older revisions of the ESPEasy firmware too.
Anyway it is the "better" way to refer to values which are also part of the event itself.

When the event is sent, you get something like this:
```
MQTT#NoSleep=1
```
In the rules you try to match an event like this:
```
on MQTT#NoSleep do
  ...
endon
```

But these events are added to an event queue and processed one by one from this queue.
However, if you refer to `[MQTT#NoSleep]` in the rules when handling this event, it is possible the value of `[MQTT#NoSleep]` has changed from when the event was fired.
Thus it is better to use the `%eventvalue1%` (which is the first item after the `=` in the actual event) when handling the event as that is the actual value from the moment the event was generated. (it is `1` in this example of event `MQTT#NoSleep=1`)

tonhuisman, 2022-10-31:
Looked over the source code, but the standard events will be generated as long as the Rules are enabled.
Most likely, the value is already reset, like TD-er suggests, and it should be resolved by applying his solution of using `%eventvalue1%`.

dkotte, 2022-11-01:
damn i'm kinda blind...
The checkbox for "Generate events" solved the problem, of course.
Thank you so much !

tonhuisman, 2022-11-01:
> The checkbox for "Generate events" solved the problem, of course.

Hm, then it's something that can be improved, as it should be backward compatible as much as possible. I'll work on that.

PR #4327 (2022-11-02): [P037] Bugfix: Enable event generation when adding task
Fixes: #4324 

Since adding the Mapping, Filtering and JSON parsing features, this option wasn't enabled after adding the task.

Also reformatted the source a little using Uncrustify.
closed 2022-11-01 by TD-er