letscontrolit/ESPEasy #4325: Rule by Value

fly74, 2022-11-01:
Hi together,

I tried a rule to set a neopixel brightness by value of a light sensor. But it don't do anything. Do I need a polling timer?

on LED1LX do
if [LED1LX#Lux] = 0.00
 nfx,dim,65
else
 nfx,dim,1
endif
endon

The log shows the values:

4783731: EVENT: LED1LX#Lux=32.92
4787380: EVENT: Clock#Time=Tue,16:50
4788727: BH1750 Address: 0x23 Mode: 0x63 : Light intensity: 32.92
4788732: EVENT: LED1LX#Lux=32.92
4793727: BH1750 Address: 0x23 Mode: 0x63 : Light intensity: 32.92
4793732: EVENT: LED1LX#Lux=32.92
4799049: BH1750 Address: 0x23 Mode: 0x63 : Light intensity: 0.00
4799052: EVENT: LED1LX#Lux=0.00

tonhuisman, 2022-11-01:
You could write that like this:
```
on LED1LX#Lux do
  if %eventvalue1%=0.00
    nfx,dim,65
  else
    nfx,dim,1
  endif
endon
```
NB: In nearly all cases the dim value for the neopixel plugin will be set to 1
NB2: `nfx` suggests you are using the NeoPixel (BusFX) plugin?

TD-er, 2022-11-01:
What Ton said, use `%eventvalue1%` in the rules block as it seems these events will be sent with a short interval.
Thus the value can already be different from when the event was sent.

fly74, 2022-11-01:
@tonhuisman works ! Thanks a lot again.

closed 2022-11-01 by fly74

fly74, 2022-11-01:
> NB2: `nfx` suggests you are using the NeoPixel (BusFX) plugin?

Yes, @tonhuisman  is the fine BusFX plugin. And the main case is, if the light is enough then dim to 1.