letscontrolit/ESPEasy #4335: P121 HMC5883 freezing LoLin NodeMCU v3

aRestless, 2022-11-06:
### Summarize of the problem/feature request
Init of plugin P121 freezes the device, requiring a reflash. 

### Expected behavior
Adding a device via P121 makes the device available in the devices list, or error out if an issue is encountered.

### Actual behavior
Upon adding the device, the serial console stops immediately and does not react anymore, web UI (including additional endpoints like /metrics) do not load anymore. This problem persists after resetting or disconnecting the device from power. Only remedy is reflashing the device, sometimes requiring a blank .bin first.

### Steps to reproduce
1. Flash any Test/Collection E build since 20220328
2. In the web UI, go to the devices tab
3. Add a device via "Position - HCM5883L" (this caption includes a typo, the name of the sensor and the plugin is HMC5883L)
4. Enable the device

The behavior occurs independently of the settings on the plugin or if the sensor is actually connected. It has been verified on two different and previously unused LoLin NodeMCU v3.

### System configuration

```
Build:⋄
20116  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20220809_57b0c0c
Plugin Count:⋄
72 [Normal] [Collection] [Collection_E]
Build Time:⋄
Aug  9 2022 09:46:40
Binary Filename:⋄
ESP_Easy_mega_20220809_collection_E_ESP8266_4M1M
Build Platform:⋄
Linux-5.10.102.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
Git HEAD:⋄
mega-20220809_57b0c0c
```

ESP Easy version: Tested versions are 20220328, 20220616, 20220809, 20221105

ESP Easy settings/screenshots: 
(Screenshot before adding the device)
[image]

### Rules or log data
Complete serial console output for a run during which the device is added (adding the device does not result in a log line):

```
sd l▄ƒ< älÓ<♥♀♦♀ä♦lõ♀#|Åé♥ý‼øsø#ä♦cä¹'o▀do'£Òõ♦c∟p─Ãdsdrd8¹gÓ►♥♀♦é♦l♀─£♦♀♀b♀'Ò|♥õlÄ♀Ãcî¹ogþ läçd`♥ÿ‼←g'♀l`♥♥orÃøÆn♀♦c☻d`☻♫{Æøo♦♦c♥l ♥£c♦♀Ä∟♥ôd;ô`☻³é'£☻¬U74 : Info   : WIFI : Set WiFi to OFF
192 : Info   :

INIT : Booting version: mega-20220809_57b0c0c (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
193 : Info   : INIT : Free RAM:29392
194 : Info   : INIT : Manual Reboot #34 - Restart Reason: External System
195 : Info   : FS   : Mounting...
220 : Info   : FS   : Mount successful, used 46686 bytes of 957314
273 : Info   : FS   : Success garbage collection
327 : Info   : FS   : Success garbage collection
345 : Info   : CRC  : SecuritySettings CRC   ...OK
347 : Info   : INIT : I2C
348 : Info   : INIT : SPI not enabled
350 : Info   : Time set to 848.000 Time adjusted by 847651.00 msec. Wander: 0.000 msec/second Source: RTC at boot
351 : Info   : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
353 : Info   : Local time: 1970-01-01 00:14:08
355 : Info   : WiFi : Start network scan all channels
367 : Info   : INIT : Free RAM:27136
539 : Info   : INFO : Plugins: 72 [Normal] [Collection] [Collection_E] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
543 : Info   : Webserver: start
1884 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 20080 WiFiStatus 0 ESPeasy internal wifi status: DISCONNECTED
2680 : Info   : WiFi : Scan finished, found: 9
2684 : Info   : WiFi : Best AP candidate: Alveran 2C:91:AB:2E:DE:4B Ch:11 (-54dBm) WPA2/PSK
2685 : Info   : WIFI : Connecting Alveran 2C:91:AB:2E:DE:4B Ch:11 (-54dBm) WPA2/PSK attempt #0
3688 : Info   : WIFI : Connected! AP: Alveran (2C:91:AB:2E:DE:4B) Ch: 11 Duration: 989 ms
5661 : Info   : WIFI : DHCP IP: 192.168.178.144 (Collection-E) GW: 192.168.178.1 SN: 255.255.255.0   duration: 1982 ms
5725 : Info   : UDP : Start listening on port 8266
5726 : Info   : firstLoopConnectionsEstablished
```

aRestless, 2022-11-06:
**Further insights:** I2C scan finds the connected HMC5883L at IC2 Address 0x0D and adding other I2C-based devices does not freeze the device (tested: Position - GPS, Accelerometer - ADXL345 (I2C), Environment - DHT22 (I2C)).

So I think it's very likely an issue isolated to P121.

TD-er, 2022-11-06:
Have you also tried the latest build from yesterday?
I had issues building it locally so right now it is still a single huge download.

tonhuisman, 2022-11-06:
> I2C scan finds the connected HMC5883L at IC2 Address 0x0D

That's a bit weird, as the HMC5883 has a fixed I2C address of `0x1E`. 🤔 
It is missing support in the I2C scanner, but that scanner will always show the I2C address for any device found, whether it is supported by ESPEasy or not, but for the Collection builds, all recognized I2C device names are excluded anyway, to save on the .bin size.

tonhuisman, 2022-11-06:
Tested here, and the HMC5883 device is working as intended.

I'll fix up the plugin code, to have it show the proper name, and fix some other minor issues, but probably not today, got IRL events to attend to 😄

aRestless, 2022-11-06:
> Have you also tried the latest build from yesterday?

Yes I have.

> That's a bit weird, as the HMC5883 has a fixed I2C address of 0x1E. 🤔

Yep, that's something I uncovered in the last hour. I have happened to receive the QMC5883 here, which likely will not be compatible with the plugin. _Actual_ HMC5883Ls are on the way.

**But** that does not change the fact the NodeMCU freezes even if nothing is attached to I2C. Is it possible that the plugin simply can't handle the sensor not being attached (as none of my tests will actually have had the sensor at 0x1E)?

tonhuisman, 2022-11-06:
> Tested here, and the HMC5883 device is working as intended.

Well, I may have written down that wrongly. I added P121 to an ESP8266 Custom build, but it wouldn't compile properly, and I just fixed that. Then it worked as intended, so I may have put you on the wrong foot. Apologies for that 😊 
But I can't understand how the Collection E builds have compiled without issues all the time, with that syntax error in place 🤯 
I'll fix that in a PR, also correcting some other minor issues with this plugin.

tonhuisman, 2022-11-06:
> freezes even if nothing is attached to I2C.

The library used isn't responding properly if no sensor is detected. I'll see if that's fixable too.

PR #4339 (2022-11-09): [P121] Bugfix for unconnected sensor, correct name-typo, code cleanup
Resolves #4335 

- Bugfix in Adafruit_HMC5883_Unified library for blocking the unit when no sensor is connected
- Correct type in plugin name
- Uncrustify code and some generic code cleanup
- Add sensor to I2C scanner
closed 2022-11-09 by TD-er