letscontrolit/ESPEasy #4336: Add Network->Connected time as System Variable

LeeNX, 2022-11-06:
Please can we expose Network->Connected timer and possible Network->Number Reconnects as System Variables?
  
Looking to trend ```connected time``` vs ```uptime``` and see if I can track down device issues.  
  
Was going to ask if we could also allow ```System Variable``` in the ```System Info``` plugin, but see the items are pretty much fixed and it is most likely easy to do with rules (still to research).  
  
I was not able to find the variable in either the August released binaries or the ReadTheDocs pages. Possible I missed it or they named something I did not think of searching for.

I see ```espeasy_wifi_connected 22319000``` and ```espeasy_wifi_reconnects 1``` in the metrics page.  
  
If I am asking, for debugging, wonder if we could also give the following variables the same full treatment ```Max WiFi TX Power:17.50``` and ```Current WiFi TX Power:14.00``` and expose them?

Thanks

### Checklist
- [x] The title of this issue is "to the point" and descriptive.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [?] filled out all applicable fields below.

TD-er, 2022-11-06:
What is available is the event when a new connection is established.
So you can already act on it via rules right now, until these may have been added.

LeeNX, 2022-11-07:
Thanks @TD-er, but I don't understand. The feature request is useful, but you offering me a workaround until feature is out.  
  
I actually setup a few rules just after opening this request and held back asking for some more variables ... ;-)
  
Wondering if the configs are accessible as variables? ie: ```Controller LWT Topic```
```
on WiFi#ChangedAccesspoint do
 Publish,status/espeasy/%sysname%/status-accesspoint,%uptime% [%rssi%] %ssid% on %wi_ch%
endon

on WiFi#ChangedWiFichannel do
 Publish,status/espeasy/%sysname%/status-channel,%uptime% [%rssi%] %ssid% on %wi_ch%
endon
``` 
but after pulling up the controller, might not be so easy to label.

TD-er, 2022-11-07:
> Thanks @TD-er, but I don't understand. The feature request is useful, but you offering me a workaround until feature is out.

Well since I don't want to hold you back until it is implemented, I gave you a quick work-around so you can continue instead of waiting for the feature to be implemented.

Publishing immediately when the channel was changed may not work very reliable since the controller may still be in the process of connecting to the broker.
If the controller queue is full, these messages may get lost.

> Wondering if the configs are accessible as variables? ie: Controller LWT Topic

Not sure what you're asking here.
Can you elaborate a bit more?

LeeNX, 2022-11-07:
Thanks @TD-er for the quick reply and awesome work!  
  
Derailed my own train of thought. Not sure I get your event suggestion, but I might not need it. I was trying to figure why one of my nodes was getting hung up. Seems after checking force to G vs N WiFi network, ended up solving this at the moment.  
  
The attached rules just return ```0```, which is not what I was thinking I would get. Might just be worth adding to syslog.  
  
As for the  ```Controller LWT Topic```, was me hoping to replace ```status/espeasy/%sysname%/``` with something like ```%lwt_topic%``` as a template, but as the controller is not fixed, I don't think it would be an easy task.
  
Must admit, with my node now stable, this feature request does not seem very useful. I was hoping to possible catch when or if the uptime vs connected time differed too much. ie: network issue or something.

TD-er, 2022-11-07:
> I was trying to figure why one of my nodes was getting hung up. Seems after checking force to G vs N WiFi network, ended up solving this at the moment.

If you're using an access point which is capable of WiFi mesh (doesn't need to be enabled to cause issues), then it might be best to force ESPEasy into "802.11g" mode.
The reasoning behind this is that such access points may try to disconnect a node every now and then to allow "fast roaming" (part of it is 802.11r standard).
However the ESP modules don't seem to handle this very well.
By telling the access point you only understand 802.11g, the AP may no longer try to disconnect you.

LeeNX, 2022-11-07:
Yip, I remember seeing the discussion, but nothing can stop a newbie like me who thought I had set that options for a long running node. I just had not been trending the data and noticing it offline.

Been adding all my ESPEasy nodes to grafana.
[image]

Also have MikroTik APs, which I have seen also can be a problem. 

Pretty pictures might help with understanding other things.

Thanks again for all the continued awesome work!