letscontrolit/ESPEasy #4337: Config (in)compatibility between firmware revisions

ghtester, 2022-11-06:
I have upgraded one ESP Easy node  from custom firmware 20220829 to custom 20221031 as it started disappearing from WiFi network.  
After the upgrade and node reboot it seemed to be working fine but after several hours it disappeared again. 
Then I tried to edit some WiFi settings but after the reboot the node did not boot properly anymore. The WiFi status LED was just changing the light intensity from low to medium after several seconds and that was all.
Repeated connecting to serial console showed just some garbage characters or even nothing.
Then I decided to reflash with empty bin, reflashed again with custom 20221031 which was working again with default configuration.
As soon as I restored the earlier config, the described issue returned... Then I reflashed with custom 20220829 firmware which is so far working fine with the restored config.
The question is what could make the config incompatible with latest firmware release?
Another question is - could we make a feature for exporting all configuration parameters in text form from ESP node? 
Also a simple viewer or config exporter could be useful (I mean an app or a script ) to be used with already existing binary config files. Maybe I could perhaps try to do some work on it if somebody reminds me with the current config binary file structure.

Jason2866, 2022-11-06:
@TD-er Could be the change of the used SDK.

TD-er, 2022-11-06:
@Jason2866 Also for ESP8266 builds?

TD-er, 2022-11-06:
@ghtester I really don't know what may have made the settings incompatible.
The only thing added to the settings is the added md5 checksum at the end (to re-introduce the feature to not saving settings when nothing was changed) and the internal build nr is updated to know if we need to perform version transitions. These transitions are only used to convert some settings if we add some features.
For example P014 now supports multiple sensors which adds the need to set an I2C address. So there is a transition on existing settings to set the previous default address.

The settings are kept as much backwards compatible as possible.
As a matter of fact, last week I upgraded a node with 4 year old settings and this worked just fine.

Jason2866, 2022-11-06:
@TD-er Only for esp8266. The change of the used SDK you did.

TD-er, 2022-11-06:
> @TD-er Only for esp8266. The change of the used SDK you did.

Hmm I wasn't aware I did change something significant for the ESP8266.
Do you know what may have changed what may cause these things? Where do I need to look?

Jason2866, 2022-11-06:
> Or do you mean this: PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190313 ?

Yes! You opened the esp8266 pandoras box by doing this ;-)

TD-er, 2022-11-06:
Why does Pandora leave these boxes laying around...
Hasn't she learned to clean up?

ghtester, 2022-11-07:
> As a matter of fact, last week I upgraded a node with 4 year old settings and this worked just fine.

I don't know, maybe my custom build uses some unusual settings / directives in Custom.h  or a plugin combinations... Or a HW issue of affected ESP node (it's permanently running for about 4 years so may be it's a bit dying)?  
Perhaps I should try upgrading another ESP node but there's always a different configuration. As the custom release 20220829 works pretty fine on my nodes, also some of them are not easily reflashable if I defunct them by upgrade so I have to be careful.
Hopefully the SDK change is the reason... ;-)

TD-er, 2022-11-07:
I will look into the SDK change.
This was mainly done to have 802.11n connections which really obey the standards and also for those problematic units that worked on really old builds but no more on builds made in the last 3 years.

ghtester, 2022-11-08:
Thanks a lot, I have just recompiled the firmware from current sources, in the evening I'll try to upgrade the node again and share the feedback.

ghtester, 2022-11-08:
OK, some news - the same issue on the same node with fresh Vagrant custom build, right after the update & reboot it did not work properly anymore.  LCD display was initialized but rules did not work, WiFi LED changing light intensity after several seconds., no connection to WiFi AP.
Reflashed back to 20220929 and working again. Checked the configuration and discovered the Serial port & Serial Log  was disabled earlier to save RAM (that's why I did not see any serial log).

cybermcm, 2022-11-08:
I may have a similar problem (or maybe it is not related at all ;-), I'm not an expert.
I have 9 ESP8266 running on mega-20220809_normal just fine (most of them only have one temperature sensor attached)
Updated all of them to mega-20221105 -> 5 working without any issue, 4 do not connect to my wifi anymore. After rebooting a faulty device, the ESP are sending their own Wi-Fi, then I'm able to connect to it (192.168.4.1) and revert to 20220809 -> issue gone, all faulty devices working again (no config changed)
The ESP are connected to different APs, the APs running on OpenWRT (latest build)

currently my free time is somehow limited, but if any further data is useful I'll try to provide it.

ghtester, 2022-11-08:
Maybe it's HW related...
My problematic ESP node:
<html>
<body>

ESP Board
--
ESP Chip ID: | 3523742 (0x35C49E)
ESP Chip Frequency: | 80 MHz
ESP Chip Model: | ESP8266
ESP Chip Cores: | 1
ESP Board Name: | Espressif Generic ESP8266 ESPEasy 4M Flash 1M FS

</body>
</html>

Working firmware:
<html>
<body>

Firmware
--
Build:⋄ | 20220829 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Aug 29 2022 09:27:48
Plugin Count:⋄ | 39 [IR]
Build Origin: | Vagrant
Build Time:⋄ | Aug 29 2022 09:27:04
Binary Filename:⋄ | ESP_Easy_mega_20220829_custom_IR_ESP8266_4M1M
Build Platform:⋄ | Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄ | mega_44b2677

</body>
</html>

TD-er, 2022-11-08:
I'm thinking about the following idea:

The SDK now used for ESP8266 builds is `PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190313`
This seems to be the (only?) one that's working on "flaky" ESP nodes, like the Sonoff Basic units.
What if I revert this SDK back to the one we used on the August build and then set those builds we now have labelled "alt WiFi" to use this 190313 SDK.
Right now, I have no idea what is the very specific cause for units that won't work with one SDK and only work with the other and vice verse. Simply because these SDK binary blobs have no release notes or anything indicating what may be different.

TD-er, 2022-11-08:
@ghtester Maybe you can already test with the current "alt WiFi" PIO env?
It is using a different SDK from the August builds, but at least it is different from the current "default" one.

ghtester, 2022-11-08:
I am not sure how the configuration option should look like exactly to use "alt WiFi".  For me it's important to have the IR related plugins compiled and I am not sure if there's any possible conflict / missing plugins or not.
So I would prefer to revert the SDK with default env. Unfortunately my build environment is inaccessible now anyway so I could recompile tomorrow.

TD-er, 2022-11-08:
There are various "custom" builds defined.
[image]

But I don't see an alt WiFi one with IR.
Let's continue this tomorrow after I have done some tests on the "problematic" units to see if they now will work well on the previous used SDK for ESP8266.

ghtester, 2022-11-08:
OK, thanks for the details, missing IR is a trouble for me.
BTW. this is the serial log after upgraded again to "bad" 20221108:
 ▒U11756 : Info   : WIFI : Set WiFi to AP+STA
16546 : Info   : WIFI : Set WiFi to OFF
16763 : Info   :

INIT : Booting version: ESP_Easy_mega_20221108_custom_IR_ESP8266_4M1M, (Vagrant) My Build: Nov  8 2022 15:03:09 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(c0eb301), LWIP: 2.1.2 PUYA support)
16764 : Info   : INIT : Free RAM:25272
16766 : Info   : INIT : Cold Boot - Restart Reason: Software/System restart
16767 : Info   : FS   : Mounting...
16793 : Info   : FS   : Mount successful, used 78061 bytes of 957314
16809 : Error  : CRC  : Settings CRC           ...FAIL
16811 : Info   : CRC  : SecuritySettings CRC   ...OK
16814 : Info   : INIT : I2C
16815 : Info   : INIT : SPI not enabled
16921 : Info   : WIFI : Set WiFi to STA
19218 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-44dBm) WPA2/PSK attempt #0
19432 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 112 ms
19534 : Info   : Reset WiFi.
21730 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-43dBm) WPA2/PSK attempt #1
22680 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 112 ms
22681 : Info   : WIFI : Set WiFi to OFF

Build changed!
22869 : Info   : Skip saving SecuritySettings, not changed
22870 : Info   : INIT : Free RAM:24608
22906 : Info   : P016_PLUGIN_INIT ...
22908 : Info   : INIT: IR RX
22909 : Info   : IR lib Version: 2.8.2
22909 : Info   : P016_PLUGIN_INIT IR receiver created
22910 : Info   : P016_PLUGIN_INIT IR receiver initialized
22910 : Info   : P016_PLUGIN_INIT done
24060 : Info   : INFO : Plugins: 40 [IR][No Debug Log] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(c0eb301), LWIP: 2.1.2 PUYA support)
24061 : Info   : EVENT: System#Wake
24214 : Info   : Webserver: start
24214 : Info   : EVENT: System#Boot
24218 : Info   : ACT  : rtttl,15:d=10,o=6,b=200,c,g,e
24220 : Info   : RTTTL : pin: 15 melody: 15:d=10,o=6,b=200,c,g,e
24583 : Info   : ACT  : LCDCMD,off
24587 : Info   : ACT  : timerSet,1,5
24591 : Info   : ACT  : gpio,10,0
24593 : Info   : GPIO : port#10: set to 0
24594 : Info   : ACT  : gpio,10,1
24596 : Info   : GPIO : port#10: set to 1
24598 : Info   : ACT  : gpio,10,0
24599 : Info   : GPIO : port#10: set to 0
24604 : Info   : WIFI : Disconnected! Reason: '(202) Auth fail' Connected for 951 ms
24805 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 24 s
24906 : Info   : Reset WiFi.
25010 : Info   : WIFI : Set WiFi to STA
27310 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-47dBm) WPA2/PSK attempt #2
27417 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 5 ms
27518 : Info   : Reset WiFi.
29715 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-47dBm) WPA2/PSK attempt #3
30664 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 5 ms
30665 : Info   : WIFI : Set WiFi to OFF
30989 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 1173 ms
31090 : Info   : Reset WiFi.
31195 : Info   : WIFI : Set WiFi to STA
33493 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-47dBm) WPA2/PSK attempt #4
33500 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 1173 ms
33501 : Info   : WIFI : Set WiFi to OFF
33715 : Info   : EVENT: WiFi#APmodeDisabled
33833 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 238 ms
33934 : Info   : Reset WiFi.
34039 : Info   : WIFI : Set WiFi to STA
36338 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-47dBm) WPA2/PSK attempt #5
36345 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 238 ms
36345 : Info   : WIFI : Set WiFi to OFF
36560 : Info   : EVENT: WiFi#Disconnected
[... 48 lines cut ...]
56479 : Info   : EVENT: TaskInit#Status=10,1
56598 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 239 ms
56699 : Info   : Reset WiFi.
56804 : Info   : WIFI : Set WiFi to STA
59103 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-45dBm) WPA2/PSK attempt #13
59110 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 239 ms
59110 : Info   : WIFI : Set WiFi to OFF
59325 : Info   : EVENT: TaskInit#LCD=11,1
59444 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 239 ms
59545 : Info   : Reset WiFi.
59649 : Info   : WIFI : Set WiFi to STA
61949 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-44dBm) WPA2/PSK attempt #14
61956 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 239 ms
61956 : Info   : WIFI : Set WiFi to OFF
62171 : Info   : EVENT: TaskInit#PIR=12,1
62289 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 239 ms
62390 : Info   : Reset WiFi.
62495 : Info   : WIFI : Set WiFi to STA
64795 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-44dBm) WPA2/PSK attempt #15
64802 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 239 ms
64803 : Info   : WIFI : Set WiFi to OFF
65017 : Info   : EVENT: WiFi#Disconnected
65135 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 238 ms
65236 : Info   : Reset WiFi.
65341 : Info   : WIFI : Set WiFi to STA

ghtester, 2022-11-08:
BTW. when I entered the **WifiAPMode** command to serial console, the node connected to AP and started working. :-) To be analyzed a bit...

TD-er, 2022-11-08:
hmm can you set it to " force B/G"?
Thing is, with AP enabled, the ESP radio cannot use 802.11n, thus is forced to use 802.11g anyway.

TD-er, 2022-11-08:
By the way, this does look like a big clue to me.
The SDK currently used is the only one (I am aware of) that does allow for 802.11n according to the standards.
The later SDK versions (at least for the 2.x SDK revisions) do not connect fully according to the specs.
So it could be the other SDK versions were already treated by the access point as 802.11g clients.

ghtester, 2022-11-08:
See the log:
....
62217 : Info   : Reset WiFi.
62321 : Info   : WIFI : Set WiFi to STA
64622 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-51dBm) WPA2/PSK attempt #19
64629 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 239 ms
64630 : Info   : WIFI : Set WiFi to OFF
64844 : Info   : EVENT: WiFi#Disconnected
>WifiAPMode
64955 : Info   : WIFI : Set WiFi to AP
65061 : Error  : WIFI : Error while starting AP Mode with SSID: ESP11 IP: 192.168.4.1

OK
65119 : Info   : ADC  : Analog value: 1 = 1.00
65197 : Info   : Dummy: value 1: 0
65199 : Info   : Dummy: value 2: 0
65199 : Info   : Dummy: value 3: 0
65200 : Info   : Dummy: value 4: 0
65297 : Info   : EVENT: WiFi#Disconnected
65426 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 13736 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
65596 : Info   : EVENT: WiFi#Disconnected
65662 : Info   : EVENT: Rules#Timer=1,1
65670 : Info   : ACT  : TaskValueSet 10,3,1
65674 : Info   : ACT  : rtttl,15:d=100,o=7,b=200,g
65676 : Info   : RTTTL : pin: 15 melody: 15:d=100,o=7,b=200,g
65691 : Info   : ACT  : gpio,10,1
65693 : Info   : GPIO : port#10: set to 1
66104 : Info   : EVENT: WiFi#Disconnected
66658 : Info   : EVENT: WiFi#Disconnected
66873 : Info   : ADC  : Analog value: 1 = 1.00
67213 : Info   : EVENT: WiFi#Disconnected
67729 : Info   : EVENT: WiFi#Disconnected
68132 : Info   : EVENT: WiFi#Disconnected
68154 : Info   : IR: No replay solutions found! Press button again or try RAW encoding (timings are in the serial output)
IR: RAW TIMINGS: uint16_t rawData[17] = {186, 7698,  588, 2530,  522, 532,  306, 5202,  332, 5524,  446, 1864,  302, 17282,  328, 3524,  382};  // UNKNOWN 8D5ED574
   68246 : Info   : EVENT: WiFi#Disconnected
68363 : Info   : EVENT: WiFi#Disconnected
68415 : Info   : EVENT: WiFi#Disconnected
68433 : Info   : EVENT: WiFi#Disconnected
68451 : Info   : EVENT: WiFi#Disconnected
68481 : Info   : EVENT: WiFi#Disconnected
68499 : Info   : EVENT: WiFi#Disconnected
68516 : Info   : EVENT: ADC#Voltage=1.00
68535 : Info   : EVENT: SW1#State=1
68541 : Info   : ACT  : TaskValueSet 10,2,1
68558 : Info   : EVENT: Status#1=0
68592 : Info   : EVENT: Status#2=0
68611 : Info   : EVENT: Status#3=0
68628 : Info   : EVENT: Status#4=0
68646 : Info   : EVENT: PIR#State=0
68671 : Info   : ACT  : TaskValueSet 10,1,0
68700 : Info   : EVENT: ADC#Voltage=1.00
68719 : Info   : EVENT: IR#input=2371802484
71601 : Info   : ADC  : Analog value: 1 = 1.00
71619 : Info   : EVENT: ADC#Voltage=1.00
71802 : Info   : Dummy: value 1: 0
71803 : Info   : Dummy: value 2: 1
71804 : Info   : Dummy: value 3: 1
71805 : Info   : Dummy: value 4: 0
71823 : Info   : EVENT: Status#1=0
71841 : Info   : EVENT: Status#2=1
71859 : Info   : EVENT: Status#3=1
71889 : Info   : EVENT: Status#4=0
74400 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 14960 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
74728 : Info   : WIFI : Set WiFi to AP+STA
74836 : Info   : WIFI : Connecting MyTestAP 17:C4:C7:FE:11:84 Ch:12 (-51dBm) WPA2/PSK attempt #20
76586 : Info   : WIFI : DHCP IP: 192.168.36.102 (ESP11) GW: 192.168.36.1 SN: 255.255.255.0   duration: 536 ms
76587 : Info   : IP   : Fixed IP octet:192.168.36.141
76591 : Info   : firstLoopConnectionsEstablished
76597 : Info   : WIFI : Connected! AP: MyTestAP (17:C4:C7:FE:11:84) Ch: 12 Duration: 1199 ms
76598 : Info   : WIFI : Starting mDNS...
76601 : Info   : WIFI : mDNS started, with name: ESP11.local
77721 : Info   : EVENT: WiFi#Connected
77730 : Info   : EVENT: WiFi#ChangedAccesspoint
77749 : Error  : MQTT : Intentional reconnect
78014 : Info   : MQTT : Connected to broker

I think B/G  won't help as I have enabled it right before the issue was encountered first time. I'll try reboot

TD-er, 2022-11-08:
Another thing changed with AP enabled is that the WiFi radio never sleeps.
So this may be -again- one of those awfully timing critical issues.
You could try to reduce the max. TX power and still have the "send with max. TX power" checked.

For example by setting it to 7 dBm, you already reduce the TX power (in Watt) considerably and "7" is probably still enough given your AP is seen at an RSSI of -51.

ghtester, 2022-11-08:
Yes I can confirm forcing B/G mode does not help.
The node is now next to AP but the original location is a bit far from it and the issue was exactly the same. So I think struggling with TX power won't bring any useful feedback. It should work reliable with any settings...

ghtester, 2022-11-09:
FYI the second mentioned ESP node which I upgraded from 20220829 to 20221108 without issue (a different configuration & plugins used but connected to the same AP and with the **Do Not Start AP:** option checked) :

<html>
<body>

ESP Board
--
ESP Chip ID: | 5859987 (0x596A93)
ESP Chip Frequency: | 80 MHz
ESP Chip Model: | ESP8266
ESP Chip Cores: | 1
ESP Board Name: | Espressif Generic ESP8266 ESPEasy 4M Flash 1M FS

</body>
</html>
<html>
<body>

Firmware
--
Build:⋄ | 20221108 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(c0eb301), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Nov 8 2022 15:03:09
Plugin Count:⋄ | 40 [IR][No Debug Log]
Build Origin: | Vagrant
Build Time:⋄ | Nov 8 2022 15:02:21
Binary Filename:⋄ | ESP_Easy_mega_20221108_custom_IR_ESP8266_4M1M
Build Platform:⋄ | Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄ | mega_ba182c5

</body>
</html>

TD-er, 2022-11-09:
```
13063 : Info : WIFI : Disconnected! Reason: '(202) Auth fail' Connected for 951 ms
13265 : Info : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 13 s
```
This part is a bit strange... 
It does receive the disconnect event twice.
But it may give me a hint where to look as I did 'fix' this last week for ESP32.
On ESP32 you register callback functions to capture the events.
But before re-registering them, you need to unregister them.
The reason you need to re-register them (which I found out last week) is that the SDK only registers those events which are applicable for the currently active interfaces. Thus causing issues when switching between STA/AP/Ethernet.
When not de-registering the callback and/or using the same callback function for all interface related events, you get them multiple times.
So maybe this also applies to ESP8266...

ghtester, 2022-11-09:
So the question is what's the main reason of this different behaviour.
Nevertheless, the topic name now looks a bit misleading and perhaps I should change it... 
BTW. in this latest firmware release there's no ESP Easy icon in web browser's tab like on earlier version...
[image]
[image]

TD-er, 2022-11-09:
> BTW. in this latest firmware release there's no ESP Easy icon in web browser's tab like on earlier version...

Yep, I noticed too.
A bit strange, as it is being served when you query it.

TD-er, 2022-11-09:
Fix for the Favicon: https://github.com/letscontrolit/ESPEasy/pull/4346

ghtester, 2022-11-09:
I am just recompiling, as the PR4347 was already merged, I hope I don't need any change in config files.
The result to be shared in evening as the node is at different location.

TD-er, 2022-11-09:
> I hope I don't need any change in config files.

Nope, that should still be the same.
Looking forward to the test results

cybermcm, 2022-11-09:
@TD-er: I can now confirm that I'm talking about the same issue. I tried your new build from today 20221109 and now my device can connect again. If I check "Do not start AP" then there is no successful connection like you described
But at least in my case there is still a problem, the connection is not stable, there are WiFi "reboots". I managed to capture the output:
```
Info   : WIFI : Set WiFi to AP+STA
4957 : Info   : WIFI : Set WiFi to OFF
5174 : Info   :

INIT : Booting version: ESP_Easy_mega_20221109_normal_ESP8266_4M1M, (GitHub Actions) HEAD_96ae578 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
5175 : Info   : INIT : Free RAM:26520
5176 : Info   : INIT : Cold Boot - Restart Reason: Software/System restart
5177 : Info   : FS   : Mounting...
5203 : Info   : FS   : Mount successful, used 75802 bytes of 957314
5290 : Info   : FS   : Success garbage collection
5310 : Error  : CRC  : Settings CRC           ...FAIL
5312 : Info   : CRC  : SecuritySettings CRC   ...OK
5325 : Info   : INIT : I2C
5326 : Info   : INIT : SPI not enabled
5339 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK
5443 : Info   : WIFI : Set WiFi to STA
5546 : Info   : WiFi : Start network scan all channels
7730 : Info   : WiFi : Scan finished, found: 4
7733 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-67dBm) WPA2/PSK
7733 : Info   : WiFi : Added known candidate, try to connect
7735 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-67dBm) WPA2/PSK attempt #0
7736 : Info   : WIFI : Set to 802.11g
7820 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
7822 : Info   : WiFi : WifiDisconnect()
7924 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 87 ms
8025 : Info   : Reset WiFi.
8027 : Info   : WiFi : Start network scan all channels
10210 : Info   : WiFi : Scan finished, found: 4
10213 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK
10214 : Info   : WiFi : Added known candidate, try to connect
10216 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK attempt #1
10223 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
11165 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 87 ms
11166 : Info   : WIFI : Set WiFi to OFF

Build changed!
11282 : Info   : SaveToFile: free stack: 3712
11358 : Info   : FILE : Saved config.dat offset: 0 size: 1316
11359 : Info   : SaveToFile: free stack after: 3712
11359 : Info   : Skip saving SecuritySettings, not changed
11371 : Info   : INIT : Free RAM:25912
11484 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
11485 : Error  : WiFi : Connecting not possible, unprocessed WiFi events:  disconn
11486 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
11489 : Info   : Webserver: start
11503 : Info   : WIFI : Disconnected! Reason: '(202) Auth fail' Connected for 951 ms
11604 : Info   : WiFi : WifiDisconnect()
11705 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 11 s
11806 : Info   : Reset WiFi.
11910 : Info   : WIFI : Set WiFi to STA
12013 : Info   : WiFi : Start network scan all channels
14198 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK
14199 : Info   : WiFi : Scan finished, found: 3
14201 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK
14202 : Info   : WiFi : Added known candidate, try to connect
14204 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK attempt #2
14211 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
14214 : Info   : WiFi : WifiDisconnect()
14316 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 10 ms
14417 : Info   : Reset WiFi.
14420 : Info   : WiFi : Start network scan all channels
16604 : Info   : WiFi : Scan finished, found: 4
16607 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-67dBm) WPA2/PSK
16608 : Info   : WiFi : Added known candidate, try to connect
16610 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-67dBm) WPA2/PSK attempt #3
16617 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
17559 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 10 ms
[... 68 lines cut ...]
326370 : Info   : WiFi : Scan finished, found: 6
326373 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-67dBm) WPA2/PSK
326374 : Info   : WiFi : Added known candidate, try to connect
326376 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-67dBm) WPA2/PSK attempt #6
326383 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
327330 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 10 ms
327331 : Info   : WIFI : Set WiFi to OFF
327546 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
327605 : Error  : MQTT : Connection lost, state: Connection lost
342845 : Info   : WD   : Uptime 6 ConnectFailures 0 FreeMem 17752 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
[... cut ...]
349791 : Info   : Subscribed to: ESP_E_Arbeitszimmer/#
372835 : Info   : WD   : Uptime 6 ConnectFailures 0 FreeMem 18144 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
402836 : Info   : WD   : Uptime 7 ConnectFailures 0 FreeMem 18144 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
432836 : Info   : WD   : Uptime 7 ConnectFailures 0 FreeMem 18144 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
```
This doesn't happen with 20220809

ghtester, 2022-11-09:
Yeah, here also the same issue with the fresh update to 20221109. So the SDK change did not help unfortunately.
Only the plus is that the icon in browser tab is back...  ;-)

TD-er, 2022-11-09:
The "disconnect" after 5 minutes is to be expected.
I do disable the AP mode after 5 minutes, as long as there is no client connected to it.
When switching WiFi modes, I now clear the connection and reconnect.
So this single disconnect is "perfectly fine".

ghtester, 2022-11-10:
Do you have an idea what source changes from 20220829 to 20221031 could have brought this issue?

TD-er, 2022-11-10:
> Do you have an idea what source changes from 20220829 to 20221031 could have brought this issue?

+/- 2 months of debugging sounds like a plausible source...

cybermcm, 2022-11-11:
I tried to update another "faulty" module, it can't connect after an update to 20221109. Output attached. Reverting to 20220809 with same config -> all working again
```
¬U178 : Info   : WIFI : Set WiFi to AP+STA
4953 : Info   : WIFI : Set WiFi to OFF
5171 : Info   :

INIT : Booting version: ESP_Easy_mega_20221109_normal_ESP8266_4M1M, (GitHub Actions) HEAD_96ae578 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
5171 : Info   : INIT : Free RAM:26520
5173 : Info   : INIT : Manual Reboot #2 Last Action before Reboot: Const Interval: TIMER_MQTT_DELAY_QUEUE Last systime: 335 - Restart Reason: External System
5174 : Info   : FS   : Mounting...
5200 : Info   : FS   : Mount successful, used 75802 bytes of 957314
5219 : Info   : CRC  : Settings CRC           ...OK
5223 : Info   : CRC  : SecuritySettings CRC   ...OK
5236 : Info   : INIT : I2C
5238 : Info   : INIT : SPI not enabled
5251 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-62dBm) WPA2/PSK
5356 : Info   : WIFI : Set WiFi to STA
5460 : Info   : WiFi : Start network scan all channels
7643 : Info   : WiFi : Scan finished, found: 2
7645 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-63dBm) WPA2/PSK
7647 : Info   : WiFi : Added known candidate, try to connect
7648 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-63dBm) WPA2/PSK attempt #0
7654 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
12021 : Info   : WiFi : WifiDisconnect()
12123 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 4373 ms
12124 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
12225 : Info   : Reset WiFi.
12227 : Info   : WiFi : Start network scan all channels
14410 : Info   : WiFi : Scan finished, found: 5
14413 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-63dBm) WPA2/PSK
14414 : Info   : WiFi : Added known candidate, try to connect
14416 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-63dBm) WPA2/PSK attempt #1
14423 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
18890 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
18892 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 4373 ms
18893 : Info   : WIFI : Set WiFi to OFF
19007 : Info   : INIT : Free RAM:25544
19093 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
19198 : Info   : WIFI : Set WiFi to STA
19303 : Info   : WiFi : Start network scan all channels
21488 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-63dBm) WPA2/PSK
21489 : Info   : WiFi : Scan finished, found: 4
21491 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-62dBm) WPA2/PSK
21492 : Info   : WiFi : Added known candidate, try to connect
25957 : Info   : WIFI : Set WiFi to AP+STA
26949 : Info   : WIFI : AP Mode ssid will be ESP-E-Cat with address 192.168.4.1
26953 : Info   : Webserver: start
26993 : Info   : BMx280 : Detected BME280
28309 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 18112 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
29775 : Info   : BME280: dew point 6.20C
29802 : Info   : BME280 : Address: 0x76
29804 : Info   : BME280 : Temperature: 21.35
29806 : Info   : BME280 : Humidity: 37.48
29808 : Info   : BME280 : Barometric Pressure: 1000.77
34426 : Info   : WiFi : Start network scan all channels
36611 : Info   : WiFi : Scan finished, found: 4
36613 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-65dBm) WPA2/PSK
36614 : Info   : WiFi : Added known candidate, try to connect
36616 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-65dBm) WPA2/PSK attempt #2
36622 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
40990 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
58332 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 17792 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
88338 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 17792 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
118325 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 17792 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
141004 : Info   : WiFi : Start network scan all channels
143187 : Info   : WiFi : Scan finished, found: 4
143190 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-63dBm) WPA2/PSK
143191 : Info   : WiFi : Added known candidate, try to connect
143193 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-63dBm) WPA2/PSK attempt #3
143198 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
147565 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
148339 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 17792 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
178338 : Info   : WD   : Uptime 3 ConnectFailures 0 FreeMem 17792 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
[... cut ...]
249760 : Info   : WiFi : Added known candidate, try to connect
249762 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-64dBm) WPA2/PSK attempt #4
249768 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
254136 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
268365 : Info   : WD   : Uptime 4 ConnectFailures 0 FreeMem 17792 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
```

TD-er, 2022-11-11:
OK, so it seems like I may have changed some timings which made it "more compatible" with the 1903xx SDK and less "compatible" with the later SDK binary blobs.

I've also seen the same here on my test node on my desk.
It no longer is able to make a stable connection with the 'default' SDK and when I flash the alt.wifi build, it works perfectly fine.

I'm now looking into options to erase the RFcal partition and the WiFi partition.
Made also a "partition chart" for the ESP8266 builds, similar to what already was present on ESP32 builds:
[image]

TD-er, 2022-11-11:
See this PR: https://github.com/letscontrolit/ESPEasy/pull/4354

Could any of you flash one of the builds (either alt.WiFi, or the regular builds) which do perform badly with WiFi?
I did add 2 commands, which can be executed via serial or rules (restricted commands, thus cannot be given from any source) to clear the RFcal partition or the SDK WiFi partition.

The commands are:
- `ClearSdkWifi`
- `ClearWifiRFcal`

After giving these, it is best to reboot the node.
Not sure if a power cycle instead of the warm reboot really makes a difference here.

Jason2866, 2022-11-11:
@TD-er  Tasmota has a command to erase the wifi calib data too. A power cycle is needed to start the wifi calib process and using the new values. Just erasing does mean the current values are still used.

TD-er, 2022-11-11:
Thanks @Jason2866 
Does Tasmota erase both partitions (Noticed there is a function for that in the `ESP` class) or only the RFcal sector?

Jason2866, 2022-11-11:
It is here https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_support/support_esp.ino#L256-L293

cybermcm, 2022-11-11:
@TD-er tried normal 20221111 build -> no luck -> still no connection.
Tried command `ClearSdkWifi` -> worked
command `ClearWifiRFcal` -> command unknown?
did power cylce

tonhuisman, 2022-11-11:
That might be a typo in the source...: https://github.com/letscontrolit/ESPEasy/pull/4354/files#diff-0c0d79067d2a3b128344094f5387f9708c268f35cdb99d1b865d4e74ae3dfa75R280

TD-er, 2022-11-11:
Yep, made a typo, so with this build you may need to use `clearwifirfcall` (with 2x `l`)
For any next build, it should be according to the documentation.

TD-er, 2022-11-11:
> It is here https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_support/support_esp.ino#L256-L293

That looks like ESP32 code.

I meant the ESP8266 code in `bool EspClass::eraseConfig(void)`

Jason2866, 2022-11-11:
@TD-er  Discord? https://discord.gg/Ks2Kzd4  More responsive.

TD-er, 2022-11-11:
OK, I am on Discord apparently but I have no clue how to add someone there as I apparently need some 4 digit tag nr.

Jason2866, 2022-11-11:
Have you got an invite? And accepted? If yes whats your @name there?
You can tag me with @Jason2866 there too.

TD-er, 2022-11-11:
> Have you got an invite? And accepted? If yes whats your @name there? You can tag me with @Jason2866 there too.

@TD-er of course :)
Tried to send a DM, but then you apparently need to add 'friends' first...

cybermcm, 2022-11-11:
@TD-er tried again with both commands (`clearwifirfcall`) both commands worked -> power cycle -> works for some time and then breaks and doesn't come back online again
```
INIT : Booting version: ESP_Easy_mega_20221111_normal_ESP8266_4M1M, (GitHub Actions) HEAD_103c3b6 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
5169 : Info   : INIT : Free RAM:26640
5171 : Info   : INIT : Manual Reboot #1 Last Action before Reboot: Background Task Last systime: 0 - Restart Reason: External System
5172 : Info   : FS   : Mounting...
5197 : Info   : FS   : Mount successful, used 75802 bytes of 957314
5216 : Error  : CRC  : Settings CRC           ...FAIL
5219 : Info   : CRC  : SecuritySettings CRC   ...OK
5233 : Info   : INIT : I2C
5234 : Info   : INIT : SPI not enabled
5247 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-67dBm) WPA2/PSK
5351 : Info   : WIFI : Set WiFi to STA
5455 : Info   : WiFi : Start network scan all channels
7638 : Info   : WiFi : Scan finished, found: 4
7640 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-67dBm) WPA2/PSK
7641 : Info   : WiFi : Added known candidate, try to connect
7643 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-67dBm) WPA2/PSK attempt #0
7649 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
12014 : Info   : WiFi : WifiDisconnect()
12117 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 4372 ms
12117 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
12219 : Info   : Reset WiFi.
12221 : Info   : WiFi : Start network scan all channels
14403 : Info   : WiFi : Scan finished, found: 4
14406 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-68dBm) WPA2/PSK
14407 : Info   : WiFi : Added known candidate, try to connect
14409 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-68dBm) WPA2/PSK attempt #1
14416 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
18883 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
18885 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 4372 ms
18886 : Info   : WIFI : Set WiFi to OFF

Build changed!
19001 : Info   : Skip saving settings, not changed
19003 : Info   : Skip saving SecuritySettings, not changed
19108 : Info   : WIFI : Set WiFi to STA
19211 : Info   : WiFi : Start network scan all channels
21394 : Info   : WiFi : Scan finished, found: 4
21397 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK
21398 : Info   : WiFi : Added known candidate, try to connect
25864 : Info   : WIFI : Set WiFi to AP+STA
26855 : Info   : WIFI : AP Mode ssid will be ESP-E-Cat with address 192.168.4.1
26859 : Info   : INIT : Free RAM:22176
26945 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
26950 : Info   : Webserver: start
27001 : Info   : BMx280 : Detected BME280
28307 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 18360 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
28766 : Info   : BME280: dew point 7.32C
28792 : Info   : BME280 : Address: 0x76
28794 : Info   : BME280 : Temperature: 22.67
28795 : Info   : BME280 : Humidity: 37.30
28797 : Info   : BME280 : Barometric Pressure: 1000.49
34416 : Info   : WiFi : Best AP candidate: cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK
34418 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK attempt #2
34424 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
37425 : Info   : WIFI : DHCP IP: 192.168.10.100 (ESP-E-Cat) GW: 192.168.10.254 SN: 255.255.255.0   duration: 1890 ms
37427 : Info   : WIFI : Arduino wifi status: WL_CONNECTED 3 ESPeasy internal wifi status: IP
37430 : Info   : firstLoopConnectionsEstablished
37435 : Info   : WIFI : Connected! AP: cybermcm (32:23:03:DE:13:70) Ch: 1 Duration: 1110 ms
39465 : Error  : MQTT : Intentional reconnect
39673 : Error  : MQTT : Broker C005 connection failed (1/0)
58302 : Info   : WD   : Uptime 1 ConnectFailures 1 FreeMem 16984 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
69569 : Error  : MQTT : Intentional reconnect
69580 : Info   : MQTT : Connected to broker with client ID: ESP_E_Cat_0
69584 : Info   : Subscribed to: ESP_E_Cat/#
88295 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 16080 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
97753 : Info   : NTP  : NTP replied: delay 11 mSec Accuracy increased by 0.006 seconds
97755 : Info   : Time set to 1668180771.006
97756 : Info   : Current Time Zone:  DST time start: 2022-03-27 02:00:00 offset: 120 min STD time start: 2022-10-30 03:00:00 offset: 60 min
[... 36 lines cut ...]
340805 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 4373 ms
340806 : Info   : WIFI : Set WiFi to OFF
341066 : Error  : MQTT : Connection lost, state: Connection lost
341123 : Info   : WD   : Uptime 5 ConnectFailures 0 FreeMem 18000 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
356435 : Info   : WIFI : Set WiFi to STA
356540 : Info   : WiFi : Start network scan all channels
358723 : Info   : WiFi : Scan finished, found: 4
[... cut ...]
465295 : Info   : WiFi : Added known candidate, try to connect
465297 : Info   : WIFI : Connecting cybermcm 32:23:03:DE:13:70 Ch:1 (-66dBm) WPA2/PSK attempt #6
465303 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
469670 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
478314 : Info   : WD   : Uptime 8 ConnectFailures 0 FreeMem 19328 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED
```

Jason2866, 2022-11-11:
Can you try the invite link from above? I should see you than.

cybermcm, 2022-12-26:
tried new [mega-20221224](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20221224) build, worked for some time but now 2 of my devices are "down", will revert to https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20220809

tonhuisman, 2022-12-26:
> worked for some time but now 2 of my devices are "down"

Can you elaborate on that, like with logs captured from USB serial? (Assuming you have the serial output enabled)

cybermcm, 2022-12-26:
@tonhuisman: I think it will be the same problem as stated above, but I'll double-check tomorrow (both devices are not so easy reachable)

richik, 2022-12-26:
I observe similar issues with latest builds - 20221105 and 20221224. Signal around -62 dBm, but with latest build ping avg. 35 ms, sometimes 5 ms, usually 50-150 ms.
Reverted back to 20220809 - ping avg. 5 ms, usually 3-15 ms - can be considered as normal.

TD-er, 2022-12-26:
Did you also power cycle the ESP since you flashed it?

richik, 2022-12-27:
Tried once again - power cycle done (approx. 1 minute power unplugged) - still the same.

cybermcm, 2022-12-27:
@tonhuisman: I'm still testing. I reinstalled 20221224 and my "faulty" device is still up and running for 6 hours now. Log shows nothing unusual up until now

cybermcm, 2022-12-29:
@tonhuisman: 2 of my 3 "faulty" devices are offline. Unfortunately, I didn't catch the output when they went out. Never happened before 20221105. One of the devices is still on 20221224 and running. Still a mystery, maybe some sort of buffer issue. I'll again try to capture a USB output during failure

[16 more replies]

closed 2024-01-12 by ghtester