letscontrolit/ESPEasy #4338: NeoPixel (Basic)

fly74, 2022-11-06:
Hi @tonhuisman is it possible to put a brightness option to the NeoPixel (Basic) plugin as in the other neopixel plugins?

tonhuisman, 2022-11-06:
I'll see if that's possible.

fly74, 2022-11-08:
@tonhuisman On last question. Is it possible to get a command for brightness (like nfx,dim,X in BusFX) to set it in the basic plugin without setting the RGB values again?

tonhuisman, 2022-11-08:
I added a `NeoPixelBright,<level>`, and [documented](https://github.com/letscontrolit/ESPEasy/pull/4340/files#diff-68ec10eda82ae44162eab64c909648234b4e5a1a295b4b4aacce34babedb05f4R48-R54) it 😄

fly74, 2022-11-08:
Can make a binary für ESP8266 normal 4M1?

tonhuisman, 2022-11-08:
I prepared one yesterday, https://github.com/letscontrolit/ESPEasy/issues/4332#issuecomment-1306129774

fly74, 2022-11-08:
@tonhuisman Holy sh.. it works!

TD-er, 2022-11-08:
So this was fixed and built before you knew you needed it...
So Ton's programming skills went up to the level of predictive coding :)

tonhuisman, 2022-11-08:
> So this was fixed and built before you knew you needed it... So Ton's programming skills went up to the level of predictive coding :)

Not exactly, I just saw the request and added it, then provided it in the other issues as a test-build 😆 (it's mentioned there)

fly74, 2022-11-08:
@TD-er Ton owns a timemachine 😂

fly74, 2022-11-08:
@tonhuisman I need once more from this build ESP8266 neopixel 4M1.

tonhuisman, 2022-11-08:
No problem, though you can also download the `binaries.zip` from the Github Actions latest [P038 PR build](https://github.com/letscontrolit/ESPEasy/actions/runs/3419498632).
[ESP_Easy_mega_20221108_neopixel_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/9964449/ESP_Easy_mega_20221108_neopixel_ESP8266_4M1M.zip)

TD-er, 2022-11-08:
> @TD-er Ton owns a timemachine 😂

Not yet... but apparently in the future he will :)

fly74, 2022-11-08:
> Not yet... but apparently in the future he will :)

But maybe he's from the future and now "back in time" :)

PR #4340 (2022-11-09): [P038] Add initial and max brightness settings and command, update docs, optimizations
Resolves #4338 

[P038] Output - NeoPixel (Basic)
- Add settings for Initial and Max brightness
- Add `NeoPixelBright[,<brightness>]` command, range 0..255, where 0 sets the Initial brightness
- Updated documentation

TODO:
- [x] Testing by requester (@fly74) ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4338#issuecomment-1307011443))
closed 2022-11-09 by TD-er