letscontrolit/ESPEasy #4345: Batchupdate multiple ESPs

fly74, 2022-11-08:
Hi developers,

I want to share a little bash script to update multiple ESP to multiple versions at once. Long time ago I wrote with @TD-er about it.
My solution is very simple to use.

The run looks like this:

[image]

You need a folder structure like this

```
├── ESP32-D0WDQ6_neopixel_ESP32_4M316k
│   └── ESP_Easy_mega_20221107_neopixel_ESP32_4M316k.bin
├── ESP32-D0WDQ6_normal_ESP32_4M316k
│   └── ESP_Easy_mega_20221107_normal_ESP32_4M316k.bin
├── ESP8266_neopixel_ESP8266_4M1M
│   └── ESP_Easy_mega_20221108_neopixel_ESP8266_4M1M.bin
├── ESP8266_normal_ESP8266_4M1M
│   └── ESP_Easy_mega_20221107_normal_ESP8266_4M1M.bin
├── ESP.csv
└── ESP.sh
```

The csv-file must have this format (column1 the DNS-Hostname, column2 the foldername for the target binary):

ESPXX;ESP8266_normal_ESP8266_4M1M
ESPXX1;ESP8266_normal_ESP8266_4M1M
ESPXX2;ESP8266_neopixel_ESP8266_4M1M
ESPXX3;ESP32-D0WDQ6_normal_ESP32_4M316k
...

The script get the actual version from the json of each ESP (since the version the json provide .System."Binary Filename") and compares it to the version which is in the folder the csv points on. If it differs it uploads the new version via curl to update.

That's the script:

`
#! /bin/bash
while IFS=";" read -r dnshostname firmware
do
  p=$(echo $firmware | tr -d '\r')
  i=$(ls -AU $p | head -1 | tr -d '\r')
  ip=$(host $dnshostname | awk '{ print $4 }')
  j=
  j=$(curl --connect-timeout 3 -s http://$dnshostname/json | jq --raw-output '.System."Binary Filename"')

if [ "$i" != "$j.bin" ]
then

if [ -z "$j" ]
then
	#echo "$dnshostname offine"
        printf '%-15s' "$dnshostname"
        printf '%-15s' "offline"
	echo ""

else
	#echo "\nupdate $dnshostname"
	printf '%-15s' "$dnshostname"
        printf '%-15s' "update to"
        printf '%-50s' "$i"
	printf "IP: $ip"
	echo ""

curl --progress-bar --connect-timeout 3 -F "file=@$p/$i" http://$dnshostname/update
        echo "\n"
fi
else
	#echo "$dnshostname schon aktuell $j"
	printf '%-15s' "$dnshostname"
	printf '%-15s' "up to date"
	printf '%-50s' "$j"
        printf "IP: $ip"
	echo ""
fi
done < ESP.csv
`

So it is easy to put all the new versions to it's folders and update all to the correct build at once. Some thoughts are welcome.

tonhuisman, 2022-11-08:
Great script.

A few minor remarks:
- The script part can be wrapped in triple-backticks (instead of the single-backticks) to have it look like a script, and if you add `sh` after the first triple-backtick, it will use Bash syntax coloring (that's a markdown feature)
```sh
echo "Hello world"
```
- It might be helpful to add a comment that the `jq` command needs to be installed separately

fly74, 2022-11-08:
> A few minor remarks

Should look better now.

TD-er, 2022-11-08:
> Some thoughts are welcome.

Cool :)

One idea...
You could see all other nodes also being listed in the json.
So maybe also an idea to loop over those IPs and create a list of what's available on the network. Just as a starting point for the CSV.

fly74, 2022-11-08:
> You could see all other nodes also being listed in the json.

What do you mean? The json shows only information about itself in my case.

tonhuisman, 2022-11-08:
If you have the UDP Port set on the Tools/Advanced page (current default for new installs is IANA registered port 8266), the json will include all 'neighbors' on the same ESPEasy P2P network.

fly74, 2022-11-08:
Yes, but not all of my nodes are in the P2P. I think you should know what nodes are running. But I understand what you mean.

fly74, 2022-11-08:
But maybe a list of hosts would be enough. The binary variant could be extracted from json and then looking for a new fitting binary version.

TD-er, 2022-11-09:
> But maybe a list of hosts would be enough. The binary variant could be extracted from json and then looking for a new fitting binary version.

Yep, that's what I meant.
Just to generate a 'starting point' for new users.

I think your script would be great to be posted in the docs and/or  the misc folder.

fly74, 2022-11-09:
@TD-er can we provide a json property wich combines 

"Binary Filename":"ESP_Easy_mega_20221108_neopixel_ESP32_4M316k",
"ESP Chip Model":"ESP32-D0WDQ6",

without version date to categorize them for a binary folder for the script? Without cutting in the bash script.

eg. "ESP32-D0WDQ6_neopixel_ESP32_4M316k" in this case

fly74, 2022-11-09:
Forget it. The binary filename without the date would be enough.

tonhuisman, 2022-11-09:
> "ESP Chip Model":"ESP32-D0WDQ6",

I don't fully understand why you would handle separate ESP32 chip revisions differently? ESPEasy doesn't have specific builds for those chips. There where, for a really short time, a few builds that enabled PSRAM for some 16M8M builds (that was and is already enabled for _all_ other ESP32 builds), but that distinction is not there any longer, and not specific to that chip revision either.

fly74, 2022-11-10:
@tonhuisman Understand. But can we make a json property like "ESP_Easy_mega_normal_ESP8266_4M1M" as a firmware category?

uzi18, 2022-11-22:
you can add also rules backup

small advice - should have this code in python as we need multiplatform tool

fly74, 2022-11-24:
Python is unfortunately not my skill. @TD-er could we have a json property to represent all the files shown in filebrowser (.dat/.bmp,..)? So we could extend the backup to save rules and all other files separately.

TD-er, 2022-11-24:
Sure, that sounds like a good idea.

closed 2025-04-24 by fly74