letscontrolit/ESPEasy #4350: ESP8266's on TP-Link Mesh WIFI Network - Intermittent - HTTP: C001 Connection Failed

Humanoid-XXX, 2022-11-10:
### Checklist
- [X] The title of this issue is "to the point" and descriptive.
- [X] This issue describes what is happening.
- [X] This issue describes what components are affected (e.g. name of plugin/controller)
- [X] This issue describes how to reproduce it.
- [X] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [X] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [X] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [X] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [X] given a list of active plugins or controllers when applicable.
- [X] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [X] Tested previous/other build (mention which one already tested)- Tested random ones
- [X] Tested on other node to make sure hardware isn't defective. - Tested on 3 units
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request
Ever since we switched to a TP-Link Deco S4 Mesh Wifi system we have periodic issues with our ESPs communicating with our domoticz server. 
-It sometimes doesnt send data or recieve commands through domoticz but we are able to visit the ESP's webpage just fine on computer.
Logs show HTTP: C001 Connection Failed and HTTP Code: -1

### Steps to reproduce
Use TP-Link Deco S4 Wifi Mesh with HTTP Domoticz

Happens intermittently, not sure how to reproduce but it happens dozens of times a day.

Most times it does persist after cycling power and a wifi network reboot is usually what fixes it. 

We are using the MCP23017 and SHT3X Plugins on these units.
We havent had as much of an issue on the units we use without MCP230177 and SHT3X (Especially SHT3x) 
But we still have had issues at times with other units not using these plugins. 

Hardware:
D1 Wemos  ESP8266

ESP Easy version: ESP_Easy_mega_20220809_collection_A_ESP8266_4M1M

ESP Easy settings/screenshots: 
[image]

### Rules or log data

```
Build:⋄
20116  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20220809_57b0c0c
Plugin Count:⋄
79 [Normal] [Collection] [Collection_A]
Build Time:⋄
Aug  9 2022 09:42:55
Binary Filename:⋄
ESP_Easy_mega_20220809_collection_A_ESP8266_4M1M
Build Platform:⋄
Linux-5.10.102.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
Git HEAD:⋄
mega-20220809_57b0c0c

```

TD-er, 2022-11-10:
Given you mentioned "WiFi Mesh", I strongly suggest to set the WiFi on the ESP to "B/G only" mode.
This will fix the typical mesh issues most of the times.
Apart from that, there was an issue with the HTTP controllers, which is fixed in the latest release build from 5 days ago.

TD-er, 2022-11-14:
Any update in testing with a newer build?
Last night I merged another fix in making HTTP calls, also used in C001.
It was sending more in the HTTP header than needed (or even allowed) when no password was set.
Not sure if that was also causing issues in your setup, but you may want to test using the latest code.

Humanoid-XXX, 2022-11-29:
I tested with b/g mode only and it reduced the issue 95% but it still
experiences the same HTTP Error occasionally.
We do have a password set, so not sure if that was the issue but we will
try the newest build and report back.

TD-er, 2022-11-29:
What build did you test with?

Humanoid-XXX, 2022-11-29:
Tested with the latest one: 11/5

[image]
[image]