letscontrolit/ESPEasy #4351: I2C Bus Drops out occasionally when using SHT3X and MCP23017

Humanoid-XXX, 2022-11-10:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable. -SHT3X and MCP23017
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested) (Tried many)
- [x] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

We have 3 ESP8266's that use 2 SHT3X sensors and MCP23017 and ocassionaly the I2C bus drops out and the sensors read NAN and the MCP23017 status read -1. The only way i have fixed is by cycling power. 

At first i thought it was a bus capacitance issue because i am using the SHT3X sensors over a 10ft cable but an oscilliscope read looks ok. I also lowered bus speed to 1000 hz to counter this.

When the error occurs the oscilliscope just flat lines.

The issue seems to be particularly tied to the SHT3X sensor, I have a few other ESPs with the MCP23017 and theres no issue.

### Expected behavior
No bus drop out, reliable SHT3X readings.

### Actual behavior
i2c Bus drops out 

### Steps to reproduce
Intermittent issue,
2 SHT3x sensors reading every 5 seconds
MCP23017 reading statuses of relays that its triggering.
Sending HTTP data to Domoticz
I2C drops out dozens of times a day with a hard reboot the only fix. 

### System configuration

Hardware:
d1 mini wemos esp8266
dfRobot SHT31 sensor
ebay sht35 probe sensor

ESP Easy version: 
ESP_Easy_mega_20220809_collection_A_ESP8266_4M1M_VCC

ESP Easy settings/screenshots: 
[image]
[image]

### Rules or log data

```
Build:⋄
20116  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20220809_57b0c0c
Plugin Count:⋄
79 [Normal] [Collection] [Collection_A]
Build Time:⋄
Aug  9 2022 09:43:28
Binary Filename:⋄
ESP_Easy_mega_20220809_collection_A_ESP8266_4M1M_VCC
Build Platform:⋄
Linux-5.10.102.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
Git HEAD:⋄
mega-20220809_57b0c0c
```

TD-er, 2022-11-10:
Please try the special setting near the bottom of the tools->Advanced page:
"Try clear I2C bus when stuck"

It seems like one of the sensors is holding the SCL line low, which may be an indication one of the sensors wants to use "I2C clock stretching"
On the same "Advanced Settings" page, you can also set the "I2C ClockStretchLimit"
Sadly the exact unit for this is not the same on ESP8266 and ESP32, nor does the manufacturer specify what would be expected values. This means you may need to experiment a bit with these values.
I would suggest to start with values around 1500 - 2000.

Humanoid-XXX, 2022-11-11:
Thank you.
I have enabled  ""Try clear I2C bus when stuck"  and the problem is significantly reduced. Thank you so much! A huge relief. 
Is there any detriment to enabling this?

I do have 1000 ohm pullup resistors, I chose such a strong pull up to help with the long bus length but perhaps too strong.

I will investigate I2C clock stretching and report back.

I also want to mention that the SHT3X sensors have been amazing (Especially the ones with the teflon sensor barrier) and much more reliable in harsh conditions (100% humidity) compared to the BME280.

TD-er, 2022-11-11:
How long is "long"?
Not sure if there is an official max length on I2C, but I know there are sources classifying "40 cm" as "long" for I2C cabling.

Maybe you can 'twist' the cables like is done in Ethernet cables.
For example SDA & GND and SCL & GND each in a pair.
You can also try to experiment with placing the resistors close to the sensor, to force some current through the cables when the ESP is sending. Or higher resistance resistors on both ends (thus parallel) to make sure there is some current flowing when either end pulls the SDA or SCL down.

This process of clearing a stuck I2C bus is something I implemented as an experiment.
It does swap SCL and SDA pins and restarts the I2C wire class.
Then it does perform a scan and swaps the pins back + restart the I2C wire class.

The theory behind this is that some I2C slave device may have missed a clock pulse and thus is out of sync with the master.
So on a new clock pulse, the slave device may continue with its next bit, but the master may also try to pull down the SDA pin for sending something.
This corrupts all communication.
By swapping the signals, you get "data pulses" as clock, which is almost guaranteed not a multiple factor of 8 pulses and also the interval between ticks may be just long enough for the device to act on it.
This may allow the stuck device to get out of its state of being half-way in processing a byte.

Another reason why the I2C bus may get stuck is when a device is not yet ready to reply.
A slave device may then hold the clock line low.
However, the master must know what is an acceptible clock stretch limit to wait for the line to become "high" again.

tonhuisman, 2023-09-09:
This can probably be closed, as there hasn't been a reply in a long time, and the issue seems to be resolved.

closed 2023-09-09 by TD-er