letscontrolit/ESPEasy #4352: Release-20221105 Build fail with Arduino-Eesp32 2.0.5

khlabg22, 2022-11-10:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

A build with arduino-esp32 2.0.5 fail with several messages like:
src/DataStructs/PluginStats.h:158:27: error: 'taskVarIndex_t' has not been declared
src/DataStructs/PluginStats.h:159:28: error: 'taskVarIndex_t' has not been declared
/src/Helpers/ESPEasy_time.cpp:209:7: error: 'START_TIMER' was not declared in this scope
       START_TIMER;
src/Helpers/ESPEasy_time.cpp:282:18: error: 'SYSTIME_UPDATED' was not declared in this scope
       STOP_TIMER(SYSTIME_UPDATED);
and so on.

src/Helpers/ESPEasy_time.cpp:899:32: error: 'concat' was not declared in this scope
     addLogMove(LOG_LEVEL_INFO, concat(
and so on

/home/espeasy/arduino-1.8.16/portable/sketchbook/ESPEasy/src/Helpers/StringProvider.cpp: In function 'String getValue(LabelType::Enum)':
/home/espeasy/arduino-1.8.16/portable/sketchbook/ESPEasy/src/Helpers/StringProvider.cpp:352:59: error: 'useStaticIP' was not declared in this scope
     case LabelType::IP_CONFIG:              return String(useStaticIP() ? getLabel(LabelType::IP_CONFIG_STATIC) : getLabel(
                                                           ^~~~~~~~~~~
/home/espeasy/arduino-1.8.16/portable/sketchbook/ESPEasy/src/Helpers/StringProvider.cpp:352:59: note: suggested alternative: 'WiFiUseStaticIP'
     case LabelType::IP_CONFIG:              return String(useStaticIP() ? getLabel(LabelType::IP_CONFIG_STATIC) : getLabel(
                                                           ^~~~~~~~~~~
                                                           WiFiUseStaticIP

### Expected behavior

Able to make build without any issue

### Actual behavior

build fail

some header are not included.

### Steps to reproduce

1. make a build of latest release with arduino-esp32 2.0.5
2. 
3. 

### System configuration

Hardware:
Adafruit ESP32 Feather
LOLIN D32
TTGO T-Display

ESP Easy version: 
ESPEasy-mega-20221105

Attached a zipped patch to fix the issue.

[ESPEasy-mega-20221105-arduino-ide.diff.gz](https://github.com/letscontrolit/ESPEasy/files/9983791/ESPEasy-mega-20221105-arduino-ide.diff.gz)

```

TD-er, 2022-11-10:
Those have already been fixed.
To be honest, I have no idea why/how the GH Actions build managed to complete with these.

If you fetch the latest code in the `mega` branch, you can again compile it.

TD-er, 2022-12-23:
Is this still an issue?
Please also check the ESP32 SDK 2.0.6 PR: https://github.com/letscontrolit/ESPEasy/pull/4429

tonhuisman, 2023-09-09:
Compiling the ESPEasy code using Arduino IDE is not endorsed, better use VSCode/PlatformIO as [documented](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html).
This issue can most likely be closed, as also the original issue was already resolved.

closed 2023-09-09 by TD-er