letscontrolit/ESPEasy #4355: ESPEasy sends header data with 'Authorization: Basic Og=='

roobbb, 2022-11-11:
Hello.

I have running devices with older releases of ESPEasy and one with a fresher version. 
It seems the newer ESPEasy sends data to my controller with different content compared to my older devices.

`Authorization: Basic Og==` 
Is now included. I do not have any auth or credentials set. 
**Is this an intended behavior?**

Thank you and kind regards
roobbb

---
affected: controller
how to reproduce it: let a device with current version send its data by an interval; capture raw data  nc -l 8383

not affected version: mega-20190116 (ESP_Easy_mega-20190116_normal_ESP8285_1024.bin)
affected version: mega-20220809_57b0c0c (ESP_Easy_mega_20220809_normal_ESP8266_4M1M) till current

### Steps already tried...
- [n/a ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [X] Tested previous/other build (mention which one already tested)
- [X] Tested on other node to make sure hardware isn't defective.
- [X] Verified if the problem is limited to a single plugin/controller

---
[image]

---
example data
new version:
```
POST /ESPEasy HTTP/1.1
Host: 192.168.x.x:8383
User-Agent: ESP Easy/20116/Aug  9 2022 10:01:11
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Authorization: Basic Og==
Connection: keep-alive
Content-Length: 449

{"module":"ESPEasy","version":1.04,"data":{"ESP":{"name":"GasCounter","unit":0,"version":2,"build":20116,"build_notes":" - Mega","build_git":"mega-20220809_57b0c0c","node_type_id":17,"sleep":0,"ip":"192.168.x.x"},"SENSOR":{"0":{"deviceName":"GasCounter","valueName":"Count","type":6,"value":4},"1":{"deviceName":"GasCounter","valueName":"Total","type":6,"value":31412},"2":{"deviceName":"GasCounter","valueName":"Time","type":6,"value":86020}}}}
```
older version:
```
POST /ESPEasy HTTP/1.1
Content-Length: 291
Host: 192.168.x.x:8383
User-Agent: ESP Easy/20103/Jan 16 2019 03:12:25
Connection: close

{"module":"ESPEasy","version":"1.04","data":{"ESP":{"name":"Sonoff03","unit":0,"version":2,"build":20103,"build_notes":" - Mega","build_git":"mega-20190116","node_type_id":17,"sleep":0,"ip":"192.168.x.x"},"SENSOR":{"0":{"deviceName":"LED_green","valueName":"key","type":10,"value":"1"}}}}
```

TD-er, 2022-11-11:
> `Authorization: Basic Og==`
> Is now included. I do not have any auth or credentials set.
> **Is this an intended behavior?**

Nope, don't think so. Will look into this.

TD-er, 2022-11-11:
I had no proper webserver setup from which I could capture the full headers. So it was not actually tested here.
Can you try this test build when it is ready? https://github.com/letscontrolit/ESPEasy/actions/runs/3448559019

roobbb, 2022-11-12:
Thank you very much for you super fast response and fix 😀

I've flashed ESP_Easy_mega_20221111_normal_ESP8266_4M1M.bin on my test device (Wemos D1 mini).
And caught the raw data via nc (set controller to my laptop's IP):
`nc -l 8383`

This is what I caught:
```
POST /ESPEasy HTTP/1.1
Host: 192.168.x.x:8383
User-Agent: ESP Easy/20285/Nov 11 2022 23:38:02
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: close
Content-Length: 431

{"module":"ESPEasy","version":1.04,"data":{"ESP":{"name":"Watercounter","unit":1,"version":2,"build":20285,"build_notes":" - Mega","build_git":"HEAD_c4cf52d","node_type_id":17,"sleep":0,"ip":"192.168.x.x"},"SENSOR":{"0":{"deviceName":"Sensor","valueName":"Count","type":6,"value":0.00},"1":{"deviceName":"Sensor","valueName":"Total","type":6,"value":0.00},"2":{"deviceName":"Sensor","valueName":"Time","type":6,"value":0.00}}}}
```
My config for the controller is the same as I have on my other devices (no credentials / basic auth set for controller). Looks good 😎 😎

I even tried the opposite case: set credentials for basic auth. This is what I've captured:
```
POST /ESPEasy HTTP/1.1
Host: 192.168.x.x:8383
User-Agent: ESP Easy/20285/Nov 11 2022 23:38:02
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Authorization: Basic dGVzdHVzZXI6dGVzdHBhc3N3b3Jk
Connection: close
Content-Length: 431

{"module":"ESPEasy","version":1.04,"data":{"ESP":{"name":"Watercounter","unit":1,"version":2,"build":20285,"build_notes":" - Mega","build_git":"HEAD_c4cf52d","node_type_id":17,"sleep":0,"ip":"192.168.x.x"},"SENSOR":{"0":{"deviceName":"Sensor","valueName":"Count","type":6,"value":0.00},"1":{"deviceName":"Sensor","valueName":"Total","type":6,"value":0.00},"2":{"deviceName":"Sensor","valueName":"Time","type":6,"value":0.00}}}}
```
Basic auth enabled is still working fine.
So I turned basic auth off again, and the header doesn't contain any auth-string. 

Perfect 😎 😎 - issue seems to be solved

Thanks a lot and best regards
roobbb

TD-er, 2022-11-12:
I tried to capture it on Apache, by using the "forensic" module.
I could not capture the header you showed, also on a node running the build you mentioned.
So therefore I was not able I fixed it :)
Never thought of `nc`. (doh!)
In the current network setup I have for my network I was not able to mirror a switch port to my PC, simply because the last switch to my PC isn't "managed", so I could not run Wireshark. Maybe I should invest some time in setting up some node to capture data.

Anyway, glad to know it is now fixed :)

closed 2022-11-14 by TD-er

TD-er, 2022-11-17:
@roobbb Can you check if with this PR: Can you check this PR: https://github.com/letscontrolit/ESPEasy/pull/4365 your setup still works?
I may have "fixed" (read: broken) too much when dealing with this issue, thus causing new issues.
So if you can test this PR, that would be great!

roobbb, 2022-11-17:
Of course. 
First case, no basic auth:
```
POST /ESPEasy HTTP/1.1
Host: 192.168.x.x:8383
User-Agent: ESP Easy/20291/Nov 17 2022 12:33:53
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: close
Content-Length: 431

{"module":"ESPEasy","version":1.04,"data":{"ESP":{"name":"Watercounter","unit":1,"version":2,"build":20291,"build_notes":" - Mega","build_git":"HEAD_c488e2d","node_type_id":17,"sleep":0,"ip":"192.168.x.x"},"SENSOR":{"0":{"deviceName":"Sensor","valueName":"Count","type":6,"value":0.00},"1":{"deviceName":"Sensor","valueName":"Total","type":6,"value":0.00},"2":{"deviceName":"Sensor","valueName":"Time","type":6,"value":0.00}}}}
```
Still looking good  😀

Next case, basic auth enabled:
```
POST /ESPEasy HTTP/1.1
Host: 192.168.x.x:8383
User-Agent: ESP Easy/20291/Nov 17 2022 12:33:53
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Authorization: Basic dGVzdHVzZXI6dGVzdHBhc3N3b3Jk
Connection: close
Content-Length: 431

{"module":"ESPEasy","version":1.04,"data":{"ESP":{"name":"Watercounter","unit":1,"version":2,"build":20291,"build_notes":" - Mega","build_git":"HEAD_c488e2d","node_type_id":17,"sleep":0,"ip":"192.168.x.x"},"SENSOR":{"0":{"deviceName":"Sensor","valueName":"Count","type":6,"value":0.00},"1":{"deviceName":"Sensor","valueName":"Total","type":6,"value":0.00},"2":{"deviceName":"Sensor","valueName":"Time","type":6,"value":0.00}}}}
```
Even looks good 😀

Last case, switch back to basic auth disabled:
```
POST /ESPEasy HTTP/1.1
Host: 192.168.x.x:8383
User-Agent: ESP Easy/20291/Nov 17 2022 12:33:53
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Connection: close
Content-Length: 431

{"module":"ESPEasy","version":1.04,"data":{"ESP":{"name":"Watercounter","unit":1,"version":2,"build":20291,"build_notes":" - Mega","build_git":"HEAD_c488e2d","node_type_id":17,"sleep":0,"ip":"192.168.x.x"},"SENSOR":{"0":{"deviceName":"Sensor","valueName":"Count","type":6,"value":0.00},"1":{"deviceName":"Sensor","valueName":"Total","type":6,"value":0.00},"2":{"deviceName":"Sensor","valueName":"Time","type":6,"value":0.00}}}}
```
seems to work fine 😀 😎

Kind regards
roobbb

TD-er, 2022-11-17:
OK, but apparently the other issue isn't fixed yet, so I'll take another attempt to break this one.. ehh fix the other one ;)
I'll get back to you about this.
Thanks for testing so far by the way.

roobbb, 2022-11-17:
No problem at all 😄. Please let me know when I should start the test.