letscontrolit/ESPEasy #4360: [Blynk] Update Blynk library (+merge ESPEasy code)

TD-er, 2022-11-14:
The [Blynk library](https://github.com/blynkkk/blynk-library) we use is outdated.
The latest Blynk release also adds support for the newer ESP32-S2 and ESP32-C3.
However for ESPEasy we made some changes to allow for defining a callback function.

See: https://github.com/letscontrolit/ESPEasy/commit/844915618f60f74405f93b3e0d79034d83a48e90#diff-d6a5c863b4326527d3a98eba149767670c8f5f7504cba4a8efef40c5a8956b68

These have to be re-incorporated into the code again (or maybe made into a pull req. to the library maintainer)

Sminter4, 2023-05-03:
Hello, i would also like to use Blynk.
Unfortunately, i can no longer find it in any current ESPEasy version. Apparently it should be under "collection":
[https://espeasy.readthedocs.io/en/latest/Controller/_Controller.html](https://espeasy.readthedocs.io/en/latest/Controller/_Controller.html)

Or is the possibility of sending data to Blynk completely lost?

tonhuisman, 2023-05-06:
Blynk is disabled in current builds, because of size, so to include it you could create a Custom build.

How to create a Custom build:
- Set up PlatformIO (PIO) as [documented here](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html)
- Copy file `src/Custom-sample.h` to `src/Custom.h` (mind the uppercase C!)
- Edit that `Custom.h` file by un-commenting or changing the settings, features, plugins and controllers as needed
- Select one of the `_custom_` environments in PIO to build (output .bin files can be found in `build_output/bin`)
- (if you require IR support, create a copy of `CustomIR-sample.h` to `CustomIR.h` and configure there, and select one of the `_custom_IR_` PIO environments to build)

NB: For Blynk support you will need Controllers C012 or C015.
NB2: Please be aware we only have older versions of the Blynk libraries incorporated in ESPEasy, as noted above, that still need to be updated, as Blynk seem to have changed their API or expected urls. (Only on ESP32 it is possible to have SSL support.)