letscontrolit/ESPEasy #4361: Add support for Ikea Vindriktning

TD-er, 2022-11-14:
Ikea link: https://www.ikea.com/nl/nl/p/vindriktning-luchtkwaliteitsensor-70498242/
Some blog post describing the sensor: https://style.oversubstance.net/2021/08/diy-use-an-ikea-vindriktning-air-quality-sensor-in-home-assistant-with-esphome/
Forum topic: https://www.letscontrolit.com/forum/viewtopic.php?p=60972#p60972

tonhuisman, 2022-11-14:
Found a link to the [datasheet](http://www.jdscompany.co.kr/download.asp?gubun=07&filename=PM1006_LED_PARTICLE_SENSOR_MODULE_SPECIFICATIONS.pdf) (English content) that also briefly describes the particle data it will return, and the command to send to get it send that data.

manjh, 2022-11-16:
tried to open the datasheet, but "cannot be downloaded safely"...

TD-er, 2022-11-16:
Seems like it is "signed" but not with a commonly known CA.
[PM1006_LED_PARTICLE_SENSOR_MODULE_SPECIFICATIONS_printToPDF.pdf](https://github.com/letscontrolit/ESPEasy/files/10025726/PM1006_LED_PARTICLE_SENSOR_MODULE_SPECIFICATIONS_printToPDF.pdf)

This is the document printed to PDF, so it removes those, including any scripts if they were present.

manjh, 2022-11-16:
Got it. 
Another bit of info: while snooping around for datasheets, I noticed that there is a PM1006, but also a PM1006K.
Not sure what the exact difference is.

TD-er, 2022-11-16:
That was also mentioned in the blog.
The IKEA one only reports 2.5 micron particles.

TD-er, 2022-11-16:
Well that's a very extremely limited protocol description in the datasheet:

[image]

flashmark, 2022-11-16:
Hi, I am playing around with ESPeasy for some time and wrote a plugin for the Vindriktning. I am still unsure how to share this plugin. I opened the box and added a Wemos D1. Hooked it up to power and the Tx of the sensor. The Vindriktning CPU is still doing the polling, but the Wemos is snooping the answer.

TD-er, 2022-11-16:
@flashmark Can you share the relevant code here?

flashmark, 2022-11-16:
Not sure how to do this. First time using github.

TD-er, 2022-11-16:
copy paste (not too large) chunks of code can be pasted in a message here.
If you prepend it with 3 backticks and also 3 backticks at the end, GitHub will consider it as a code block

flashmark, 2022-11-16:
Main trick, borrowed from others
///
// Function P240_process_rx
// Handles a single character received from the PM1006 in the Vindrikning
// It expects a response to the poll message sent by the Vindriknings own CPU
// 0x16 <len> <data> <chksum>
// <len>  = size of <data> in char [1 char]
// <data> = message payload, array of <len> chars
// <csum> = checksum, 256-(sum of all received characters) [1 char]
bool P240_process_rx(char c)
{
    switch (rxState) {
    case PM1006_HEADER:
        rxChecksum = c;
        if (c == 0x16) {
            rxState = PM1006_LENGTH;
        }
        break;

    case PM1006_LENGTH:
        rxChecksum += c;
        if (c <= bufferSize) {
            _rxlen = c;
            _index = 0;
            rxState = (_rxlen > 0) ? PM1006_DATA : PM1006_CHECK;
        } else {
            rxState = PM1006_HEADER;
        }
        break;

    case PM1006_DATA:
        rxChecksum += c;
        serialRxBuffer[_index++] = c;
        if (_index == _rxlen) {
            rxState = PM1006_CHECK;
        }
        break;

    case PM1006_CHECK:
        rxChecksum += c;
        rxState = PM1006_HEADER;
#ifdef PLUGIN_240_DEBUG
        P240_dump();
#endif
        return ((rxChecksum&0xFF) == 0);

    default:
        rxState = PM1006_HEADER;
        break;
    }
    return false;
}
///

TD-er, 2022-11-16:
Do you also have the defines you use in the switch statement?

And backticks are these: `
You're using slashes.

flashmark, 2022-11-16:
I am using a simple state machine as parser of the incoming characters. Characters are gathered "ten per second' 
'''
typedef enum {
    PM1006_HEADER,
    PM1006_LENGTH,
    PM1006_DATA,
    PM1006_CHECK
} pm1006_state_t;
'''
And some global storage
```
// Global variables TODO: attach them to the task to allow multiple instances
  ESPeasySerial *easySerial = nullptr;
  const int bufferSize = 20;
  char serialRxBuffer[bufferSize];
  char debugBuffer[3*bufferSize];
  int rxBufferPtr = 0;
  int rxChecksum = 0;
  int _index = 0;
  int _rxlen = 0;
  int pm25 = 0;
  pm1006_state_t rxState = PM1006_HEADER;
```

flashmark, 2022-11-16:
Plugin is around 375 lines of code, half of it comment. Main issue to move the global variables in something that is bound to the plugin instance. The sensor is working, but a bit boring at my home with only high rates when you fool it with burning matches. I should try to upload it to my github clone

TD-er, 2022-11-16:
I took the liberty to edit your posts to show the correct backticks.
If you press edit on your posts, you can see it.

flashmark, 2022-11-16:
Thanks, learned new tricks today :-) Next is to share the code through github. I took the free P240, used the template and reverse engineered serial link sensors like MHZ19.

tonhuisman, 2022-11-16:
We have a Developers guide, [here](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html)

flashmark, 2022-11-16:
Missing part:
```c++
 case PLUGIN_TEN_PER_SECOND:
    {
      // code to be executed 10 times per second. Tasks which require fast response can be added here
      // be careful on what is added here. Heavy processing will result in slowing the module down!
      bool new_data = false;
      while ((easySerial->available() > 0) && !new_data) 
      {
        new_data = P240_process_rx(easySerial->read());
        if (new_data) 
        {
          pm25 = float((serialRxBuffer[3] << 8) + serialRxBuffer[4]);
          String log = F("P240 : Data ");
          log += pm25;
          addLogMove(LOG_LEVEL_INFO, log);
        }
      }
      success = true;
    }
```

flashmark, 2022-11-16:
@tonhuisman Will pick it up coming days.

TD-er, 2022-11-16:
> Missing part:
> 
> ```c++
>  case PLUGIN_TEN_PER_SECOND:
>     {
>       ...
>       success = true;
>     }
> ```

I think you're also missing a `break;` statement here.. (it might have been just the next line after the `}` )

flashmark, 2022-11-17:
Thanks, break was missing, but not affecting the functionality. Tried a pull request first time, not sure the code fulfills all requirements. I don't have reviewers at home.

fly74, 2022-11-17:
I'm interested too to use the Vindriktning :)

manjh, 2022-11-17:
There is a lot of interest for this device, in several app-groups that I am part of.
Not surprising, since it is relatively cheap to buy and offers a nice looking case for those extra sensors...
Hans Man

uzi18, 2022-11-18:
@TD-er PM1006K : 
[image]

TD-er, 2022-11-18:
Thanks. It shows at least it shares its protocol with the PM1006.
So I guess we can make the plugin also support the PM1006K and make a selector indicating which values we need to use.

flashmark, 2022-11-18:
Note that the plugin I wrote is dedicated to the Ikea Vindriktning. The original Ikea CPU is talking to the sensor and driving the three LEDs. The plugin is listening only to the answers of the sensor and translates them to ESPeasy plugin values. If you wish to take control the Ikea CPU shall be removed and the plugin also has to send queries the sensor. I think this is easily possible as long as the protocol is indeed as given above.

fly74, 2022-11-20:
Here a german article about the inside [https://www.heise.de/ratgeber/Ikea-Feinstaubsensor-Vindriktning-zum-IoT-Device-aufbohren-6164149.html](https://www.heise.de/ratgeber/Ikea-Feinstaubsensor-Vindriktning-zum-IoT-Device-aufbohren-6164149.html)

fly74, 2022-11-24:
Is there an action build for testing ?

tonhuisman, 2022-11-24:
I think I've pointed you to the Actions tab before 🤔, the PR is named `Add p240` (I already suggested it to be renamed), a fresh build (2 actually) is being built atm.

fly74, 2022-11-24:
> Hi, I am playing around with ESPeasy for some time and wrote a plugin for the Vindriktning. I am still unsure how to share this plugin. I opened the box and added a Wemos D1. Hooked it up to power [...]

@flashmark Can you describe what the wiring between D1 and Vindriktning has to be?

TD-er, 2022-11-24:
I renamed the PR, so it is easier to find it in the GH Actions builds.
I also approved the latest changes to be built by GH Actions.

fly74, 2022-11-25:
https://github.com/letscontrolit/ESPEasy/actions/runs/3543242951 which image has the plugin? Cannot find it in

ESP_Easy_mega_20221124_climate_ESP32_4M316k.bin or
ESP_Easy_mega_20221124_normal_ESP32_4M316k.bin

tonhuisman, 2022-11-25:
Don't think it is included in a regular build yet, so I've created a Custom build that includes it:

[ESP_Easy_mega_20221125_custom_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/10094641/ESP_Easy_mega_20221125_custom_ESP32_4M316k.zip)

fly74, 2022-11-25:
@tonhuisman Thx. I play a little around with it. Do you know what the wiring has to be?

tonhuisman, 2022-11-25:
I just copied the wiring instructions from the blog post, referenced in the tom message of this issue:
```
    The wire from the VINDRIKTNING’s +5V pad goes to the D1 mini’s 5V pin.
    The wire from the VINDRIKTNING’s GND pad goes to the D1 mini’s GND pin.
    The wire from the VINDRIKTNING’s REST pad goes to the D1 mini’s D2 pin.
```
This talks about a Wemos D1 mini, but it also states just about any GPIO pin can be used to connect the `REST` signal to, you will have to configure that in the plugin, AFAICS. Only the receiving part of ESPEasy Serial is used for this plugin, currently.
That blog post also has photos to illustrate the process of opening the unit and connecting the wires.

fly74, 2022-11-25:
I put it on  a breadboard :)

fly74, 2022-11-27:
@tonhuisman What is the pin for the ESP32?

TD-er, 2022-11-27:
That's the same serial port selector as used all over ESPEasy.
Just first select the port (e.g. Serial1) which is not jet used, then select the pins you want.

flashmark, 2022-11-27:
I was a bit busy with other things and have some troubles with my Wemos-D1 in my Vindriktning. I reworked the plugin according to review comments, but still work to do. I will create a pull request. Coming week I am away, so any comments will be picked up later. With the latest version from Github my module does not connect with the WiFi anymore: Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 70 ms

I wired my Wemos-D1 as described 5V, GND and D2 (GPIO4) on the Wemos. The Vindriktning has solder pads as shown here
[image]
Connect D2 (GIO4) to REST.

fly74, 2022-11-27:
> I will create a pull request.

@tonhuisman can make then a new binary?

tonhuisman, 2022-11-27:
> @tonhuisman can make then a new binary? ESP32 and ESP8266?

I'd like to bring to your attention the [development guide](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#starter-guide-for-local-development-on-espeasy), as you seem capable enough to build yourself 😃 

[ESP_Easy_mega_20221127_custom_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/10098985/ESP_Easy_mega_20221127_custom_ESP32_4M316k.zip)

[ESP_Easy_mega_20221127_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/10098987/ESP_Easy_mega_20221127_custom_ESP8266_4M1M.zip)

fly74, 2022-11-28:
@tonhuisman @flashmark @TD-er Hey not bad a for a first try!

[image]

TD-er, 2022-11-30:
What serial port do you use on ESP8266?
Is "use serial" disabled on the tools->Advanced page when using Serial0 or Serial0_swapped ?

fly74, 2022-11-30:
Physically REST -> D4

and

[image]
[image]

TD-er, 2022-11-30:
Serial1 on ESP8266 is useless.
It can only send, not receive.
It also shows it in the description, "- <- TX" means the TX of the sensor is not connected to anything on the ESP
"GPIO-2 -> RX" means GPIO-2 on the ESP is connected to RX of the sensor.

It is read from left to right, from the viewpoint of the ESP.

fly74, 2022-11-30:
@TD-er  I'm a little confused. D4 = GPIO2 

[https://microcontrollerslab.com/wp-content/uploads/2019/05/ESP8266-12E-Wemos-D1-Mini-pinout.jpg](https://microcontrollerslab.com/wp-content/uploads/2019/05/ESP8266-12E-Wemos-D1-Mini-pinout.jpg)

TD-er, 2022-11-30:
You selected "HW Serial1"
The ESP8266 does have 2 HW serial ports:
- Serial 0 => RX & TX
- Serial 1 => TX only

This means HW serial1 on ESP8266 is utterly useless. (well, maybe for outputting logs it may be useful...)

HW Serial0 is default on GPIO 1 & 3 (I will not name the D-notation as I always have to check it as it makes no sense, even though boards like NodeMCU and Wemos D1 mini use it on their silk screen)
However, it is possible to re-assign pins GPIO 13 and 15 (no idea about their D-name equivalent)
This mode is called "HW Serial 0 Swapped" in ESPEasy.
N.B. GPIO 15 is a 'special' pin, so in order to use it for serial, you need to add a PNP transistor.

This means that if you want to connect the sensor to pins other than GPIO 1 & 3, you should select "SW Serial" and pick the right pins.

fly74, 2022-12-01:
@TD-er Hi Gijs, it's a little tricky and work now with:

[image]
[image]

But why the device must be set GPIO<TX and the devicelist shows it as RX: GPIO-4 (D2)?

TD-er, 2022-12-01:
I tried to make it as unambiguous as possible, so please tell me where it is inconsistent?
The labels have to be read from left to right, where on the left is the ESP related part and on the right the attached sensor.
Thus GPIO (ESP side) <- RX (sensor side)

Only thing I can think of for serial is to change the label into "GPIO (RX) <- TX"
The idea is, you're configuring it from the ESP's viewpoint.
So you need to define the GPIO and connected to it is some pin on the sensor side which happens to be "sending data" which we call TX.
What matters is that the ESP considers it as an RX port.

fly74, 2022-12-01:
I can live with it. But for me it would be the clearest to set an "GPIO (#) <- REST Pin Sensor". But don't let us splitting hairs :)

fly74, 2022-12-01:
I made a clip for fixing the ESP inside, if anyone is interested and able to print 3D.

[ESP Clip](https://www.printables.com/de/model/329613-ikea-vindriktning-esp8266-wemos-d1-clip)

[image]

fly74, 2022-12-01:
Another Question the values are only shown if the plugin interval is set less than 10 seconds. I'm using the generic http controller to push values to a database usually every 300 seconds. Can I use a rule to push values every 300s even if the sensor is getting it every 10s?

TD-er, 2022-12-01:
Hmm that sounds like a bug...
Anyway, what you can do to overcome this issue is to copy the value to a dummy task and then call taskrun on that dummy task (you need to assign the controller to the dummy task of course)

fly74, 2022-12-01:
> Hmm that sounds like a bug... Anyway, what you can do to overcome this issue is to copy the value to a dummy task and then call taskrun on that dummy task (you need to assign the controller to the dummy task of course)

Works great. THX.

fly74, 2022-12-02:
Here the LED values from the instruction manual

[image]

And it was for me better to connect the fan to the 3.3V ESP instead the 5V from the original board. The original connect turn on the fan every ca. 30s. It was annoying, so lower rpm with constant fan is very quiet without loss of accuracy.

TD-er, 2022-12-02:
> so lower rpm with constant fan is very quiet without loss of accuracy.

That does really surprise me.
I would expect that lower air flow would result in lower reported values for the dust particle count.
Also when the fan at some point will be a bit more dirty or worn down, it may need more power to start spinning. At 3V3 this may not be possible and then the fan will no longer spin up.

fly74, 2022-12-02:
I will have an eye on it, but the constant rpm change is annoying to me.

TD-er, 2022-12-02:
Maybe you can also experiment with its orientation? If the fan is slightly tilted, maybe the fan noise is less annoying?

fly74, 2022-12-02:
The annoying thing the constant spin up and spin down from the measure cycle.

TD-er, 2022-12-02:
Yep, I understand.
But parts of such noise can also be amplified when the fan isn't rolling optimal.
Thus tilting the fan/sensor slightly can maybe also reduce this noise.

[7 more replies]

PR #4366 (2023-03-23): Add support for Ikea Vindriktning (#4361)
Adding p144_Vindriktning plugin. Ikea AIR quality sensor with added ESP.

Fixes: #4361
closed 2023-03-23 by TD-er