letscontrolit/ESPEasy #4364: C011 POST  header problem

3dalex07, 2022-11-17:
in mega 41f60a2a.  then sending header information 
`Content-Type:application/json
Authorization:Token xxxxxxxxxxxxxxxxxxxxxxxx`
server answer POST... failed HTTP code: 401  Received reply: {"detail":"Authentication credentials were not provided."}
if i use mega-20220511 (5c7470c1).  all works OK

TD-er, 2022-11-17:
Hmm, this keeps giving head-aches...

What kind of authentication does your server require? Basic auth or Digest?
Is either your user name or password empty?

3dalex07, 2022-11-17:
[image]
my configs

TD-er, 2022-11-17:
Ah check!
That's something I would never have thought of.

3dalex07, 2022-11-17:
esp32

TD-er, 2022-11-17:
Can you check this PR: https://github.com/letscontrolit/ESPEasy/pull/4365 ?

TD-er, 2022-11-17:
Ah you're right... I needed some more coffee....

3dalex07, 2022-11-17:
dont helps

TD-er, 2022-11-17:
I don't get it....
See [here the commit](https://github.com/letscontrolit/ESPEasy/pull/4358/files) that caused this issue.
Apart from stripping extra spaces from credentials, that commit should have been reverted by these changes.

3dalex07, 2022-12-01:
replace to this file  helps [https://github.com/TD-er/ESPEasy/blob/bugfix/http_token_auth_header/src/src/Helpers/Networking.cpp](url)

TD-er, 2022-12-01:
You mean this PR? https://github.com/letscontrolit/ESPEasy/pull/4365

3dalex07, 2022-12-01:
yes

TD-er, 2022-12-01:
thanks for testing

commit d729d34a (2022-12-01): [HTTP] Restore Authorization headers used in C011 (#4364)

Fixes: #4364

Need to check if #4355 still works
closed 2022-12-01 by TD-er