letscontrolit/ESPEasy #4367: sendto not working anymore since probably mega-20221105

chromoxdor, 2022-11-18:
its me with a bug again... 

as you can see sendto is not working anymore unless the syntax was secretly changed

[image]

TD-er, 2022-11-18:
Not sure if that ever should have worked as the last argument should be what you're trying to send.
Thus something like this:
```
sendto,1,"event,toggle"
```

TD-er, 2022-11-18:
Oh and this is a typical example of why I added several types of quotes, so you could do this:
```
sendto,1,`sendto,2,"event,toggle"`
```
(just for fun :) )

chromoxdor, 2022-11-18:
ohhh..thanks a lot!
That is new for me. In the past it was definitely less strict. :)

TD-er, 2022-11-18:
Maybe you had checked the checkbox on the tools Advanced page to be less strict in parsing the last argument?

chromoxdor, 2022-11-18:
[image]

TD-er, 2022-11-18:
Ah, check, some really old documentation which needs to be updated

TD-er, 2022-11-18:
N.B. just to show how "old" this is: https://github.com/letscontrolit/ESPEasy/issues/2724

chromoxdor, 2022-11-18:
Jep, thats quite some time...
I also just realized, that there are (don’t know since when) html status codes for `SendToHTTP` which generate events.
That is awesome!!!!!! 🤩

PR #4371 (2022-11-21): [Docs] Improvements nov 2022
Resolves #4367

Features
- Improvements for `SendTo` and `SendToHTTP` commands using quotes where required
closed 2022-11-21 by TD-er