letscontrolit/ESPEasy #4369: [feature request] 12bit pwm

chromoxdor, 2022-11-18:
This request i am doing for my brother.
He asked me if it is possible to use 12bit pwm in espeasy. He wants to use it over a longer period of time so for smother transitions he wants to have more steps.
In general this is possible by limiting the frequency . I also found this: https://github.com/StefanBruens/ESP8266_new_pwm

So i relay the question to the professionals. Is this possible to  implement this into ESPeasy?

TD-er, 2022-11-18:
Sure it is... actually for servo's we already use a different resolution.
However higher PWM resolution does limit the max. frequency.
Not 100% sure, but I think the PWM resolution is global, so it then does affect all instances of PWM.

I have to think on how to implement it, as the current pwm command already has a lot of parameters and also setting something global via a command doesn't feel right.
Maybe I have to add a setting for it, so it is clear it is global.

chromoxdor, 2022-11-18:
> I have to think on how to implement it, 

That would be most appreciated! 🙂

chromoxdor, 2022-12-01:
I will close this because i found the corresponding entry in the code and changed it to make a special build for my brother.
And for me it is not so important, that this will find it´s way as an option to an official release.

closed 2022-12-01 by chromoxdor