letscontrolit/ESPEasy #4370: Rules: int-variable countdown into negative converts to float after 24 iterations

canufarm, 2022-11-18:
Hi,

try this rule with ESP8266 with current (and before) ESPeasy-firmware:

```
on System#Boot do
  Let,1,0
  LoopTimerSet,1,1
endon

on Rules#Timer=1 do
  Let,1,[var#1]-1
  LogEntry,'----############### [var#1]'
endon

```

Look at the logs:
```
...
629103: ----############### -22
630163: ----############### -23
631227: ----############### -24.00000000000001
632289: ----############### -25.00000000000002
633352: ----############### -26.00000000000003
...
```

I don't know if this is known.

TD-er, 2022-11-18:
What is the issue here? It suddenly showing those extra "noise bits" when printing it?
I have tried to mitigate this effect in the printing routine, but apparently I need to look into the epsilon factor here.

canufarm, 2022-11-18:
If one wants to compare these values this could be an issue with unpredictive results. I don't think this should be normal when only counting integer values.

tonhuisman, 2022-11-18:
If you require the value of the counter to be of type `int`, you can use `[int#1]` instead of `[var#1]`, that will convert correctly.

canufarm, 2022-11-18:
Yes, I know, but nevertheless there must be an issue. How can it be that integers are not straight on subtracting integers? What's the background? Even if these numbers were floats, they can not get this way by only adding round numbers.

TD-er, 2022-11-18:
Sure they can.
This is very common for floating point operations.
The only "issue" here may be that I convert floating point values to a string first and then parse it again to a floating point value.
This way such errors may manifest itself a bit earlier.
See for some examples: https://stackoverflow.com/questions/2100490/floating-point-inaccuracy-examples
The "amplification" mentioned in point 2 of the first answer is somewhat similar to what is happening here.

N.B. I did use a few tricks in the string formatting function to format floating point values to make this happen less often.
Still it isn't something I can prevent from happening.

It is a fact of handling floating point values on a computer.
Some calculators try to prevent this by using a completely different representation of floating point values. But this extra effort isn't worth the added complexity.

canufarm, 2022-11-18:
Ah, you are using floats instead of integers and you convert them several times? Now I understand. OK, than this seems not a valid bug.

TD-er, 2022-11-18:
Yep and internally the values are converted to strings before being processed in rules.
This way task values will also be formatted according to the set nr. of decimals before being processed.

canufarm, 2022-11-18:
OK, lets close it :) Thank you.

closed 2022-11-18 by TD-er