letscontrolit/ESPEasy #4373: Release-20221105 Generic - Pulse counter problem

tomatos-st, 2022-11-19:
I have a problem with the counter in ESP32 in the compilation ESP_Easy_mega_20221105_normal_ESP32_4M316k - it does not work on any port, I tried practically on every port (GPIO27,25,32,39,35,33,36) it just does not count.

Interestingly, on the previous version of ESP_Easy_mega_20220809_normal_ESP32_4M316k it works without any problems
also works on Release-20220809 on ESP8266 ESP_Easy_mega_20220809_normal_ESP8266_1M

I also noticed that it often resets when trying to use the counter.
In the previous build Release-20220809 it works stably.
I use ESP D1 mini WEMOS

TD-er, 2022-11-19:
That's correct and already fixed in the latest code.
There has not yet been an official new build including this fix, but you could try one of the latest GitHub Actions builds. For example this one: https://github.com/letscontrolit/ESPEasy/actions/runs/3461165253

TD-er, 2022-12-23:
This has been fixed, so will be included in the next official build.

closed 2022-12-23 by TD-er