letscontrolit/ESPEasy #4374: Generic - Dummy Device and TaskValueSet,TASKname,VARname,Value

tomatos-st, 2022-11-19:
In the latest build of ESP_Easy_mega_20221119_normal_ESP32_4M316k, TaskValueSet control via HTTP with (TaskValueSet,TASKname,VARname,Value) stopped working
After posting http://192.168.100.92/control?cmd=TaskValueSet,data,dat1,10
receives the message: NOT_A_DUMMY_TASK
if I enable the option Allow TaskValueSet on all plugins the message is: OK but it still does not work

Only TaskValueSet,TASKnr,VARnr,Value works
http://192.168.100.92/control?cmd=TaskValueSet,32,1,10

In ESP_Easy_mega_20221105_normal_ESP32_4M316k both methods worked

TD-er, 2022-11-19:
Have you also tried to call it from the rules using the taskname?
Just to make sure where to look for the bug.

tomatos-st, 2022-11-20:
The rules:
      TaskValueSet Data,dat1,10
       TaskValueSet 31,2,10

result:
log:
2264699: ACT  : TaskValueSet Data,dat1,10
2264703: ACT  : TaskValueSet 31,2,10

Devices:
[image]

TD-er, 2022-11-20:
OK, so the problem is not related to the HTTP source for commands.
Thanks for testing.
@tonhuisman Another use case where the cache of tasknames and taskvaluenames may be used wrong?

tomatos-st, 2022-11-20:
I haven't found it at the moment

chromoxdor, 2022-12-03:
I just stumbled over the same issue... it appears to only be an issue for the esp32... on an 8266 this works.

TD-er, 2022-12-04:
Should be fixed now... At least Ton was having a clear moment to fix my 2 year old bug...
I am curious to what other issues this may have caused.

PR #4398 (2022-12-04): [Command parsing] Non-numeric arguments could return uninitialized data
Fixes #4374 

Non-numeric command arguments, as can be used in the Task* commands, could return uninitialized data. This was detected on ESP32, but not (yet) on ESP8266. Initializing the used variable to 0 solves this.
closed 2022-12-04 by TD-er