letscontrolit/ESPEasy #4375: _P121_HMC5883L.ino doesn't send events

RoBra81, 2022-11-20:
### Summarize of the problem/feature request

I want to use _P121_HMC5883L.ino to count gas meter. Unfortunately it doesn't send events and values neither to the controller nor to the rules.

### Expected behavior

Rules are executed and values are sent to the controller.

### Actual behavior

Values are displayed and changing within devices view but don't execute rules and aren't sent to the controller. Other values (for example Generic - System) are sent to the controller.

### Steps to reproduce

1. Did clean install
2. configured controller
3. configured HMC5883L device including send to controller
4. configured Generic - System device

### System configuration

Hardware:
Wemos D1 mini

ESP Easy version: 
ESP_Easy_mega_20221105_collection_E_ESP8266_4M1M Nov 5 2022

ESP Easy settings/screenshots: 
[image]
[image]

tonhuisman, 2022-11-20:
Can you get any logging from the ESP?
Is the Sysinfo data received at the FHEM Controller?

RoBra81, 2022-11-20:
Where to find the logging info? 
Yes, Sysinfo data is received at the FHEM controller.

tonhuisman, 2022-11-20:
Logging can be obtained via the (USB) Serial port of many ESP boards, like the Wemos D1 Mini or NodeMCU, or via the Tools/Log page in the Web UI.
It seems that it might fail to send to FHEM if memory is too low, can you report the `Freemem` value in the `WD   : Uptime...` log line that is output every 30 seconds?

RoBra81, 2022-11-20:
4044252: WD : Uptime 67 ConnectFailures 0 FreeMem 19136 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init

tonhuisman, 2022-11-20:
Well, that's not too bad, if it get down to ~5 kB it would be problematic, but this should be fine.

No messages like `Could not initialize HCM5883L` I assume? As that would be the only other reason it's not sending out data, AFAICS.

tonhuisman, 2022-11-20:
Do you have multiple ESPEasy instances active, sending to the same FHEM server? And do the ESPEasy units have a unique Unit number assigned on the Config page? And that number should not be 0 or 255.
The Unit number is included in the message sent to FHEM, so when having duplicates, FHEM _might_ get confused by that.

tonhuisman, 2022-11-20:
I can see that you haven't enabled Rules on the Tools/Advanced page, so **no** events will be generated. Controller data _should_ still be sent though.

RoBra81, 2022-11-20:
Yes, I have multiple EspEasy but even after changing unit number only SysInfo is received within FHEM. Rules are enabled. Message `Could not initialize HCM5883L` isn't there and values for HCM5883 are displayed.

tonhuisman, 2022-11-20:
> values for HCM5883 are displayed.

Can you show a set from the log?

RoBra81, 2022-11-20:
No, they are not in the log only in the devices view they change

tonhuisman, 2022-11-20:
Ah, I've been looking at the most current `mega` code. There's been a [fix](https://github.com/letscontrolit/ESPEasy/pull/4339/files#diff-940d1c82a254743974fc205b7c976f8cc14efef8129f0944a1655ab052e0879dR151) to this plugin that's merged 11 days ago but not included in a release yet, and definitely explains why the data is never sent both to rules events or controllers...

Can you download from  a [recent Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/3501419232) the Binaries.zip file, and install from there the same configuration you currently have (Collection E)? That should fix it.

RoBra81, 2022-11-20:
Thank you very much, no I get the values to FHEM - only have to check, if rules are working too

tonhuisman, 2022-11-20:
> have to check, if rules are working too

You can see that in the Log

tonhuisman, 2023-09-09:
This can be closed as the issue has been resolved.

closed 2023-09-09 by TD-er