letscontrolit/ESPEasy #4389: Wiegand - conversion hex to dez

Heimweh76, 2022-11-28:
if the conversion from hex to dec is activated for the wiegand device, the 0 is not transmitted for a PIN starting with 0. 
Example: the PIN is 0815 - but only 815 is transmitted.
Is this the way it should be?

```

tonhuisman, 2022-11-29:
I assume you have also tried with that option turned off? Because it will be exactly the same. ESPEasy works with numeric values, and having a leading zero can have meaning in some systems/interpretations, so it's not usual to have that included.

It's not currently known/counted what length is input by the user, but I'll see if that can be made available, so you can format the output to the desired length from rules.

Heimweh76, 2022-11-29:
Yes, I have already deactivated this option - I just didn't get around to adding the hez -> dez conversion to my script yesterday. Yes, I understand that with the length. Would not necessarily help me because I work with manually entered pins (4 digits) and with RFID tags (7 digits).
I think it is easiest for me - not to choose 0 at the beginning of the pin. Or do you have a better idea?

tonhuisman, 2022-11-29:
> I think it is easiest for me - not to choose 0 at the beginning of the pin.

For the short term that's the best workaround, as I expect the leading `0`s _are_ included in the code received from the Wiegand device.

TD-er, 2022-11-29:
As far as I know, all instances of code where strings need to be converted to ints, we are using our own code.
So there should be no parsing from octal notation.
As long as those functions are being used on this plugin, you're good when using leading zeroes.

tonhuisman, 2022-11-29:
> So there should be no parsing from octal notation.

Yes, be we can't know what external systems do/expect.

I'm assuming this is about the leading 0s not being sent to some external system, as it states `transmitted` in the OP.