letscontrolit/ESPEasy #4391: A fresh custom Vagrant build not working

ghtester, 2022-11-30:
Today I tried to compile a fresh custom Vagrant build (custom_IR_ESP8266_4M1M) and tried to upgrade 2 nodes, both continuously crashing with wdt, and never connecting to WiFi.
Reverting back to ESP_Easy_mega_20220829_custom_IR_ESP8266_4M1M worked on one node, second node without serial connection needs to be restored later.
BTW. I have encountered a significant change in the Custom-sampl.h in the section I was using:
[image]
Now it looks like that - the section was removed:
[image]
And in the serial console I saw the error messages related to P016 plugin after the upgrade. So it looks something is really wrong.

TD-er, 2022-11-30:
Can you also try the "custom_beta_IR_ESP8266_4M2M" ?

ghtester, 2022-11-30:
Does not it mean the config is cleared up (when upgrading from custom_IR_ESP8266_4M1M)?

TD-er, 2022-11-30:
Nope, the settings files and file system layout are the same.
As long as you keep using the "ESP8266_4M1M" build envs. the settings should be kept.
Of course on crashing nodes or suspected to be crashing, it is best to backup the settings files and rules etc.

The differences between "alt.wifi" and "beta" and those builds without such being mentioned are:
- "alt.WiFi" now uses the same WiFi pre-build binary blob as we used in 20220829. The "default" wifi binary blob now uses one released by Espressif in March of 2019 as it seems to be working better on some nodes and at least makes a 802.11n connection according to the standards. (later SDK blobs don't)
- "beta" builds use the SDK3.x version

ghtester, 2022-11-30:
OK, thanks for the info, I'll give it a try tomorrow.

ghtester, 2022-12-01:
Just tried to compile but the recommended pio env. variable does not exist.
  default: Error: Unknown environment names 'custom_beta_IR_ESP8266_4M2M'. Valid names are 'minimal_core_274_ESP8266_1M_OTA_Domoticz, custom_IR_ESP32s2_4M316k, custom_ESP8266_4M2M, custom_beta_ESP8266_4M1M, custom_ESP32_16M8M_LittleFS, collection_E_ESP32s2_4M316k, display_ESP8266_4M1M, collection_D_ESP8266_4M1M, neopixel_ESP8266_1M_Domoticz, custom_ESP8266_4M1M, normal_302_ESP8266_4M1M, custom_beta_ESP8266_1M, neopixel_ESP32s2_4M316k, custom_alt_wifi_ESP8266_4M1M, collection_C_ESP32_IRExt_4M316k, normal_ESP8266_4M1M_VCC, collection_A_ESP32s2_4M316k, custom_IR_ESP8266_4M1M, neopixel_ESP8266_1M_FHEM_HA, collection_B_ESP32_4M316k, minimal_core_302_ESP8266_1M_OTA_Domoticz, normal_ESP8266_4M1M, custom_ESP32_4M316k, collection_D_ESP32_4M316k, custom_alt_wifi_ESP8266_1M, minimal_core_302_ESP8266_1M_OTA_FHEM_HA, hard_other_POW_ESP8285_1M, normal_beta_ESP8266_16M_LittleFS, normal_ESP32s2_4M316k, normal_IRext_no_rx_ESP8266_4M2M, custom_ESP8266_2M256, collection_A_ESP32_4M316k_ETH, display_ESP32s2_4M316k, collection_B_ESP8266_4M1M, minimal_core_302_ESP8266_1M_OTA_Domoticz_MQTT, collection_A_ESP32_4M316k, spec_debug_max_ESP32_16M8M, normal_WROOM02_2M256, climate_ESP32_4M316k_ETH, minimal_IRext_ESP8266_1M, collection_A_ESP8266_4M1M_VCC, custom_ESP8266_1M, normal_alt_wifi_ESP8266_4M1M, max_ESP32_16M8M_LittleFS_ETH, display_ESP32_4M316k, max_ESP32_16M1M, normal_beta_ESP8266_4M1M, minimal_IRext_ESP8266_4M2M, collection_D_ESP8266_4M1M_VCC, display_ESP32_4M316k_ETH, spec_debug_custom_ESP32_16M8M, collection_A_ESP8266_4M1M, custom_ESP32_4M2M_NO_OTA_LittleFS, normal_ESP32_4M316k_ETH, hard_LCtech_relay_x2_1M, collection_E_ESP32_4M316k_ETH, collection_A_ESP32_IRExt_4M316k, max_ESP32_16M1M_ETH, collection_B_ESP32s2_4M316k, collection_E_ESP8266_4M1M_VCC, custom_ESP32s2_4M316k, energy_ESP32s2_4M316k, minimal_core_274_ESP8266_1M_OTA_Domoticz_MQTT, collection_E_ESP32_IRExt_4M316k, neopixel_ESP8266_4M1M, hard_Ventus_W266_1M, custom_beta_2ndheap_ESP8266_4M1M, custom_beta_IR_ESP8266_4M1M, collection_D_ESP32s2_4M316k, climate_ESP8266_4M1M, energy_ESP8266_4M1M, normal_alt_wifi_ESP8266_1M_VCC, custom_ESP32_4M316k_ETH, collection_C_ESP32_4M316k_ETH, custom_IR_ESP32_4M316k, collection_C_ESP32_4M316k, collection_D_ESP32_4M316k_ETH, minimal_IRext_ESP8266_4M1M, custom_ESP32_4M316k_LittleFS, collection_B_ESP32_4M316k_ETH, collection_B_ESP8266_4M1M_VCC, hard_Shelly_1_2M256, normal_ESP32_4M316k, climate_ESP32s2_4M316k, spec_debug_beta_custom_ESP8266_4M1M, normal_ESP8266_1M, collection_B_ESP32_IRExt_4M316k, custom_ESP8266_4M2M_LittleFS, minimal_core_274_ESP8266_1M_OTA_FHEM_HA, custom_302_ESP8266_4M1M, spec_memanalyze_ESP8266, energy_ESP32_4M316k, normal_ESP32_4M316k_LittleFS, energy_ESP32_4M316k_ETH, neopixel_ESP32_4M316k, neopixel_ESP32_4M316k_ETH, collection_C_ESP32s2_4M316k, climate_ESP32_4M316k, normal_ESP8266_1M_VCC, hard_Shelly_PLUG_S_2M256, max_ESP32_16M8M_LittleFS, collection_E_ESP82

TD-er, 2022-12-01:
Ah missed the "4M2M" variant you're using.
Have to run now, but I can have a look at it later this evening to add that one.

tonhuisman, 2022-12-01:
> Ah missed the "4M2M" variant you're using.

I don't think the OP is using that type of config, he only mentioned the 4M1M with IR build...

ghtester, 2022-12-01:
Yeah I am using custom_IR_ESP8266_4M1M (currently built but not working properly) but adviced to try custom_beta_IR_ESP8266_4M2M which does not exist. 
I can try any other (existing) pio env. , preferably the version which can preserve the current ESP node configuration.
I can also flash a new - empty ESP node. But I need IRRX and IRTX plugins working.

ghtester, 2022-12-01:
So upgraded again ESP node running custom_IR_ESP8266_4M1M dated 20220829 with today's custom_beta_IR_ESP8266_4M1M  - the same issue, it's unusable.

5251 : Info   :

INIT : Booting version: ESP_Easy_mega_20221201_custom_beta_IR_ESP8266_4M1M, (Vagrant) My Build: Dec  1 2022 15:16:46 (ESP82xx Core fe451a81, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.3 PUYA support)
5252 : Info   : INIT : Free RAM:23392
5253 : Info   : INIT : Manual Reboot #1 - Restart Reason: External System
5255 : Info   : FS   : Mounting...
5288 : Info   : FS   : Mount successful, used 77057 bytes of 957314
5313 : Info   : CRC  : Settings CRC           ...OK
5451 : Info   : WIFI : Set WiFi to STA
11145 : Info   : WIFI : Set WiFi to OFF
12618 : Info   : WIFI : Set WiFi to STA
15047 : Info   : WIFI : Set WiFi to AP+STA
16641 : Info   : GPIO : port 16. Pulse set for 100 ms
16646 : Info   : GPIO : port#12: set to 0
17560 : Info   : GPIO : port#16: set to 1
36014 : Info   : WIFI : Set WiFi to OFF
36348 : Info   : WIFI : Set WiFi to STA
38777 : Info   : WIFI : Set WiFi to AP+STA
60868 : Info   : WIFI : Set WiFi to OFF
61202 : Info   : WIFI : Set WiFi to STA
63632 : Info   : WIFI : Set WiFi to AP+STA
85727 : Info   : WIFI : Set WiFi to OFF
86061 : Info   : WIFI : Set WiFi to STA
88491 : Info   : WIFI : Set WiFi to AP+STA
110594 : Info   : WIFI : Set WiFi to OFF
110928 : Info   : WIFI : Set WiFi to STA
                                        --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000df2f epc2=0x00000000 epc3=0x40000f68 excvaddr=0x00000033 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff3430 end: 3fff3840 offset: 0190
3fff35c0:  3fff75d4 402ceee0 00000020 00000000
3fff35d0:  3fff75d4 402ceee0 00000020 401017d0
3fff35e0:  402aabd3 402ceee0 40000000 402c428e
3fff35f0:  00000033 00000010 402ab4ed 3ffef360
3fff3600:  3fff75d4 00000000 3ffea958 3fff4834
3fff3610:  3fff75d4 402ab415 3ffef360 00000001
3fff3620:  3ffea958 00000004 402bb051 60009a00
3fff3630:  3fff6466 3fff58cc 00000000 00000000
3fff3640:  3ffea958 3ffea958 00000033 402ad35b
3fff3650:  3ffefa54 402acf54 40209d24 3ffefa54
3fff3660:  402acad8 3fff8ad4 3fff58cc 402c63f4
3fff3670:  00000001 00000001 3fff025e 00000000
3fff3680:  402acf8f 3ffefb48 3ffefb48 00000000
3fff3690:  402aef14 70400000 00007040 00000000
3fff36a0:  3ffef3e4 402aeed8 3ffef360 00000012
3fff36b0:  402acf44 402acf18 3ffef3e4 00000012
3fff36c0:  402acee3 3ffef360 00000012 00000001
3fff36d0:  402ad494 3ffef360 3ffef4dc 40209d24
3fff36e0:  402acec1 3ffef4dc 00000003 3ffef4dc
3fff36f0:  402af2f5 3fff3740 0000010f 3ffef4dc
3fff3700:  00000000 0000010f 069e4bfb 40240d0e
3fff3710:  40209d24 402c8c29 00000001 00000000
3fff3720:  00000000 3fff58cc 3fff58cc 4010144e
3fff3730:  3fff3740 40209d24 3fff58cc 40209ecd
3fff3740:  00000000 00000000 00000100 00000000
3fff3750:  00000000 00000000 3fff75d4 402554a2
3fff3760:  00000000 00000000 3fff15a4 402554c4
3fff3770:  3fff472c 00000000 3fff15a4 00000000
3fff3780:  00000000 00000000 00000000 4024139a
3fff3790:  069e4c14 00000000 02103830 402099d8
3fff37a0:  3fff08c8 3fff1ba8 069b0f2b 40240e31
3fff37b0:  00000000 00000000 00000000 3fff3874
3fff37c0:  3fff0fe8 3fff38a0 40280ac0 3fff38a0
3fff37d0:  3fff0fe8 00000000 3fff1ba8 4023e728
3fff37e0:  40242204 3fff1ba8 3fff1ba8 402421ca
3fff37f0:  3fffdad0 00000000 3fff3870 40243f78
3fff3800:  000049b8 feefeffe feefeffe 3fff38a0
3fff3810:  3fffdad0 00000000 3fff3870 3fff38a0
3fff3820:  3fffdad0 00000000 3fff3870 40280ce7
3fff3830:  feefeffe feefeffe 3fffdab0 40100535
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8
tail 0
chksum 0x2b
csum 0x2b
v000e9ab0
~ld
▒U336 : Info   : WIFI : Set WiFi to AP+STA
5181 : Info   : WIFI : Set WiFi to OFF
5405 : Info   :

INIT : Booting version: ESP_Easy_mega_20221201_custom_beta_IR_ESP8266_4M1M, (Vagrant) My Build: Dec  1 2022 15:16:46 (ESP82xx Core fe451a81, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.3 PUYA support)
5406 : Info   : INIT : Free RAM:23184
5408 : Info   : INIT : Exception #2 - Restart Reason: Exception
5409 : Info   : FS   : Mounting...
5442 : Info   : FS   : Mount successful, used 77057 bytes of 957314
5467 : Info   : CRC  : Settings CRC           ...OK
5605 : Info   : WIFI : Set WiFi to STA
11382 : Info   : WIFI : Set WiFi to OFF
12856 : Info   : WIFI : Set WiFi to STA
15285 : Info   : WIFI : Set WiFi to AP+STA
16880 : Info   : GPIO : port 16. Pulse set for 100 ms
16885 : Info   : GPIO : port#12: set to 0
17803 : Info   : GPIO : port#16: set to 1
36256 : Info   : WIFI : Set WiFi to OFF
36590 : Info   : WIFI : Set WiFi to STA
39019 : Info   : WIFI : Set WiFi to AP+STA
61130 : Info   : WIFI : Set WiFi to OFF
61464 : Info   : WIFI : Set WiFi to STA
63894 : Info   : WIFI : Set WiFi to AP+STA
86018 : Info   : WIFI : Set WiFi to OFF
86352 : Info   : WIFI : Set WiFi to STA
88780 : Info   : WIFI : Set WiFi to AP+STA

TD-er, 2022-12-01:
Which plugins are active here?

ghtester, 2022-12-01:
I don't know at the moment as the node is inaccessible now, wifiapmode command does not help. Can't connect to WiFi AP.
But tgere's AM2301 sensor and AFAIK yesterday it was disabled (automatically after several crashes), I had to reenable it when I restored the 20220829 firmware.

ghtester, 2022-12-01:
FYI... After reboot command entered in serial console the node crashed soon. Looks unstable.

2622177 : Info   : WIFI : Set WiFi to OFF
2622511 : Info   : WIFI : Set WiFi to STA
2624941 : Info   : WIFI : Set WiFi to AP+STA
2647034 : Info   : WIFI : Set WiFi to OFF
2647368 : Info   : WIFI : Set WiFi to STA
2649797 : Info   : WIFI : Set WiFi to AP+STA
2671886 : Info   : WIFI : Set WiFi to OFF
2672220 : Info   : WIFI : Set WiFi to STA
2674650 : Info   : WIFI : Set WiFi to AP+STA
>reboot
2685561 : Info   : WIFI : Set WiFi to OFF
2685884 : Info   : WIFI : Set WiFi to STA
2688314 : Info   : WIFI : Set WiFi to AP+STA
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3424, room 16
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8
tail 0
chksum 0x2b
csum 0x2b
v000e9ab0
~ld
▒U339 : Info   : WIFI : Set WiFi to AP+STA
5184 : Info   : WIFI : Set WiFi to OFF
5407 : Info   :

INIT : Booting version: ESP_Easy_mega_20221201_custom_beta_IR_ESP8266_4M1M, (Vagrant) My Build: Dec  1 2022 15:16:46 (ESP82xx Core fe451a81, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.3 PUYA support)
5408 : Info   : INIT : Free RAM:23336
5410 : Info   : INIT : Soft Reboot #3 - Restart Reason: Software/System restart
5411 : Info   : FS   : Mounting...
5445 : Info   : FS   : Mount successful, used 77057 bytes of 957314
5470 : Info   : CRC  : Settings CRC           ...OK
5607 : Info   : WIFI : Set WiFi to STA
11387 : Info   : WIFI : Set WiFi to OFF
12861 : Info   : WIFI : Set WiFi to STA
                                       --------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000df2f epc2=0x00000000 epc3=0x402c6938 excvaddr=0x00000033 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff3340 end: 3fff3840 offset: 0190
3fff34d0:  3fff5f44 402ceee0 00000020 00000000
3fff34e0:  3fff5f44 402ceee0 00000020 401017d0
3fff34f0:  402aabd3 402ceee0 40000000 402c428e
3fff3500:  00000033 00000010 402ab4ed 3ffef360
3fff3510:  3fff5f44 00000000 3ffea958 3fff4834
3fff3520:  3fff5f44 402ab415 3ffef360 00000001
3fff3530:  3ffea958 00000004 402bb051 60009a00
3fff3540:  3fff446e 3fff58cc 00000000 00000000
3fff3550:  3ffea958 3ffea958 00000033 402ad35b
3fff3560:  3ffefa54 402acf54 40209d24 3ffefa54
3fff3570:  402acad8 3fff6fc4 3fff58cc 402c63f4
3fff3580:  00000001 00000001 3fff025e 00000000
3fff3590:  402acf8f 3ffefb48 3ffefb48 00000000
3fff35a0:  402aef14 c7c00000 0000c7c0 00000000
3fff35b0:  3ffef3e4 402aeed8 3ffef360 00000012
3fff35c0:  402acf44 402acf18 3ffef3e4 00000012
3fff35d0:  402acee3 3ffef360 00000012 00000001
3fff35e0:  402ad494 3ffef360 3ffef4dc 40209d24
3fff35f0:  402acec1 3ffef4dc 00000003 3ffef4dc
3fff3600:  402af2f5 3fff3650 0000010f 3ffef4dc
3fff3610:  00000000 0000010f 00c5e60a 40240d0e
3fff3620:  40209d24 402c8c29 00000001 00000000
3fff3630:  00000000 3fff58cc 3fff58cc 4010144e
3fff3640:  3fff3650 40209d24 3fff58cc 40209ecd
3fff3650:  00000000 00000000 00000100 00000000
3fff3660:  00000000 00000000 3fff5f44 402554a2
3fff3670:  00000000 00000000 3fff15a4 402554c4
3fff3680:  3fff5eb4 00000000 3fff15a4 00000000
3fff3690:  00000000 00000000 00000000 4024139a
3fff36a0:  00c5e624 00000000 00000000 402099d8
3fff36b0:  3fff08c8 3fff1ba8 00c2a8f1 40240e31
3fff36c0:  00000000 00000000 00000000 00399e85
3fff36d0:  3fff37a0 3fff0be3 00000000 3fff0c04
3fff36e0:  3fff37a0 00000000 00c2a8de 40243d05
3fff36f0:  00000039 3fff0be3 3fff373c 3fff37e0
3fff3700:  3fff37a0 3fff0be3 00000000 4023ff80
3fff3710:  3fff3750 3fff374c 3fff3748 3fff3753
3fff3720:  3fff3752 00000000 3fff3770 4027f4ae
3fff3730:  00000000 0009004f 00000000 00000000
3fff3740:  0005004f 00000000 3fff73dc 402d6318
3fff3750:  00000000 00000000 3fff37a0 4027f89d
3fff3760:  00000352 00000001 00c2a7de 00000000
3fff3770:  00000000 00000000 00000000 3fff37e0
3fff3780:  00000000 00000000 3fff37a0 3fff37e0
3fff3790:  00000000 00000001 00000000 4024003d
3fff37a0:  656c7572 742e3173 0a007478 3fff20c0
3fff37b0:  00c17a80 00000000 3fff37e0 4027f2ac
3fff37c0:  3fff2350 402d5cff 3fff37e0 3fff20c0
3fff37d0:  3fff2350 3ffe8664 3fff1cc0 4024450f
3fff37e0:  3fff61fc 000b000f 80000000 00000000
3fff37f0:  00000000 00000000 feefeffe feefeffe
3fff3800:  3fff15a4 feefeffe feefeffe feefeffe
3fff3810:  feefeffe feefeffe feefeffe 3fff38a0
3fff3820:  3fffdad0 00000000 3fff3870 40280cdb
3fff3830:  feefeffe feefeffe 3fffdab0 40100535
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3424, room 16
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8
tail 0
chksum 0x2b
csum 0x2b
v000e9ab0
~ld
▒U332 : Info   : WIFI : Set WiFi to AP+STA
5177 : Info   : WIFI : Set WiFi to OFF
5401 : Info   :

INIT : Booting version: ESP_Easy_mega_20221201_custom_beta_IR_ESP8266_4M1M, (Vagrant) My Build: Dec  1 2022 15:16:46 (ESP82xx Core fe451a81, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.3 PUYA support)
5402 : Info   : INIT : Free RAM:23208
5404 : Info   : INIT : Exception #4 - Restart Reason: Exception
5405 : Info   : FS   : Mounting...
5438 : Info   : FS   : Mount successful, used 77057 bytes of 957314
5463 : Info   : CRC  : Settings CRC           ...OK
5601 : Info   : WIFI : Set WiFi to STA
11294 : Info   : WIFI : Set WiFi to OFF
12768 : Info   : WIFI : Set WiFi to STA
15197 : Info   : WIFI : Set WiFi to AP+STA
16791 : Info   : GPIO : port 16. Pulse set for 100 ms
16796 : Info   : GPIO : port#12: set to 0
17713 : Info   : GPIO : port#16: set to 1
36169 : Info   : WIFI : Set WiFi to OFF
36503 : Info   : WIFI : Set WiFi to STA

ghtester, 2022-12-01:
I have flashed a new - empty - ESP node with today's custom_IR_ESP8266_4M1M build. It could not connect to WiFi due the #4337 issue (I have the **Do Not Start AP:** option enabled in default config). Connected after entering **wifiapmode**
 command. Disabled the **Do Not Start AP:** option, rebooted the node and it connected to WiFi. There are no plugins and the node seems to be working, so far without crash.
5177 : Info   :

INIT : Booting version: ESP_Easy_mega_20221201_custom_IR_ESP8266_4M1M, (Vagrant) My Build: Dec  1 2022 15:00:39 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
5178 : Info   : INIT : Free RAM:26072
5179 : Info   : INIT : Cold Boot - Restart Reason: External System
5180 : Info   : FS   : Mounting...
5206 : Info   : FS   : Mount successful, used 75802 bytes of 957314
5224 : Info   : CRC  : Settings CRC           ...OK
5227 : Info   : CRC  : SecuritySettings CRC   ...OK
5240 : Info   : INIT : I2C
5242 : Info   : INIT : SPI not enabled
5357 : Info   : WIFI : Set WiFi to STA
7648 : Info   : WIFI : Connecting TestAP 00:14:0C:25:58:B4 Ch:1 (-79dBm) WPA2/PSK attempt #0
7875 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 125 ms
7977 : Info   : Reset WiFi.
10165 : Info   : WIFI : Connecting TestAP 00:14:0C:25:58:B4 Ch:1 (-76dBm) WPA2/PSK attempt #1
11114 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 125 ms
11114 : Info   : WIFI : Set WiFi to OFF
11228 : Info   : INIT : Free RAM:25440
11302 : Info   : INFO : Plugins: 40 [IR][No Debug Log] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
11303 : Info   : EVENT: System#Wake
11392 : Info   : Webserver: start
11403 : Info   : EVENT: System#Boot
11405 : Info   : WIFI : Disconnected! Reason: '(202) Auth fail' Connected for 949 ms
11617 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 1350 ms
11718 : Info   : EVENT: WiFi#APmodeDisabled
11829 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 1562 ms
11931 : Info   : EVENT: WiFi#Disconnected
11975 : Info   : EVENT: WiFi#Disconnected
11987 : Info   : EVENT: WiFi#Disconnected
11998 : Info   : EVENT: WiFi#Disconnected
12744 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 20072 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
>wifiapmode
16733 : Info   : WIFI : Set WiFi to AP
16839 : Error  : WIFI : Error while starting AP Mode with SSID: IR IP: 192.168.4.1

OK
20269 : Info   : WIFI : Set WiFi to AP+STA
20377 : Info   : WIFI : Connecting TestAP 00:14:0C:25:58:B4 Ch:1 (-76dBm) WPA2/PSK attempt #2
27392 : Info   : WIFI : DHCP IP: 192.168.22.109 (IR) GW: 192.168.22.253 SN: 255.255.255.0   duration: 6024 ms
27465 : Info   : NTP  : NTP replied: delay 11 mSec Accuracy increased by 0.757 seconds
27466 : Info   : Time set to 1669925867.757
27468 : Info   : Current Time Zone:  DST time start: 2022-03-27 02:00:00 offset: 120 min STD time start: 2022-10-30 03:00:00 offset: 60 min
27471 : Info   : Local time: 2022-12-01 21:17:47
27476 : Info   : UDP : Start listening on port 8266
27477 : Info   : firstLoopConnectionsEstablished
27480 : Info   : EVENT: WiFi#Connected
27483 : Info   : WIFI : Connected! AP: TestAP (00:14:0C:25:58:B4) Ch: 1 Duration: 987 ms
27485 : Info   : WIFI : Starting mDNS...
27488 : Info   : WIFI : mDNS started, with name: IR.local
27491 : Info   : EVENT: Time#Initialized
27519 : Info   : EVENT: WiFi#ChangedAccesspoint
27521 : Info   : EVENT: WiFi#ChangedWiFichannel
28182 : Info   : EVENT: Clock#Time=Thu,21:17
40182 : Info   : EVENT: Clock#Time=Thu,21:18
42738 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 17808 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP

TD-er, 2022-12-01:
Yep there is something really funky with the latest code, where it seems the ESP needs to have AP+STA mode enabled to be successful in connecting to an AP.

ghtester, 2022-12-01:
And also a config compatibility issue as the old node upgraded to latest code is crashing and can't connect to WiFi at all... The **wifiapmode** command does not work on it, seems to be ignored...

tonhuisman, 2022-12-01:
Could clearing the SDK WiFi settings, using the `EraseSDKwifi` command, clear things up here?

ghtester, 2022-12-01:
It did not help... Command accepted with OK response. Rebooted node with **reboot** command,  after boot several crashes... Not connecting to WiFi. The same after cold boot.