letscontrolit/ESPEasy #4393: esp8266 & esp32 can't connect to wifi after wake from deepsleep latest release

dertobias, 2022-12-01:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Since Release mega-20221105 esp8266 and esp32(lolin lite) won't connect to wifi after a wake from deep sleep.

rolled back to mega-20220809 and everything works like it should.

### Expected behavior

both esp's wake, publish to mqtt and go to sleep again

### Actual behavior

esp's wont connect to wifi with
`Disconnected! Reason: '(8) Assoc leave' Connected for ...`
also i have to re-enable the devices, controller & activate rules again

### Steps to reproduce

1. update to latest release
2. enable sleep mode

### System configuration

Hardware: wemos d1 mini with battery shield and liion cell. lolin32 lite (az-delivery) with lipo battery

ESP Easy version:  mega-20221105

`ESP_Easy_mega_20221105_normal_ESP8266_4M1M_VCC.bin`
`ESP_Easy_mega_20221105_normal_ESP32_4M316k.bin`

ESP Easy settings/screenshots: 

### Rules or log data

```
Disconnected! Reason: '(8) Assoc leave' Connected for
```

dertobias, 2022-12-01:
tried to increase sleep awake time (from 6 to 120 sec) cause i noticed it takes more time until it connects…but every time i actually use deep sleep it does not connect ....so it has to be connected to deep sleep.

without deep sleep (gpio16 to GND and then wait until it connects to WiFi) it connects...after a while

```
load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
�U309 : Info   : WIFI : Set WiFi to AP+STA
5099 : Info   : WIFI : Set WiFi to OFF
5316 : Info   :

INIT : Booting version: ESP_Easy_mega_20221105_normal_ESP8266_4M1M, (GitHub Actions) mega-20221105 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(c0eb301), LWIP: 2.1.2 PUYA support)
5317 : Info   : INIT : Free RAM:26176
5319 : Info   : INIT : Soft Reboot #8 Last Action before Reboot: Intended Reboot: CommandReboot Last systime: 1669907316 - Restart Reason: Software/System restart
5320 : Info   : FS   : Mounting...
5346 : Info   : FS   : Mount successful, used 75802 bytes of 957314
5366 : Info   : CRC  : Settings CRC           ...OK
5370 : Info   : CRC  : SecuritySettings CRC   ...OK
5373 : Info   : INIT : I2C
5374 : Info   : INIT : SPI not enabled
5376 : Info   : Time : Set Ext. Time Source: RTC at boot time: 1669907316
5388 : Info   : Time set to 1669907316.000
5391 : Info   : Current Time Zone:  DST time start: 2022-03-27 02:00:00 offset: 120 min STD time start: 2022-10-30 03:00:00 offset: 60 min
5394 : Info   : Local time: 2022-12-01 16:08:36
5408 : Info   : WiFi : Best AP candidate: Test_NW_2.4G 54:AF:97:DD:1A:5F Ch:12 (-44dBm) WPA2/PSK
5513 : Info   : WIFI : Set WiFi to STA
5617 : Info   : WiFi : Start network scan all channels
7808 : Info   : WiFi : Scan finished, found: 10
7812 : Info   : WiFi : Best AP candidate: Test_NW_2.4G 54:AF:97:DD:1A:5F Ch:12 (-46dBm) WPA2/PSK
7813 : Info   : WiFi : Added known candidate, try to connect
7815 : Info   : WIFI : Connecting Test_NW_2.4G 54:AF:97:DD:1A:5F Ch:12 (-46dBm) WPA2/PSK attempt #0
7817 : Info   : IP   : Static IP : 192.168.1.193 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1
7823 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
7825 : Info   : WiFi : WifiDisconnect()
7927 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 10 ms
8029 : Info   : Reset WiFi.
8032 : Info   : WiFi : Start network scan all channels
10225 : Info   : WiFi : Scan finished, found: 9
10230 : Info   : WiFi : Best AP candidate: Test_NW_2.4G 54:AF:97:DD:1A:5F Ch:12 (-46dBm) WPA2/PSK
10231 : Info   : WiFi : Added known candidate, try to connect
10233 : Info   : WIFI : Connecting Test_NW_2.4G 54:AF:97:DD:1A:5F Ch:12 (-46dBm) WPA2/PSK attempt #1
10235 : Info   : IP   : Static IP : 192.168.1.193 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1
10242 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
11183 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 10 ms
11184 : Info   : WIFI : Set WiFi to OFF
11298 : Info   : INIT : Free RAM:24568
11385 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(c0eb301), LWIP: 2.1.2 PUYA support)
11387 : Info   : EVENT: System#Wake
11514 : Error  : WiFi : Connecting not possible, unprocessed WiFi events:  disconn
11515 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
11518 : Info   : Webserver: start
11520 : Info   : EVENT: System#Boot
11523 : Info   : WIFI : Disconnected! Reason: '(202) Auth fail' Connected for 952 ms
11727 : Info   : WIFI : Set WiFi to STA
11831 : Error  : WiFi : Scan not needed, good candidate present
11933 : Info   : WIFI : Set WiFi to AP+STA
12926 : Info   : WIFI : AP Mode ssid will be ESP-Easy-F2 with address 192.168.4.1
12938 : Info   : EVENT: WiFi#APmodeDisabled
12970 : Info   : EVENT: Time#Initialized
13038 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 17272 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: DISCONNECTED
13073 : Info   : SI7013 : sn=a63f605-15ffffff
13074 : Info   : P014: chip_id=21
13087 : Info   : EVENT: WiFi#Disconnected
13124 : Info   : EVENT: TaskInit#sensor=1,0
13150 : Info   : EVENT: WiFi#Disconnected
13176 : Info   : EVENT: WiFi#APmodeEnabled
13276 : Info   : P014: Temperature: 23.17 Humidity: 55.70
13306 : Info   : EVENT: sensor#Temperature=23.17
13375 : Info   : EVENT: sensor#Humidity=55.70
30239 : Info   : WiFi : Best AP candidate: Test_NW_2.4G 54:AF:97:DD:1A:5F Ch:12 (-46dBm) WPA2/PSK
30241 : Info   : WIFI : Connecting Test_NW_2.4G 54:AF:97:DD:1A:5F Ch:12 (-46dBm) WPA2/PSK attempt #2
30243 : Info   : IP   : Static IP : 192.168.1.193 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1
30249 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
30374 : Info   : WIFI : Connected! AP: Test_NW_2.4G (54:AF:97:DD:1A:5F) Ch: 12 Duration: 107 ms
30375 : Info   : WIFI : Static IP: 192.168.1.193 (ESP-Easy-F2) GW: 192.168.1.1 SN: 255.255.255.0   duration: 25 ms
30389 : Info   : EVENT: WiFi#ChangedAccesspoint
30415 : Info   : EVENT: WiFi#ChangedWiFichannel
30440 : Info   : EVENT: WiFi#Connected
30471 : Error  : MQTT : Intentional reconnect
30494 : Info   : MQTT : Connected to broker with client ID: ESP_Easy_F2_12
30498 : Info   : Subscribed to: ESP_Easy_F2/#
30503 : Info   : UDP : Start listening on port 8266
30503 : Info   : firstLoopConnectionsEstablished
30517 : Info   : NTP  : NTP replied: delay 10 mSec Accuracy increased by 0.539 seconds
30519 : Info   : Time set to 1669907346.539 Time adjusted by 5406.81 msec. Wander: 0.0 ppm Source: NTP
30523 : Info   : Local time: 2022-12-01 16:09:06
30530 : Info   : EVENT: MQTT#Connected
30534 : Info   : EVENT: Time#Set
31002 : Info   : static_file: esp.css
31148 : Info   : static_file: favicon.ico
31299 : Info   : EVENT: Clock#Time=Thu,16:09
42855 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 15544 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
```

flashmark, 2023-01-01:
I see tsimilar behavior for two EPS8266 modules after an upgrade to a new, self built, version. AP is recognized, but connection fails with "Disconnected! Reason: '(8) Assoc leave' Connected for 10 ms". However this will be repeated forever. I am not using deepsleep. Running exactly the same image on a 3rd board works fine. It looks like the WiFi behavior has changed such that some modules won't connect anymore. I am not sure to open a separate ticket or go with this one.

TD-er, 2023-01-01:
Such behavior should be fixed in the 20221224 build....
But make sure to set the nr. of extra WiFi scans to 0 (as was reported by some other user).

richik, 2023-05-17:
I used to set "Extra WiFi scan loops" to 1 - on 20220809 without any issues - worked perfectly (AP reboot => WiFi disconnected and ESP connected to another AP within few seconds).
The same I set on 20230409, but when AP is lost then ESP does not connect to secondary configured AP and even not to original one (once it come back online) => reboot needed. But setting "Extra WiFi scan loops" to 0 fixes this and it connects to of the configured SSIDs in a few seconds after disconnect (or directly by wifidisconnect command).

TD-er, 2023-05-18:
Yep, this extra WiFi scan does seem to cause more issues than it solves.
I think I will remove the setting.
On the latest build, there are 2 more settings, which may be useful in some specific situations:

- Extra Wailt WiFi Connect
- Enable SDK WiFi Auto Reconnect

See: https://espeasy.readthedocs.io/en/latest/Tools/Tools.html#extra-wait-wifi-connect