letscontrolit/ESPEasy #4395: P118 Itho plugin crashes/reboot on SPI and CS settings (ESP8266)

PhamNguyenNL, 2022-12-02:
### Checklist
- [X] The title of this issue is "to the point" and descriptive.
- [X] This issue describes what is happening.
- [X] This issue describes what components are affected (e.g. name of plugin/controller)
- [X] This issue describes how to reproduce it.
- [X] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [X] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [X] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [X] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [X] given a list of active plugins or controllers when applicable.
- [X] filled out all applicable fields below.

### Steps already tried...
- [X] Tried a clean install (empty `.bin` files are included in the ZIP)
- [X] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

There are 2 situations.

- CS pin (CC1101 CSN) is set. Using the command tool will crash/reboot (fast).
- SPI and CS pin set will crash/reboot directly and takes long to go up. 

I think problems with new changes on de CS pin field???

I've tried the latest 4 versions. They all work except the latest. But I have an "Orcon" so need that support.
Other versions work well. I could retrieve the RF id's. No crashes/reboot.

Already erased flash memory and tried the blank.bin and a lot of versions for testing.

more info: https://gathering.tweakers.net/forum/list_message/73629286#73629286

### Expected behavior

Save settings, doesn't disable plugin and crash/reboot

### Actual behavior

Disable plugin and reboot

### Steps to reproduce

1. CS pin (CC1101 CSN) is set. Using the command tool will crash/reboot (fast).
2.  SPI and CS pin set will crash/reboot directly and takes long to go up. 

### System configuration

Hardware: CC1101 868 MHz transceiver + Wemos D1 mini pro (ESP-8266 4M1M)
[image]

ESP Easy version:  Release mega-20221105 -> ESP_Easy_mega_20221105_climate_ESP8266_4M1M.bin

ESP Easy settings/screenshots: 
[image]
[image]
[image]

### Rules or log data

```
Date    Time    Level   Host Name   Category    Program Messages
2022-11-29  11:26:28    Notice  ESP-Itho    user        <7>ESP-Itho EspEasy: WiFi : Missed connected event
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: ITHO: Extra Settings PLUGIN_118 loaded
2022-11-29  11:23:01    Notice  ESP-Itho    user        <7>ESP-Itho EspEasy: Read settings: CustomTaskSettings index: 0
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: INIT PLUGIN_118
2022-11-29  11:23:01    Notice  ESP-Itho    user        <7>ESP-Itho EspEasy: Read settings: TaskSettings index: 0
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: Skip saving SecuritySettings, not changed
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: SaveToFile: free stack after: 3424
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: FILE : Saved config.dat offset: 0 size: 1316
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: SaveToFile: free stack: 3424
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: SaveToFile: free stack after: 3360
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: FILE : Saved config.dat offset: 4096 size: 536
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: SaveToFile: free stack: 3360
2022-11-29  11:23:01    Notice  ESP-Itho    user        <7>ESP-Itho EspEasy: Write settings: TaskSettings index: 0
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: SaveToFile: free stack after: 2768
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: FILE : Saved config.dat offset: 5120 size: 27
2022-11-29  11:23:01    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: SaveToFile: free stack: 2768
2022-11-29  11:23:00    Notice  ESP-Itho    user        <7>ESP-Itho EspEasy: Write settings: CustomTaskSettings index: 0
2022-11-29  11:23:00    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: EXIT PLUGIN_118
2022-11-29  11:23:00    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: SaveToFile: free stack after: 3104
2022-11-29  11:23:00    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: FILE : Saved config.dat offset: 4096 size: 536
2022-11-29  11:23:00    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: SaveToFile: free stack: 3104
2022-11-29  11:23:00    Notice  ESP-Itho    user        <7>ESP-Itho EspEasy: Write settings: TaskSettings index: 0
2022-11-29  11:23:00    Notice  ESP-Itho    user        <5>ESP-Itho EspEasy: Webserver 23 Arguments (First 20): 0: 'index' length: 1 1: 'page' length: 1 2: 'TDNUM' length: 3 3: 'TDN' length: 4 4: 'TDE' length: 2 5: 'taskdevicepin1' length: 1 6: 'taskdevicepin2' length: 2 7: 'pID1' length: 8 8: 'pID2' length: 0 9: 'pID3' length: 0 10: 'plog' length: 2 11: 'prflog' length: 2 12: 'pdevid1' length: 3 13: 'pdevid2' length: 3 14: 'pdevid3' length: 3 15: 'orcon' length: 2 16: 'TDSD1' length: 2 17: 'TDVN1' length: 5 18: 'TDVN2' length: 5 19: 'TDVN3' length: 11

```

Side question:
What does this exactly mean? Note: For Orcon: This is the destination ID a.k.a. the ID of the Ventilation unit.
[image]

The way I read it was the RF id of the Orcon ventilation unit and I should set it at the device id bytes. Which I could since I retrieved "77,9c,94" as the Unit ID. The Device ID bytes are only numbers.
I was told it was the same the note above "Note: Device ID of your ESP, should not be the same as your neighbours ;-). Defaults to 10,87,81 which corresponds to the old Itho library" So I could set my own Device ID for my ESP to use.
If that's correct maybe you should consider rephrase that line.

TD-er, 2022-12-02:
Do you happen to have some serial dumps of these crashes?
Maybe it will give some info on the crash reason.
Right now I have no idea where to start looking, as it can be something as simple as running out of memory.
Other options:
Watchdog reboot due to not enabling interrupts again.

TD-er, 2022-12-02:
Oh and the problems you mentioned about getting the device to boot again made me think about you using GPIO 15.
This pin might be pulled up by the cc1101. See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266

So maybe you need to use a PNP transistor between GPIO 15 and this cc1101 module?

PhamNguyenNL, 2022-12-06:
> Do you happen to have some serial dumps of these crashes? Maybe it will give some info on the crash reason. Right now I have no idea where to start looking, as it can be something as simple as running out of memory. Other options: Watchdog reboot due to not enabling interrupts again.

`——————— CUT HERE FOR EXCEPTION DECODER ———————`

Exception (3):
epc1=0x40209d3d epc2=0x00000000 epc3=0x00000000 excvaddr=0x402cf2ac depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff3400 end: 3fff3840 offset: 0190
3fff3590:  3fff0878 3fff5be4 0000007e 40209d36  
3fff35a0:  00000008 3fff0878 3fff5be4 40209bed  
3fff35b0:  3fff5be4 00000030 3fff5be4 3fff1d74  
3fff35c0:  3fff5be4 00000001 3fff5be4 4020a338  
3fff35d0:  3fff5be4 00000001 3fff5b74 402687b8  
3fff35e0:  00000000 00000002 80000003 4022c32e  
3fff35f0:  00002c2e 00000000 80ff36e0 4024dbfa  
3fff3600:  00000008 00000101 00000000 3fff5b74  
3fff3610:  3fff1d74 00000000 3fff36e0 402226d7  
3fff3620:  00001a78 0000034f 0000034f 40101313  
3fff3630:  3fff36e0 0000002c 3fff3670 3fff36e0  
3fff3640:  3fff36e0 0000002c 00000020 401015cb  
3fff3650:  3fff5b00 0010001f 3fff3690 40283184  
3fff3660:  0000010c 0000002c 3fff3690 40283165  
3fff3670:  402e0000 3fff2114 00000000 3fff36e0  
3fff3680:  3fff36e0 0000002c 00000000 402428af  
3fff3690:  3fff5b00 000c000f 80ff36c0 40283165  
3fff36a0:  00000002 3fff37c0 00000160 00ac8275  
3fff36b0:  00000000 00000001 3fff3740 40106031  
3fff36c0:  4028133c 00000001 3fff36e0 00000001  
3fff36d0:  3fff37c0 00000002 3fff36e0 402434d3 <
3fff36e0:  00005d00 00ac6fa3 80ff3710 40283100  
3fff36f0:  00001a60 8000008d 3fff3700 4022bf32 <
3fff3700:  80000000 3fff1b00 3fff3710 80ff1bc8 <
3fff3710:  3fff1b00 40243eac 80000020 00000000  
3fff3720:  00000000 00000000 00000000 00000000  
3fff3730:  00000000 00000000 03030000 3f000600  
3fff3740:  67756c00 61436e69 80006c6c 40243e89  
3fff3750:  00000000 3fff5b44 80000000 00000000  
3fff3760:  3fff37a0 80000000 03030c00 40106031  
3fff3770:  4028133c 000e0000 00000000 00000001  
3fff3780:  3fff1840 3fff1c8c 00005760 4024945c  
3fff3790:  00ac7f00 00000000 00000000 4024945c  
3fff37a0:  00000087 3fff1e46 3fff1e45 00000001  
3fff37b0:  0000031a 3fff1e46 3fff1e46 4024a08e  
3fff37c0:  3fff1800 3fff22a8 80ff22ac 00000001  
3fff37d0:  3fff1840 00004f37 3fff1d74 402411b1  
3fff37e0:  65637800 6f697470 80ef006e 3fff5d00  
3fff37f0:  0000007f 80efeffe feefeffe feefeffe  
3fff3800:  feefeffe feefeffe feefeffe 3fff388c  
3fff3810:  3fffdad0 00000000 3fff3848 40211218  
3fff3820:  feefeffe feefeffe feefeffe 40284b9f  
3fff3830:  feefeffe feefeffe 3ffe86d8 401004d1  
<<<stack<<<

——————— CUT HERE FOR EXCEPTION DECODER ———————

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
ªU329 : Info   : WIFI : Set WiFi to AP+STA
5134 : Info   : WIFI : Set WiFi to OFF
5351 : Info   : 

INIT : Booting version: ESP_Easy_mega_20221105_climate_ESP8266_4M1M, (GitHub Actions) mega-20221105 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(c0eb301), LWIP: 2.1.2 PUYA support)
5352 : Info   : INIT : Free RAM:24968
5354 : Info   : INIT : Exception #23 Last Action before Reboot: Background Task Last systime: 1670283946 - Restart Reason: Exception
5355 : Info   : FS   : Mounting…
5380 : Info   : FS   : Mount successful, used 75802 bytes of 957314
5400 : Info   : CRC  : Settings CRC           …OK
5534 : Info   : WIFI : Set WiFi to STA
11185 : Info   : WIFI : Set WiFi to OFF
11581 : Info   : WIFI : Set WiFi to STA
13982 : Info   : WIFI : Set WiFi to AP+STA

`——————— CUT HERE FOR EXCEPTION DECODER ———————`

It's stuck in a bootloop and repeating this in the log.

> Oh and the problems you mentioned about getting the device to boot again made me think about you using GPIO 15. This pin might be pulled up by the cc1101. See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266
> 
> So maybe you need to use a PNP transistor between GPIO 15 and this cc1101 module? See as example: https://www.letscontrolit.com/forum/viewtopic.php?t=8914#p56426

I've looked at the schemes and please correct me if I'm wrong.

`15 (D8) | pulled to GND | OK | Boot fail if pulled high`
So I need to use a PNP transistor to pull the GPIO15 to the ground.

- B to GPIO15 (D8)
- C to Ground
- E to CSN

I don't think it make sense to put a PNP transistor there. Since it's boot fine (the article you referring has a boot problem) and it worked fine with older firmware.

>Watchdog reboot due to not enabling interrupts again.
Not sure how to set it. I think this refer to it right? -> https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html

TD-er, 2022-12-06:
The watchdog due to not enabling interrupts is a software thing (if it is the cause here), so that needs to be fixed in ESPEasy.

If your unit boots well, then you don't need a PNP transistor.
But then I wonder if it will be pulled high well enough to switch the "CS" signal.

Can you also test a later build, from the GitHub Actions test builds?
For example this one: https://github.com/letscontrolit/ESPEasy/actions/runs/3614982471
Just to make sure we haven't already fixed this since the 20221105 build.

PhamNguyenNL, 2022-12-06:
@TD-er I've just tested it. Unsuccessfully.

`Full log on update until it boot again (with probably some bootloops?)`
Update: ESP_Easy_mega_20221204_climate_ESP8266_4M1M.bin
sleep disable
…………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………………….Update Success: 1012608
Rebooting…
3a5ac
@cp:0
ld
 : Set WiFi to OFF
21711 : Info   : 

INIT : Booting version: ESP_Easy_mega_20221204_climate_ESP8266_4M1M, (GitHub Actions) mega_26b4fa9 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
21712 : Info   : Mounting…
21740 : Info   : FS   : Mount successful, used 75802 bytes of 957314
21759 : Info   : CRC  : Settings CRC           …OK
21890 : Info   : WIFI : Set WiFi to STA
27517 : Info   : WIFI : Set WiFi to OFF
——————— CUT HERE FOR EXCEPTION DECODER ———————

Exception (3):
epc1=0x4020e369 epc2=0x00000000 epc3=0x000 3fff0b00 00000002 80000000 4022c829  
3fff36c0:  00006c55 00000000 3fff37b0 4024edca  
3fff36d0:  00000008 00000101 00000000 3fff5bf4  
3fff36e0:  3fff12cc 00000000 3fff37b0 40222aa3  
3fff36f0:  00001a28 00000345 00000345 40101313  
3fff3700:  3fff37b0 0000002c 3fff3740 00000002  
3fff3710:  3fff37b0 0000002c 00000020 401015cb  
3fff3720:  3fff5c00 0010001f 3ff0 000c000f 80ff3790 402836c9  
3fff3770:  3fff37b0 00000160 3fff3890 01a71a8d  
3fff3780:  00000000 00000001 3fff3810 40106031  
3fff3790:  402818a8 00000001 3fff37b0 00000001  
3fff37a0:  3fff3890 00000002 3fff37b0 40243823 <
3fff37b0:  00005d00 01a70803 80ff37e0 40283600  
3fff37c0:  00001a10 8000008d00000 00000000 03030000 3f000600  
3fff3810:  67756c00 61436e69 80006c6c 4024461d  
3fff3820:  00000000 3fff5bc4 80000000 00000000  
3fff3830:  3fff3870 80000000 03030c00 40106031  
d 00000001  
3fff3880:  0000031a 3fff139e 3fff139e 4024e506  
3fff3890:  3fff1700 3fff0de0 80ff0de4 00000001  
3fff38a0:  3fff17f0 00004f37 3fff12cc 402418e5  
3fff38b0:  3fff5f00 0017001f 80efeffe 3fff5f00  
3fff38c0:  0000003f 80efeffe feefeffe feefeffe  
3fff38d0:  feefeffe feefeffe feefeffe 3fff395cT HERE FOR EXCEPTION DECODER ———————

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40103a5ac
~ld
ªU303 : Info   : WIFI : Set WiFi to AP+STA
5083 : Info   : WIFI : Set WiFi to OFF
5300 : Info   : 

INIT : Booting version:dev(38a443e), LWIP: 2.1.2 PUYA support)
5300 : Info   : INIT : Free RAM:24760
5302 : Info   : INIT : Exception #1 Last Action before Reboot: Background Task Last systime: 21 - Restart Reason: Exceptionfo   : WIFI : Set WiFi to STA
Exception (3):
epc1=0x4020e369 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402d649e depc=0x00000000

>>>stack>>>

ctx: 0 3fff44fc 4020e219  
3fff3680:  3fff44fc 00000030 3fff44fc 3fff36a0:  3fff44fc 00000001 3fff44ac 40269188  
3fff36b0:  3f 3fff44ac  
3fff36e0:  3fff12cc 00000000 3fff37b0 40222aa3  
3fff36f0:  00000278 0000004f 0000004f 40101313  
3fff3700:  3fff37b0 0000002c 3fff3740 00000002  
3fff3710:  3fff37b0 0000002c 00000020 401015cb  
[... 350 lines cut ...]
3fff3700:  3fff37b0 0000002c 3fff3740 00000002  
3fff3710:  3fff37b0 0000002c 00000020 401015cb  
3fff3720:  3fff5c00 0010001f 3fff3760 402836e8  
3fff3730:  0000010c 0000002c 3fff3760 40283650:  3fff37b0 0000002c 00000000 40242c00  
3fff3760:  3fff5c0fff3890 00ab9210  
3fff3780:  00000000 00000001 3fff3810 40106031  
3fff3790:  402818a8 00000001 3fff37b0 00000001  
3fffd00 00ab7f87 80ff37e0 40283600  
3fff37c0:  00001a50 8000008dfff17f0 3fff2348 00005690 40246d7c  
3fff3860:  00000000 0000
3fff3890:  3fff1700 3fff0de0 80ff0de4 00000001  
3fff38a0:  3fff17f0 00004f54 3fff12cc 402418e5  
3fff38b0:  65637800 6f697470 80ef006e 3fff5e00  
3fff38c0:  0000006f 80efeffe feefeffe feefeffe  
3fff38d0:  feefeffe feefeffe feefeffe 3fff395c:  feefeffe feefeffe feefeffe 40285103  
3fff3900:  feefeffe T HERE FOR EXCEPTION DECODER ———————

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40103a5ac
~ld
n: ESP_Easy_mega_20221204_climate_ESP8266_4M1M, (GitHub Actions) mega_26b4fa9 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2- Reason: Exception
5290 : Info   FI : Set WiFi to OFF
11524 : Info   : WIFI : Set WiFi to STA
13915 : Info   : WIFI : Set WiFi to AP+STA

PhamNguyenNL, 2022-12-06:
`This is the full log after enabling the device until it boot again. (probably with some bootloops too?`
00000 excvaddr=0x402d649e depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff33d0 end: 3fff3910 offset: 0190
3fff3560:  30950 3fff8cec 4020e219  
3fff3580:  3fff8cec 00000030 3fff8cec 3fff12cc  
3fff3590:  3fff8cec 00000001 3fff8cec 4020e964  d29c  
3fff3690:  00000000 00000000 80000000 00000000  
3fff000 00000000 03030000 4000ff00  
3fff36c0:  54494e00 61436e00 80006c6c 3fff5700  
3fff36d0:  0011001f 80ff12cc 3fff3700 40001a0fa 80ffffff 3fff8500 00000001  
3fff3760:  800000dd 3fff2100 00000020 801015cb  
3fff3770:  00000000 3fff37f8 8023f8a 00000000 3fff2104 80ff8be8 00000008  
3fff37b0:  00000001 3f800 40276bec  
3fff37d0:  00000003 00000001 3fff6bec 40290b16:  3fff6bec 3fff2120 3fff6bec 40276c24  
3fff3800:  76656400 f20e0 40240ec6  
3fff3820:  7665642f 73656369 88ff8d00 0000008f  
3fff3830:  80006976 4bc6a7f0 0001a0ed 3fff395c  
3fff3840:  00000000 00000001 3fff2190 000000dd  
3fff3850:  00000003 0000000f 40100bc6 27ef9db2  
3fff3860:  00000000 3fff85d4 3fff20e0 3fff395c  
3fff3870:  00000001 3fff2104 3fff20e0 4024cc4 00000000 065c9dd3 40241111  
3fff38b0:  3ffe8568 00000000efeffe  
3fff38d0:  00000000 00000000 00000001 3fff395c  
3ffeffe feefeffe feefeffe 4028510f  
3fff3900:  feefeffe feefefroom 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
n: ESP_Easy_mega_20221204_climate_ESP8266_4M1M, (GitHub Actions) mega_26b4fa9 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
5295 : Info   : INIT :ion before Reboot: Background Task Last systime: 1670287522 - Restart Reason: Exception
5298 :    …OK
5474 : Info   : WIFI : Set WiFi to STA
11101 : Info   : WIFI : Set WiFi to OFF00000 excvaddr=0x402d649e depc=0x00000000

>>>stack>>>

ctfff0950 3fff6b74 0000007e 4020e362  
3fff3670:  00000008 3fff4 3fff12cc  
3fff3690:  3fff6b74 00000001 3fff6b74 4020e964  000000 3fff37b0 4024edca  
3fff36d0:  00000008 00000101 00000  
3fff36f0:  00001a68 0000034d 0000034d 40101313  
3fff37000000002c 00000020 401015cb  
3fff3720:  3fff5c00 0010001f 3fff3760 402836e8  
3fff3730:  0000010c 0000002c 3fff3760 40283637a0:  3fff3890 00000002 3fff37b0 40243823 <
3fff37b0:  00005 3fff3700 4022c432  
3fff37d0:  80000000 3fff2200 3fff37e0 80ff37f0:  00000000 00000000 00000000 00000000  
3fff3800:  00000000000  
3fff3830:  3fff3870 80000000 03030c00 40106031  
3fff3840:  402818a8 00000000 80ff5cc4 00000001  
3fff3850:  30000 00000000 40246d7c  
3fff3870:  00000087 3fff139e 3fff139
3fff3890:  3fff1700 3fff0de0 80ff0de4 00000001  
3fff38a0: 697470 80ef006e 3fff5e00  
3fff38c0:  0000007f 80efeffe feefe  
3fff38e0:  3fffdad0 00000000 3fff3918 4021138c  
3fff38f03a5ac
~ld
 Free RAM:24824
5300 : Info   : INIT : Exception #13 Last Action before Reboot: Background Task Last systime: 1670287522 - Restart Reason: Exception
5301 : tings CRC           …OK
5476 : Info   : WIFI : Set WiFi to STA
11104 : Info
——————— CUT HERE FOR EXCEPTION DECODER ———————

[... 339 lines cut ...]
3fff3760:  3fff5b00 000c000f 80ff3790 402836c9  
3fff3770:  3fff37b0 00000160 3fff3890 00ab5020  
3fff3780:  00000000 00000001 3fff3810 40106031  
3fff3790:  402818a8 00000001 3fff37b0 00000001  
3fff37a0:  3fff3890 00000002 3fff37b0 40243823 <
3fff37b0:  00005d00 00ab3dae 80ff37e0 40283600  
3fff37c0:  000019f0 8000008dff37f0:  00000000 00000000 00000000 00000000  
3fff3800:  00000000 00000000 03030000 3f000600  
3fff3810:  67756c00 61436e3fff3840:  402818a8 00000000 80ff5d0c 00000001  
3fff3850:  3fff17f0 3fff2348 000056e8 40246d7c  
3fff3860:  00000000 0000d 00000001  
3fff3880:  0000031a 3fff139e 3fff139e 4024e506  
3fff3890:  3fff1700 3fff0de0 80ff0de4 00000001  
3fff38a0: ffe feefeffe  
3fff38d0:  feefeffe feefeffe feefeffe 3fff395cfeefeffe 3ffe86d8 401004d1  
<<<stack<<<

——————— CU3a5ac
~ld
ev(38a443e), LWIP: 2.1.2 PUYA support)
5296 : Info   : INIT :Restart Reason: Exception
5299 : tings CRC           …OK
5475 : Info   : WIFI : Set WiFi to STA
11102 : Info   : WIFI : Set WiFi to OFF
11506 : Info   : WIFI : Set WiFi to STA

TD-er, 2022-12-06:
Does the P118 task get disabled after some time? (typically (10 + task index nr) reboots)

PhamNguyenNL, 2022-12-06:
After enabling it crashes and the task is disabled after boot. I've only created one task for the P118 on the first line.

TD-er, 2022-12-21:
Is there any update on this?
It sounds a bit like a hardware issue.
Even though it should not crash. But the problem is that SPI devices may keep the SPI bus busy if they do not act as expected, so not sure what we can do in software to mitigate this.

PhamNguyenNL, 2022-12-21:
I thought you were busy with this issue. Are there major changes within the SPI settings in the mega-20221105 in general or in the plugin? I still want to fix the problem. Because the mega-20220809 was working.

What are my options to get the Orcon support to work?

- Add the PNP transistor as discussed before to test if it doesn't keep SPI busy?
- Check changes for the Orcon support and add it to a custom build of mega-20220809?

Any possibility for you to create a custom test build of mega-20220809 with the new P118 plugin with Orcon support to test?

I've tried to check docs and tutorials to create a custom build but it failed somehow and stopped working on it.

Do you have some links for me to read/see how to create a custom esp easy build? So I want to have the compiled version of mega-20220809 esp easy version with the mega-20221105 p118 plugin to test the Orcon support. Later I want to solve the crash as well.

TD-er, 2022-12-21:
Well "busy with this issue" is only partly true.
Ton was looking into it and we both discussed this offline. (well, via Signal, but not in public here :) )
Ton mentioned this issue as one that we still need to look into before making the new "Christmas build".

You mention it was working fine on 20220809, so we can see what related code might have changed since.

I cannot find here whether you already have tested it with a newer test build or not.
Can you test to see if it perhaps already is fixed in some other commit in one of the latest test builds?
For example this build: https://github.com/letscontrolit/ESPEasy/actions/runs/3744004947

tonhuisman, 2022-12-22:
@PhamNguyenNL I've found and fixed the issue that caused the crash.
Can you verify using this [Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/3761451590) (once it's completed), and report your findings?

closed 2022-12-23 by TD-er

gjstroom, 2022-12-23:
I tried the ESP_Easy_mega_20221223_climate_ESP8266_4M1M.bin and it looks like it's working !
I had a problem when I set "GPIO → CS pin (CC1101 CSN)" to "GPIO-15 (D8) the ESP froze and resulted in a boot loop.

TD-er, 2022-12-23:
Great!
Thanks for reporting back.

PhamNguyenNL, 2022-12-24:
@tonhuisman and @TD-er I can confirm that it's working! Thanks!

Is there any good documentation how to get the Orcon functionality working?

In the build of https://github.com/svollebregt/ESPEASY_Plugin_ITHO (ESP_Easy_mega_20211105_normal_ESP8266_4M1M) I could retrieve the remote id in with the log (weblog in dev).

Any instructions how to retrieve the RF id of the remote or unit? 
What does this exactly mean? Note: For Orcon: This is the destination ID a.k.a. the ID of the Ventilation unit.

Please correct me if I'm wrong. I need to set the Unit ID remote 1 (remote RF id) and Device ID bytes 1,2,3 Orcon Ventilation unit? Or Unit ID remote 2 = Orcon Ventilation unit and Device ID bytes 1,2,3 = ID of ESP?

It's kinda unclear to me.

tonhuisman, 2023-01-04:
> Any instructions how to retrieve the RF id of the remote or unit (in current builds)? What does this exactly mean? Note: For Orcon: This is the destination ID a.k.a. the ID of the Ventilation unit.

I would expect that information to be available on the sticker of the Orcon device, but I don't own such device (nor do I know anyone who does), so can't confirm...

PR #4428 (2022-12-23): [P118] Bugfix for crash during startup
Fixes #4395 

Features:
- Bugfix for crash during initialization of CC1101
- Make destructor virtual
- Move function from .h to .cpp