letscontrolit/ESPEasy #4400: [Feature Request]  more unique plugins for inputs / outputs

chromoxdor, 2022-12-06:
Actually I wanted to write a feature request but then i found out, that someone beat me to it. :)

[image]

But i would like to add:
      `// - switch output (relays) and pwm output (led) - show the duty cycle as a value`

TD-er, 2022-12-06:
Then at least make a PR for that comment and please also add a link to this issue to it. :)

chromoxdor, 2022-12-06:
> Then at least make a PR for that comment and please also add a link to this issue to it. :)

I am not there yet (in terms of competence). The PR would be an empty/nothing changed one. 
Don't overestimate me. :)

TD-er, 2022-12-06:
> Don't overestimate me. :)

I call it "encouragement" ;)

tonhuisman, 2022-12-06:
> I call it "encouragement" ;)

Worked for me 🤣

LeeNX, 2022-12-06:
You are all very funny!  
  
I can assist with the PR ... Not very good at code thou ... 🤣

TD-er, 2022-12-06:
The "code" for that PR would just be updating a comment, so no functional changes....
Unless of course you want to implement it too :)  (hint: encouragement)

LeeNX, 2022-12-06:
I get that, I was offering @chromoxdor to talk him through the process, if I understand it correctly. Long time user & lurker. Not much on PR.

TD-er, 2022-12-06:
But it will be quite a bit of code to actually implement it, as it is about splitting the functionality in the P001 plugin, which is on my todo list for a looooong time since it is probably a new box left by someone called Pandora and I wouldn't be surprised if that box is filled with other boxes too.

LeeNX, 2022-12-06:
Sounds like @TD-er needs a walk before putting his daughter to sleep and a short coding session. 😱

chromoxdor, 2022-12-06:
> I call it "encouragement" ;)

I definitely am engouraged :)

> The "code" for that PR would just be updating a comment, so no functional changes....

Ahhh... this i actually can do

> I get that, I was offering @chromoxdor to talk him through the process, if I understand it correctly. Long time user & lurker. Not much on PR.

If you mean with "process" creating a PR, than i am fine, thank you.
If you mean the actual coding of the Plugin....

> But it will be quite a bit of code to actually implement it, as it is about splitting the functionality in the P001 plugin, which is on my todo list for a looooong time since it is probably a new [...]

...than call me discouraged 😁
(i mean i´ll look into it anyway..just to see if i understand how it works)

TD-er, 2022-12-06:
What an emotional rollercoaster thread :)

TD-er, 2022-12-06:
> Sounds like @TD-er needs a walk before putting his daughter to sleep and a short coding session. 😱

Not sure if it will be a short coding session, splitting up P001.
But given this comment, you are for sure a long-time-lurker :)

chromoxdor, 2022-12-06:
😂

LeeNX, 2022-12-06:
First merged PR! 🥳