letscontrolit/ESPEasy #4404: _P052_SenseAir.ino values too low (<400)

DittelHome, 2022-12-07:
Hi,
i am using the S8 Senseair Modul connected via TX GPIO 1 (D10) and RX GPIO 3 (D9).
Every Morning after opening window i get CO2 values below 400.
After power off/on, same values.
During the day, the value grows up to "normal" values.
The Automatic Baseline Correction seems not to work.

[image]

[image]

[image]

TD-er, 2022-12-07:
The default ABC period of a SenseAir S8 is 7 or 8 days if I remember correctly.
I don't know if this period will reset/restart when the sensor gets power cycled.

Did you change the ABC on this sensor using some tools from SenseAir, or some other firmware?

DittelHome, 2022-12-07:
> The default ABC period of a SenseAir S8 is 7 or 8 days if I remember correctly.
I don't know if this period will reset/restart when the sensor gets power cycled.

Normally the ESP is awlays up, today's reboot was only a test.
Doku says: ABC period 15 days

> Did you change the ABC on this sensor using some tools from SenseAir, or some other firmware? (other than ESPEasy)

The Sensor is from day one connected to ESPEASY.
Can i verfiy, that ABC is active ?

TD-er, 2022-12-07:
> Can i verfiy, that ABC is active ?

I have some pending requests regarding ABC and SenseAir sensors.
I will see what I can do to add such info on the task config page.
Right now, the only way you can "see" it is when it suddenly appears to make a "jump" on the lowest values. (max. 50 ppm jumps I think)

DittelHome, 2022-12-08:
I think the problem is already known since 2018 with issue #759.
It seems that the S8 sensor is not really useable, right ?

TD-er, 2022-12-08:
That issue is the reason why I asked whether the sensor was ever used on another firmware.
It is not about the usability of the sensor.

Problem with ABC is that it may introduce issues when the sensor isn't in a well ventilated room for longer than the ABC period.
Then it may try to correct its "400 ppm" level to whatever the lowest measured value was in the past N days.
If the CO2 concentration was never near 400 ppm, then the ABC will set its auto-calibration to the wrong level.

The ABC period of the S8 if by default 7 or 8 days (not 100% sure), where others like the Winsen MH-Z19 sensors have it at 24h (which is quite unusable to be honest)
Also the SenseAir sensor will not correct it with the full computed correction.
e.g. new base value should be corrected with 200 ppm, it will only correct this with a fraction of it, or 50 ppm.

But this longer ABC period can also work against you as you may need to have the sensor operate in a well ventilated room for a longer period to get it back to normal again.

I can re-add the ABC setting instructions again. But since I have no idea how to detect whether a sensor is affected by the linked bug or not.

DittelHome, 2022-12-11:
Ok, shame on my head.
The sensor works as it should, I was just too impatient.
Sorry for the rush, please close the ticket

closed 2022-12-11 by TD-er