letscontrolit/ESPEasy #4407: [Libraries] Spaces in names and duplicated libs

clumsy-stefan, 2022-12-07:
ArduinoIDE (and probably other) don't like Library names with spaces in it:
`Adafruit GFX Library`
`Adafruit NeoMatrix`
`SparkFun SCD4x Arduino Library`

Also it seems that due to this some libraries are now duplicated (and outdated) in the repository:
`Adafruit NeoPixel` and `Adafruit_NeoPixel`
`SparkFun MAX1704x Fuel Gauge Arduino Library` and `SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library`

Probably I missed something as I was offline for 3 month, but I think things should neither be duplicated nor have spaces in names due to various systems not really like dealing with spaces on filesystem level.

TD-er, 2022-12-07:
Those library names are not something we have any saying in.
And yep, some are outdated and some are changed by us to make them work.
But the idea is that they are now all included in the libs folder.
So ideally you should be able to just copy this folder and use it as an Arduino IDE library folder.

clumsy-stefan, 2022-12-07:
>So ideally you should be able to just copy this folder and use it as an Arduino IDE library folder.

Yep, that's what I normally do, but it does not work anymore as ArduinoIDE doesn't accept spaces in libraries and therefore doesn't use them... So I needed to rename them before/after copying, which is fine for me, just wanted to share that probably others also have this issue..

> Can you show how the folder names are when Arduino IDE does install the same libs?

I think ArduinoIDE just uses underscores instead of spaces (as it was before in ESPEasy).

PR #4408 (2022-12-08): [Build] Deduplicate libs and un-space lib foldernames
Resolves: #4407 
Features:
- Remove duplicated libraries (keeping newest)
- Use library folder names without spaces for compatibility with Arduino IDE
closed 2022-12-08 by TD-er