letscontrolit/ESPEasy #4410: [Wifi] No connection with mega from Dec,8th 2022

uwekaditz, 2022-12-09:
@TD-er 
With the latest sources (merged Dec,8th 20:22 @ 09:24) I do not get a working Wifi connection. Attached is a log.

With the mega sources merged Dec,3rd 20:22 @ 13:32 it worked well!
[Log.txt](https://github.com/letscontrolit/ESPEasy/files/10196140/Log.txt)

TD-er, 2022-12-09:
And here is (hopefully) the fix: https://github.com/letscontrolit/ESPEasy/pull/4411
Not that it was fixed in 39 minutes.... Have been working on it the whole day as I messed up one of my Sonoff POW nodes yesterday when uploading a build based on the latest code.

uwekaditz, 2022-12-09:
It works! 🥇 👍 
Attached is a log.
[Working.log](https://github.com/letscontrolit/ESPEasy/files/10196625/Working.log)

TD-er, 2022-12-09:
Great!
So I can merge it :)

PR #4411 (2022-12-09): [WiFi] Fix loop WIFI_STA and WIFI_OFF preventing boot/wificonnect
Some strange situation where ESP8266 SDK seems to toggle WiFi mode, leaving the state unexpected for ESPEasy. This ended up in an infinite loop switching between WIFI_STA and WIFI_OFF. This happened on an old Sonoff POW.
No idea what may have caused this.

Other fixes:
- No longer trying to automatically reconnect when trying to reboot.
- Deal with disconnect events before trying to reconnect.
- Don't force WiFi disconnect when AP is turned off.
- Register WiFi event handlers on ESP8266 only once (as we already did on ESP32) to  [... cut]
closed 2022-12-09 by TD-er