letscontrolit/ESPEasy #4412: on Clock#Time=All,**:**  disables other on Clock#Time=All,With_specific_Time

TommoT1, 2022-12-10:
Hi, I'm using the  mega-20220809 release and I noticed that if you have a on Clock#Time=All,**:** running for example publishing something, any other on Clock#Time=All,xx:xx with a specific time to do anything else is obmitted.
In earlier days this worked flawless.

Also on Clock#Time=All,"[Var#1]:[Var#2]" do doesn't work at my setup anymore (discussed here : https://letscontrolit.com/forum/viewtopic.php?t=6957)

Any Ideas?
Greets
Thomas

tonhuisman, 2022-12-10:
What was the previous release you had installed on the unit?

Can you, for testing purposes only, turn off the Tools/Advanced setting `Enable Rules Cache`, to see if that fixes it?

TD-er, 2022-12-10:
Hmm it could be the [var#1] part may indeed be already interpreted before it is stored in the cache.
That would be a bug.

Another thing to quickly test would be this:
```
Clock#Time=All,%v1%:%v2%
```
N.B. I also removed the quotes as I don't see why it should contain a comma or spaces.

TD-er, 2022-12-10:
Can you give the exact line as how it is used in the rules?
I suspect you're missing something.

> When I turn off the "Enable Rules Cache" nothing works anymore......

That's even more strange and indicative there might be something else wrong with your rules.

TommoT1, 2022-12-10:
Here are the rules, this device is a SonOff S26 switchable plug.
If you switch it on, 1min is added to the current time and then it shall switch off again.
I use those timers in a weekend House to switch on a heating or right now a frost protection and with writing the time and state to a dummy variable, the device restores the current status (heating or not, till when?) after a reboot.

on Taster#Switch=1 do event,toggle

on toggle do
 if [Relay#State]= 0
   let,1,%syshour%+0 // +h till off
   let,2,%sysmin%+1  // +min till off
   event,SteckdoseEin
 else
   gpio,12,0
 endif
TimerSet,1,2
endon

//on Clock#Time=All,**:** do 
//  event,pub
//  logentry,[Uhr#hr#D2]:[Uhr#min#D2]
//endon

on SteckdoseEin do 
 if [VAR#2] > 59
  let,2,[VAR#2]-60
  let,1,[VAR#1]+1
 endif
 if [VAR#1] >= 24
  let,1,[VAR#1]-24
 endif 
 TaskValueSet,11,1,([VAR#1])
 TaskValueSet,11,2,([VAR#2])
 gpio,12,1
 TimerSet,1,2
endon

on Clock#Time=All,"[VAR#1#D2]:[VAR#2#D2]" do 
  event,pub
  gpio,12,0
  TimerSet,1,2
endon

on Rules#Timer=1 do event,pub 

on pub do Publish QWERTZ1311/S26,"%sysname%.%ip4% %rssi% %syshour_0%:%sysmin_0% Rel:[Relay#State]"

TommoT1, 2022-12-10:
When I turn off "enable rules chache" even the toggle does nothing anymore

TD-er, 2022-12-10:
I think the "one-liner" lines may break it using the current implementation.

```
on Taster#Switch=1 do 
  event,toggle
endOn

on Rules#Timer=1 do 
  event,pub
endOn

on pub do 
  Publish QWERTZ1311/S26,"%sysname%.%ip4% %rssi% %syshour_0%:%sysmin_0% Rel:[Relay#State]"
endOn
```

TD-er, 2022-12-10:
```
on toggle do
 if [Relay#State]= 0
   let,1,%syshour%+0 // +h till off
   let,2,%sysmin%+1  // +min till off
   event,SteckdoseEin
 else
   gpio,12,0
 endif
TimerSet,1,2
endon
```
followed by:
```
on Clock#Time=All,"[VAR#1#D2]:[VAR#2#D2]" do 
```
Is a very elaborate way to simply start a timer for 1 minute.
And it isn't even an exact way as it might only last 1 second in worst case.

Better use a `timerset,2,60` and then use `on Rules#Timer=2 do` instead of the `Clock#Time=....` stuff.

TommoT1, 2022-12-10:
Ok again the one-liners... yes than it works when I disable "enable rules chache". But still a  "clock#time=All,**:**" disables the one at a specific time.
Yes of course for a one minute timer it makes no sense, it's just for testing. In the weekend house I switch on a heating and from that time I want it to be one several hours and  "survive" a reboot.
And just now with temperatures getting colder I noticed via MQTT that the devices don't work anymore that expected.

TommoT1, 2022-12-10:
And the on Clock#Time=All,[int#1#D2]:[int#2#D2] do was ment to be 
on Clock#Time=All,"[Uhr#hr#D2]:[Uhr#min#D2]" do since Uhr#hr and Uhr#min ist set via TaskValueSet to be reboot save

TD-er, 2022-12-10:
Can you create a logentry with this, so we can see what the output is that the rules parser eventually has to parse?

Thus:
```
LogEntry,"Clock#Time=All,[int#1#D2]:[int#2#D2]"
```
You can even do this from the tools->Command line field and keep another window open on the log.

TommoT1, 2022-12-10:
I'm away from home and will get the logentry this evening

TommoT1, 2022-12-10:
Now I moved the "on Clock#Time=All,**:** do" part below the "on Clock#Time=All,[Uhr#hr#D2]:[Uhr#min#D2] do" and with 
 disabled rules chache it seems to work like expected. 
The log output looks like this:
25478936: ACT : TaskValueSet,11,1,(16)
25478940: ACT : TaskValueSet,11,2,(36)
25478944: ACT : gpio,12,1
25478945: GPIO : port#12: set to 1
25478947: ACT : TimerSet,1,2
25478953: ACT : LogEntry,'Clock#Time=All,16:36'
25478954: Clock#Time=All,16:36
25478960: ACT : TimerSet,1,2
25478984: EVENT: Relay#State=1
25479183: EVENT: Taster#Switch=0
25480962: EVENT: Rules#Timer=1,1
25480990: ACT : event,pub
25480991: EVENT: pub
25481025: ACT : Publish QWERTZ1311/S26,'S26¹.65 -35 16:35 Rel:1'
25501671: WD : Uptime 425 ConnectFailures 0 FreeMem 20768 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
25531699: WD : Uptime 426 ConnectFailures 0 FreeMem 20656 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
25533117: EVENT: Clock#Time=Sat,16:36
25533138: ACT : event,pub
25533141: EVENT: pub
25533175: ACT : Publish QWERTZ1311/S26,'S26¹.65 -32 16:36 Rel:1'
25533186: ACT : gpio,12,0
25533188: GPIO : port#12: set to 0
25533190: ACT : TimerSet,1,2
25533208: EVENT: Relay#State=0
25535192: EVENT: Rules#Timer=1,1
25535220: ACT : event,pub
25535221: EVENT: pub
25535253: ACT : Publish QWERTZ1311/S26,'S26¹.65 -35 16:36 Rel:0'

TD-er, 2022-12-10:
huh?
I have not seen a `on Clock#Time=All,: do` part in your code.

If that was present, then it for sure makes perfect sense that it wasn't working.

You need to have the most specific one on top and you can have some kind of "catch all" variant at the end.

Problem is that `on Clock#Time=All,: do` will match always and thus the rules parsing will not continue when it has found one to match.

TommoT1, 2022-12-10:
it should read on Clock#Time=All,**:** do
...ok makes sense, the most specific on on top and further down the broader ones.
But is it assured that if I have e.g.
on Clock#Time=All,12:25 do Task1
on Clock#Time=All,**:** do Task2

at 12:25 both Task will be performed or just Task 1?

tonhuisman, 2022-12-10:
Because of missing back-ticks the formatting for `on Clock#Time=All,**:** do` is broken in several comments.

@TommoT1 can you use back-ticks around literal strings, and triple-back-ticks before and after code listings?
[Github markdown: Basic writing and formatting syntax](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax)

TD-er, 2022-12-10:
> Ah,ok makes sense, the most specific on on top and further down the broader ones. But is it assured that if I have e.g. on Clock#Time=All,12:25 do Task1 on Clock#Time=All,**:** do Task2
> 
> at 12:25 both Task will be performed or just Task 1?

Only the first matching event block will be processed.
In older implementations, the rules parser continued even after an event was handled, which does take a lot of resources on the ESP.

TommoT1, 2022-12-10:
@tonhuisman 
Yes, Thank you, I noticed it on my last post, will look into the document...

@TD-er
Thanks for explaining me that!

TD-er, 2022-12-10:
Using 3 back ticks (at begin and end of a block) you can make multi-line blocks use the code markup.
See also the post of you I edited to add these: https://github.com/letscontrolit/ESPEasy/issues/4412#issuecomment-1345235813

TommoT1, 2022-12-10:
> Using 3 back ticks (at begin and end of a block) you can make multi-line blocks use the code markup. See also the post of you I edited to add these: [#4412 (comment)](https://github.com/letscontrolit/ESPEasy/issues/4412#issuecomment-1345235813)

Yes, thanks found it in the document ton mentioned

burdal1, 2022-12-11:
> Only the first matching event block will be processed. In older implementations, the rules parser continued even after an event was handled, which does take a lot of resources on the ESP.

oh, this is very uncomfortable for my use of `on Clock#Time=zz,xx:yy do`. I used this as a weekly scheduler in combination with the one-minute monitoring with `on Clock#Time=All,**:** do`.

Now the first `on Clock#Time=All,**:** do` blocks the other parts with specific time `on Clock#Time=zz,xx:yy do`.

What are the other options? It seems I have to redesign everything 

This looks much better for me:
```
on Clock#Time=All,**:** do
 let,1,[var#1]+1
endon

on Clock#Time=Wrk,7:00 do
  Publish zigbee/Tice/set, 25
endon
```

than this:

```
on Clock#Time=All,**:** do
 let,1,[var#1]+1
 If %syshour% = 7 and %sysmin%  = 0 and %sysweekday% > 1 and %sysweekday% < 7
  Publish zigbee/Tice/set, 25
 Endif 
endon
```
is there a better solution?

Anyway, thanks for the great work on ESPEasy. It's amazing project.

tonhuisman, 2022-12-11:
> is there a better solution?

As already written above, by specifying these rules from most specific to more generic, they should work, as the rules processing is stopped after the first matching rule is found (to reduce execution time). If you would want to add a rule to only execute on every Thursday, it should be _before_ the Workdays `Wrk` rule, like this:
```
on Clock#Time=Thu,12:00 do // Most specific Clock#Time (Thursday, fixed time)
  // Commands
endon

on Clock#Time=Wrk,7:00 do // Somewhat specific Clock#Time (Workdays, fixed time)
  Publish zigbee/Tice/set, 25
endon

on Clock#Time=All,**:** do // Most generic Clock#Time (All days, every minute)
 let,1,[var#1]+1
endon
```

So, the only thing you have to change is the order or your rules 😃

TommoT1, 2022-12-11:
But if I get it correct, on Thu at 12:00 the second and third rule is not checked/executed and on Workdays at 7:00 the third rule ist skipped also. Am I right?

TD-er, 2022-12-11:
That seems right indeed.

TD-er, 2022-12-11:
If you want to make sure you're not copy/pasting code into duplicate code, then I can suggest to move that code into a separate event block.

```
on weekdays do
  ...
endon
```

Then you can "call" this using `asyncevent,weekdays`.
Or whatever name you want to give it.

TD-er, 2022-12-21:
Is this issue now 'resolved' by following the explanations?

TommoT1, 2022-12-21:
Yes, with disabled Rules Cache and only one matching "on clock#time"  at a given time everything works

tonhuisman, 2022-12-21:
It is expected to work with the Rules Cache _enabled_ too, can you confirm that?

TommoT1, 2022-12-21:
Not anymore, I changed everything and now I have only one If then Block wich ist called every Minute and makes sure everything which needs to be done is done....
I can check If this one works with Cache enabled but it is different from where we started this issue...

tonhuisman, 2022-12-21:
> I can check If this one works with Cache enabled but it is different from where we started this issue...

My suggestion of testing it without caching enabled was intended to turn it off temporarily, to see if it made a difference, but it didn't seem to. So I have assumed you turned it back on after testing. That's the suggested, and default, state, as it nicely improves performance of rules processing.

TommoT1, 2022-12-31:
Good morning!
I played around with another device and release 20221224 and stepped again over  ``` on clock#time ```
Again in a dummy variable I store the time something should be done and than I have the following code
```on Clock#Time=All,[Numbers#hr]:[Numbers#min] do
    gpio,12,1
   TimerSet_ms,1,200
endon```
This on is completely ignored, it does'nt even produce a logentry.

TD-er, 2022-12-31:
That doesn't surprise me it will not work.
Not sure what side effects we will see when calling parseTemplate() on the on...do line before matching.

tonhuisman, 2022-12-31:
> Not sure what side effects we will see when calling parseTemplate() on the on...do line before matching.

I'd expect a speed-bump on rules processing, as you will have to call it on _every_ cached event _before_ matching, and the events have to be stored unparsed.

TD-er, 2022-12-31:
The events lines with on...do are stored in memory when you enable the cache.
And they should be stored "unprocessed" anyway, but I'm not sure if we call parseTemplate() on this line (I guess not)

TommoT1, 2022-12-31:
> That doesn't surprise me it will not work. Not sure what side effects we will see when calling parseTemplate() on the on...do line before matching.

Not sure if I understand you correct, is it related to the rules cache? Because I have some devices with exactly such rules wich used to work with earlier releases....

TD-er, 2023-01-01:
Nope, right now it will not work at all.

Just a bit of info on how the rules are being processed:

When handling an event, the ESP has to search for the first matching line with `on ... do` in the rules.
On ESP8266, when "rules cache" is enabled, only these lines are kept in memory along with the position in the rules files where this line is located.
On ESP32, with "rules cache" enabled, also the other lines are kept in memory, but the initial search for these `on...do` lines is the same as on ESP8266.

With the rules cache disabled, this is all read from the flash file system.

That's the only difference the rules cache makes.

If you had earlier builds which worked with this, then those must have been quite old builds.
In earlier builds, the rules were parsed by interpreting all rules files and each line was parsed extensively.
This is slooooowwwww.

What we can consider to do (not saying we will, as I find it a bit hard to see the consequences) is to also call `parseTemplate()` on each `on...do` line, regardless of cache setting, when trying to match the right rules block.
So if this is added, then it will be behind a flag which is default disabled.

The reason I'm a bit reluctant to add this is that I got some feeling we're missing something here where this might backfire a lot.
Can't put my finger on it, but there's something which in the back of my mind warning me that I might have tried this before.

Anyway, the reason to have it disabled by default is that it will slow down rules parsing quite a bit.

TommoT1, 2023-01-01:
Happy New Year!
Thank you for your explanation!
So ist there a way I can have rules with a variable time setting without running in those issues? 
Does it make the difference if I use variables instead of dummy variables to store hours and minutes like it was discussed here? https://letscontrolit.com/forum/viewtopic.php?p=39254#p39254

TD-er, 2023-01-01:
You could act on every minute and compare systime.
To compare time notation to seconds, you can use this: https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#timetomin-timetosec (recently added)

Or you can set a system timer for some specific time. (don't forget to also set it at boot)

TommoT1, 2023-01-01:
Ok, that's how I modified the rules in the node I started this issue with. Every minute I compare to different stored times and decide what to do or not.
For the future I would appreciate if you find a solution to have again a '''on clock#time=all,[var#1#D2]:[var#2#D2]'''

TD-er, 2023-01-01:
> For the future I would appreciate if you find a solution to have again a `on clock#time=all,[var#1#D2]:[var#2#D2] do`

Will think about it...

TD-er, 2023-03-29:
> Ok, that's how I modified the rules in the node I started this issue with. Every minute I compare to different stored times and decide what to do or not. For the future I would appreciate if you find a solution to have again a `on clock#time=all,[var#1#D2]:[var#2#D2] do`

Can you test with the test build from this PR: https://github.com/letscontrolit/ESPEasy/pull/4579 ??

PR #4579 (2023-03-31): [Rules] Use variables in clock events
For example:
```
on clock#time=all,[int#1]:[int#2#D2] Do
```

Fixes: #4412
commit fa91c48f (2023-03-31): [Rules] Use variables in clock events

For example:
```
on clock#time=all,[int#1]:[int#2#D2] Do
```

Fixes: #4412
closed 2023-03-31 by TD-er