letscontrolit/ESPEasy #4415: Empty device name produces broken prometheus metrics

knrdl, 2022-12-13:
A device name (Devices → Edit → Name) can be set to an empty value. The generated Prometheus metrics (/metrics endpoint) are then broken because invalid keys will be generated. This gave me quite a headache when collecting metrics via Telegraf failed.

To reproduce just set an empty device name. The broken metrics look like this (invalid format):
```
# HELP espeasy_uptime current device uptime in minutes
# TYPE espeasy_uptime counter
espeasy_uptime 13264
# HELP espeasy_load device percentage load
# TYPE espeasy_load gauge
espeasy_load 7.89
# HELP espeasy_free_ram device amount of RAM free in Bytes
# TYPE espeasy_free_ram gauge
espeasy_free_ram 15392
# HELP espeasy_free_stack device amount of Stack free in Bytes
# TYPE espeasy_free_stack gauge
espeasy_free_stack 3696
# HELP espeasy_wifi_rssi Wifi connection Strength
# TYPE espeasy_wifi_rssi gauge
espeasy_wifi_rssi -75
# HELP espeasy_wifi_connected Time wifi has been connected in milliseconds
# TYPE espeasy_wifi_connected counter
espeasy_wifi_connected 153644000
# HELP espeasy_wifi_reconnects Number of times Wifi has reconnected since boot
# TYPE espeasy_wifi_reconnects counter
espeasy_wifi_reconnects 29
# HELP espeasy_device_ Values from connected device
# TYPE espeasy_device_ gauge
espeasy_device_{valueName="rssi"} -73
espeasy_device_{valueName="vcc"} 3.15
espeasy_device_{valueName="uptime"} 13264
# HELP espeasy_device_E280 Values from connected device
# TYPE espeasy_device_E280 gauge
espeasy_device_E280{valueName="Temp"} 1.70
espeasy_device_E280{valueName="Hum"} 56.8
espeasy_device_E280{valueName="Pres"} 1006.3
# HELP espeasy_device_ Values from connected device
# TYPE espeasy_device_ gauge
```
The key `espeasy_device_` will be duplicated which is not allowed.

When providing a name they look like this (valid format): 
```
# HELP espeasy_uptime current device uptime in minutes
# TYPE espeasy_uptime counter
espeasy_uptime 13257
# HELP espeasy_load device percentage load
# TYPE espeasy_load gauge
espeasy_load 7.38
# HELP espeasy_free_ram device amount of RAM free in Bytes
# TYPE espeasy_free_ram gauge
espeasy_free_ram 15112
# HELP espeasy_free_stack device amount of Stack free in Bytes
# TYPE espeasy_free_stack gauge
espeasy_free_stack 3696
# HELP espeasy_wifi_rssi Wifi connection Strength
# TYPE espeasy_wifi_rssi gauge
espeasy_wifi_rssi -76
# HELP espeasy_wifi_connected Time wifi has been connected in milliseconds
# TYPE espeasy_wifi_connected counter
espeasy_wifi_connected 153186000
# HELP espeasy_wifi_reconnects Number of times Wifi has reconnected since boot
# TYPE espeasy_wifi_reconnects counter
espeasy_wifi_reconnects 29
# HELP espeasy_device_System Values from connected device
# TYPE espeasy_device_System gauge
espeasy_device_System{valueName="rssi"} -78
espeasy_device_System{valueName="vcc"} 3.14
espeasy_device_System{valueName="uptime"} 13257
# HELP espeasy_device_E280 Values from connected device
# TYPE espeasy_device_E280 gauge
espeasy_device_E280{valueName="Temp"} 1.74
espeasy_device_E280{valueName="Hum"} 56.9
espeasy_device_E280{valueName="Pres"} 1006.3
# HELP espeasy_device_ Values from connected device
# TYPE espeasy_device_ gauge
```

Please disallow in the WebUI to set empty values as device names. Thanks!

TD-er, 2022-12-13:
There are lots of warnings shown in the web interface when you use such highly undesired setups.
But we do allow it, as it would otherwise complicate things.
For example you want to give a specific name to a task, but since it causes a conflict, it cannot be saved. You then need to change the conflicting situation first and then go back to the task to change it to the desired name.

Also there is at least one user who kindly requested to make it possible to use it like this for his specific setup as he somehow needs to have the same names. (not sure anymore why, maybe it had to do with a specific controller, don't know anymore)

tonhuisman, 2022-12-13:
Empty name could be handled in the Prometheus generator, f.e. by using `task<tasknr>` when the name is found to be empty. That should be an easy fix.

TD-er, 2022-12-13:
> That should be an easy fix.

Hmm that's a tricky statement, that's going to bite you later ;)

tonhuisman, 2022-12-14:
@knrdl Once the PR build is finished, can you test the fix? Download can be fetched [from here](https://github.com/letscontrolit/ESPEasy/actions/runs/3699262666)

knrdl, 2022-12-15:
I've tested it successfully with ESP_Easy_mega_20221214_normal_ESP8266_4M1M_VCC.bin:

Metrics with all names set (valid format):
```
# HELP espeasy_uptime current device uptime in minutes
# TYPE espeasy_uptime counter
espeasy_uptime 7
# HELP espeasy_load device percentage load
# TYPE espeasy_load gauge
espeasy_load 63.94
# HELP espeasy_free_ram device amount of RAM free in Bytes
# TYPE espeasy_free_ram gauge
espeasy_free_ram 11672
# HELP espeasy_free_stack device amount of Stack free in Bytes
# TYPE espeasy_free_stack gauge
espeasy_free_stack 3648
# HELP espeasy_wifi_rssi Wifi connection Strength
# TYPE espeasy_wifi_rssi gauge
espeasy_wifi_rssi -51
# HELP espeasy_wifi_connected Time wifi has been connected in milliseconds
# TYPE espeasy_wifi_connected counter
espeasy_wifi_connected 448000
# HELP espeasy_wifi_reconnects Number of times Wifi has reconnected since boot
# TYPE espeasy_wifi_reconnects counter
espeasy_wifi_reconnects 0
# HELP espeasy_device_Display1602 Values from connected device
# TYPE espeasy_device_Display1602 gauge
# HELP espeasy_device_DS1820 Values from connected device
# TYPE espeasy_device_DS1820 gauge
espeasy_device_DS1820{valueName="Temp"} 0.00
# HELP espeasy_device_BMP180 Values from connected device
# TYPE espeasy_device_BMP180 gauge
espeasy_device_BMP180{valueName="Temp"} 19.4
espeasy_device_BMP180{valueName="Pres"} 1000.9
# HELP espeasy_device_Info Values from connected device
# TYPE espeasy_device_Info gauge
espeasy_device_Info{valueName="load"} 64
espeasy_device_Info{valueName="uptime"} 7
espeasy_device_Info{valueName="freeheap"} 13168
espeasy_device_Info{valueName="freestack"} 3264
# HELP espeasy_device_DHT22 Values from connected device
# TYPE espeasy_device_DHT22 gauge
espeasy_device_DHT22{valueName="Temp"} nan
espeasy_device_DHT22{valueName="Hum"} nan
# HELP espeasy_device_BH1750 Values from connected device
# TYPE espeasy_device_BH1750 gauge
espeasy_device_BH1750{valueName="Lux"} 0
# HELP espeasy_device_INA219 Values from connected device
# TYPE espeasy_device_INA219 gauge
espeasy_device_INA219{valueName="Volt"} 32.76
espeasy_device_INA219{valueName="Amp"} -0.000
espeasy_device_INA219{valueName="Power"} -0.003
# HELP espeasy_device_2ina219 Values from connected device
# TYPE espeasy_device_2ina219 gauge
espeasy_device_2ina219{valueName="Voltage"} 32.76
espeasy_device_2ina219{valueName="Current"} -0.000
espeasy_device_2ina219{valueName="Power"} -0.003
# HELP espeasy_device_BME280 Values from connected device
# TYPE espeasy_device_BME280 gauge
espeasy_device_BME280{valueName="Temp"} 0.00
espeasy_device_BME280{valueName="Hum"} 0.0
espeasy_device_BME280{valueName="Pres"} 0.0
# HELP espeasy_device_GY21 Values from connected device
# TYPE espeasy_device_GY21 gauge
# HELP espeasy_device_1115 Values from connected device
# TYPE espeasy_device_1115 gauge
espeasy_device_1115{valueName="Analog"} -32768.00
# HELP espeasy_device_OLED2 Values from connected device
# TYPE espeasy_device_OLED2 gauge
```

Metrics when removing the name from the task "Info" (still valid format):
```
# HELP espeasy_uptime current device uptime in minutes
# TYPE espeasy_uptime counter
espeasy_uptime 8
# HELP espeasy_load device percentage load
# TYPE espeasy_load gauge
espeasy_load 65.38
# HELP espeasy_free_ram device amount of RAM free in Bytes
# TYPE espeasy_free_ram gauge
espeasy_free_ram 11480
# HELP espeasy_free_stack device amount of Stack free in Bytes
# TYPE espeasy_free_stack gauge
espeasy_free_stack 3648
# HELP espeasy_wifi_rssi Wifi connection Strength
# TYPE espeasy_wifi_rssi gauge
espeasy_wifi_rssi -50
# HELP espeasy_wifi_connected Time wifi has been connected in milliseconds
# TYPE espeasy_wifi_connected counter
espeasy_wifi_connected 482000
# HELP espeasy_wifi_reconnects Number of times Wifi has reconnected since boot
# TYPE espeasy_wifi_reconnects counter
espeasy_wifi_reconnects 0
# HELP espeasy_device_Display1602 Values from connected device
# TYPE espeasy_device_Display1602 gauge
# HELP espeasy_device_DS1820 Values from connected device
# TYPE espeasy_device_DS1820 gauge
espeasy_device_DS1820{valueName="Temp"} 0.00
# HELP espeasy_device_BMP180 Values from connected device
# TYPE espeasy_device_BMP180 gauge
espeasy_device_BMP180{valueName="Temp"} 19.3
espeasy_device_BMP180{valueName="Pres"} 1000.9
# HELP espeasy_device_task3 Values from connected device
# TYPE espeasy_device_task3 gauge
espeasy_device_task3{valueName="load"} 65
espeasy_device_task3{valueName="uptime"} 8
espeasy_device_task3{valueName="freeheap"} 13200
espeasy_device_task3{valueName="freestack"} 3264
# HELP espeasy_device_DHT22 Values from connected device
# TYPE espeasy_device_DHT22 gauge
espeasy_device_DHT22{valueName="Temp"} nan
espeasy_device_DHT22{valueName="Hum"} nan
# HELP espeasy_device_BH1750 Values from connected device
# TYPE espeasy_device_BH1750 gauge
espeasy_device_BH1750{valueName="Lux"} 0
# HELP espeasy_device_INA219 Values from connected device
# TYPE espeasy_device_INA219 gauge
espeasy_device_INA219{valueName="Volt"} 32.76
espeasy_device_INA219{valueName="Amp"} -0.000
espeasy_device_INA219{valueName="Power"} -0.003
# HELP espeasy_device_2ina219 Values from connected device
# TYPE espeasy_device_2ina219 gauge
espeasy_device_2ina219{valueName="Voltage"} 32.76
espeasy_device_2ina219{valueName="Current"} -0.000
espeasy_device_2ina219{valueName="Power"} -0.003
# HELP espeasy_device_BME280 Values from connected device
# TYPE espeasy_device_BME280 gauge
espeasy_device_BME280{valueName="Temp"} 0.00
espeasy_device_BME280{valueName="Hum"} 0.0
espeasy_device_BME280{valueName="Pres"} 0.0
# HELP espeasy_device_GY21 Values from connected device
# TYPE espeasy_device_GY21 gauge
# HELP espeasy_device_1115 Values from connected device
# TYPE espeasy_device_1115 gauge
espeasy_device_1115{valueName="Analog"} -32768.00
# HELP espeasy_device_OLED2 Values from connected device
# TYPE espeasy_device_OLED2 gauge
```

"task3" is correctly used as ID instead. That works for me :+1:

tonhuisman, 2022-12-15:
That will be changed to `task4` in your case, as I've not yet added the +1, required to map it to the Task number in the UI.
I'll fix that later today.

tonhuisman, 2022-12-15:
Updated Action build will be available in a short while [from here](https://github.com/letscontrolit/ESPEasy/actions/runs/3707458138)
I also used Uncrustify to format the source for uniformity.

PR #4418 (2022-12-16): [Metrics] Use default devicename `taskN` when Name is empty
Resolves: #4415 

- Use default devicename `task<nr>` when Name is empty
- Reduce used string sizes by some de-duplication

NB: Does **not** correct duplicate devicenames (that is warned for in the UI, and should best be fixed manually)
closed 2022-12-16 by TD-er