letscontrolit/ESPEasy #4424: Sonoff POW R2 is not sending data. mega-20221105

Frank-Huber, 2022-12-20:
Hi,

I downloaded the latest build today and flashed the ESP_Easy_mega_20221105_hard_SONOFF_POW_4M1M.bin image to teh Sonoff.
ESPEasy is running OK, I can get teh pushbutton and control the LED.

but after defining the CSE7766 it is not sending any data. I have for sure mains and a load connected, but nothing.
Is there any other special configuration needed? I searched for it, but could either not find or not understand.

any hint would really be welcome. 

[image]

TD-er, 2022-12-20:
What config do you use on this task?

[image]
This is taken from my POW R2 unit.

Frank-Huber, 2022-12-20:
started with "0" so it took the stored values. 
[image]

Frank-Huber, 2022-12-20:
swapped it and rebooted. but no change. all values zero

Frank-Huber, 2022-12-20:
guys, guess I figured it out. i'm just stupid....
my PCB tells POW v2 which I understood as the R2.
but in fact I have the regular POWR1. configured that one and I have my data. just need to flash back latest buld and calibrate.

sorry for confusion!

[image]

closed 2022-12-20 by Frank-Huber

TD-er, 2022-12-20:
Good to know it is working :)

Frank-Huber, 2022-12-20:
Yes, thanks. :-) but giving teh next challenge. DS18B20 is not included in the POW build and the POW not in the normal...
Have to compile my own build teh next days to get all working as should. but for sure there is no issue in ESPEasy. ;-)
Thanks again!

TD-er, 2022-12-20:
And there is a good reason for the Dallas plugins not being included!

These units, like the Sonoff POW, have the electronics connected to mains voltage.
There is absolutely no isolation!!!!
So you should not have any sensor outside the enclosure of the POW, as you can't isolate it and thus it is really dangerous when you touch it.